jwise/HoRNDIS #33: HoRNDIS kext doesn't load at startup

Obbart, 2015-01-02:
Hi, i'm trying to use HoRNDIS rel7 on my mbp w 10.10.1 to connect to the network created by a linux embedded board, Arietta G25 (http://www.acmesystems.it/arietta).
I installed the pkg with no errors but when I connect the board no interface is shown in OSX networking panel. dmesg and system doesn not show any sign of HoRNDIS being alive, so i tried using KEXT Wizard and I noticed that HoRNDIS Kext is not loaded by OSX at all. Any suggestion? Thanks in advace for support.

mikumi, 2015-07-01:
I have the same issue on 10.10, manually loading the kext after startup helped:

`sudo kextload /Library/Extensions/HoRNDIS.kext`

openhacker, 2015-09-12:
I just noticed, when I don't have a tether and boot, HoRNDIS isn't present.

When I attach a tether, automagically it loads.

Is this expected behavior?