jwise/HoRNDIS #38: kext doesn't load on Yosemite 10.10.2—claims "invalid signature for com.joshuawise.kexts.HoRNDIS"

pragdave, 2015-04-03:
~~~
4/3/15 10:27:40.593 AM sudo[892]:     dave : TTY=ttys001 ; PWD=/Users/dave ; USER=root ; COMMAND=/sbin/kextload /System/Library/Extensions/HoRNDIS.kext/
4/3/15 10:27:40.828 AM com.apple.kextd[22]: ERROR: invalid signature for com.joshuawise.kexts.HoRNDIS, will not load
~~~

jwise, 2015-04-03:
On OS X 10.10, you need to kextload /Library/Extensions/HoRNDIS.kext instead.  The version that is dropped in /System is not signed.

pragdave, 2015-04-04:
Ah, perfect. Thanks!

closed 2015-04-04 by pragdave

mojo2012, 2015-09-08:
Is there a special reason why it is there on both locations?

jwise, 2015-09-08:
@mojo2012 Yes.  Older versions of OS X will only load kexts from /System/Library/Extensions, and expect kexts in that directory not to be signed.  If a signed version of a kext is present there, an older version of OS X will not load it.  Newer versions of OS X will complain if they see an unsigned kext (or simply refuse to load it), but will prefer a signed version of a kext if it exists in /Library/Extensions.  No version of OS X will load an unsigned kext from /Library/Extensions.   More info -- https://developer.apple.com/library/mac/releasenotes/MacOSX/WhatsNewInOSX/Articles/MacOSX10_9.html