jwise/HoRNDIS #39: Homebrew support

thesabbir, 2015-04-17:
It would be nice if we can install HoRNDIS with [homebrew](http://brew.sh/)

jwise, 2015-04-17:
I think you can?

thesabbir, 2015-04-17:
Never done it before. Just a heavy brew user :)

jwise, 2015-04-17:
I am pretty sure there is a brew package already; that said, I'm pretty sure brew can't install kexts on its own, too... 

Closing; feel free to reopen if package does not exist

closed 2015-04-17 by jwise

SimenB, 2015-04-21:
Cask seems to have it, untested though (I'll test it this weekend, can report back then)

https://github.com/caskroom/homebrew-cask/blob/master/Casks/horndis.rb

joelpurra, 2015-06-27:
`brew cask install horndis` *and a reboot* worked very well for me!

Thank you very much @jwise -- now I don't have to enable/disable the buggy Bluetooth PAN to reconnect all the time =)

binaricko, 2017-12-13:
Can confirm that `brew cask install horndis` plus reboot worked for me on OSX Sierra.  Oddly, the `pkg` installer didn't work ... even though it looks like the brew installer downloads the same file?  But I'd had previous versions of HoRNDIS left over from before the Sierra upgrade, which I cleaned out manually before trying the Homebrew install.

tl;dr: After an upgrade to Sierra:
```bash
sudo rm -R /Library/Extensions/HoRNDIS.kext
sudo rm -R /System/Library/Extensions/HoRNDIS.kext
brew cask install horndis
shutdown -r now
```