jwise/HoRNDIS #40: HoRNDIS Spams Console

Blaziken584, 2015-04-25:
I do not believe there seems to be any other issue present, but HoRNDIS completely spams my console and makes it near impossible to debug anything.  My Console always is filled with the following, and usually only the following:

```4/24/15 20:32:12.000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer...
4/24/15 20:32:12.000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer...
4/24/15 20:32:12.000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer...
4/24/15 20:32:12.000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer...
4/24/15 20:32:12.000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer...
4/24/15 20:32:12.000 kernel[0]: *** kernel exceeded 500 log message per second limit  -  remaining messages this second discarded ***```

There are hundreds of thousands of these messages in my Console for an unknown reason.  I am unfortunately stuck to tethering to my phone for the time being for internet, but everything performs fine otherwise.

OS: Mac OS X 10.10.2
Phone: HTC One M7

jwise, 2015-04-27:
Hmm... that is pretty weird.  If this is happening all the time, it means that something in the USB stack is continuously locking up, and somehow is leaking packets.  I'm not really sure where to go to debug that; it indeed isn't fatal, but if you're seeing it under no load, something really is pretty wrong...  It's OK to comment out that message, I suppose, if you have to.

Blaziken584, 2015-05-02:
I'll look into my USB stack then since I am using a hackintosh.  It's possibly a glitch with the USB kext I am using.  I'll certainly be sure to comment it out.  Thanks!

jwise, 2015-05-07:
Ok.  Hackintosh is not a platform I can reasonably support, so I'm going to close this one out.  If you come up with a root cause, feel free to reopen and I'm happy to take patches for it, though.

closed 2015-05-07 by jwise