jwise/HoRNDIS #41: Kernel Panic in 10.10

vignesh0025, 2015-05-23:
Anonymous UUID:       778D6A60-C3AC-9FF8-1B8F-BC5EF68B15FB

Thu May 21 00:29:12 2015

*** Panic Report ***
panic(cpu 3 caller 0xffffff8000417cc2): Kernel trap at 0xffffff800040b6c7, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000008, CR3: 0x000000013aae500e, CR4: 0x00000000001627e0
RAX: 0x000000000000060e, RBX: 0xffffff801b3dd000, RCX: 0x0000000000000600, RDX: 0x0000000000000000
RSP: 0xffffff80a8b939d0, RBP: 0xffffff80a8b939d0, RSI: 0xffffff80a47b0000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0xffffff80005900a8, R10: 0xffffff80a8b9396c, R11: 0xffffff8000590080
R12: 0xffffff80a47b0000, R13: 0x0000000000000000, R14: 0x0000000000000156, R15: 0xffffff80a47b0000
RFL: 0x0000000000010297, RIP: 0xffffff800040b6c7, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000008, Error code: 0x0000000000000000, Fault CPU: 0x3

Backtrace (CPU 3), Frame : Return Address
0xffffff80a8b93680 : 0xffffff800032bda1 
0xffffff80a8b93700 : 0xffffff8000417cc2 
0xffffff80a8b938c0 : 0xffffff8000434b73 
0xffffff80a8b938e0 : 0xffffff800040b6c7 
0xffffff80a8b939d0 : 0xffffff7f816f4757 
0xffffff80a8b93a20 : 0xffffff7f80f48a40 
0xffffff80a8b93a60 : 0xffffff7f80f48776 
0xffffff80a8b93ad0 : 0xffffff7f80f489a2 
0xffffff80a8b93b10 : 0xffffff80005a2cac 
0xffffff80a8b93c50 : 0xffffff800058d1a3 
0xffffff80a8b93ce0 : 0xffffff8000580cbb 
0xffffff80a8b93d50 : 0xffffff800056f600 
0xffffff80a8b93dd0 : 0xffffff8000564b5f 
0xffffff80a8b93e40 : 0xffffff80007ecc05 
0xffffff80a8b93ef0 : 0xffffff80007eca52 
0xffffff80a8b93f50 : 0xffffff800084ba86 
0xffffff80a8b93fb0 : 0xffffff8000435376 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.2)[247581D8-DFC5-3AAE-B079-3D13F5514C2B]@0xffffff7f80f37000->0xffffff7f80f69fff
         com.joshuawise.kexts.HoRNDIS(6.0)[A615230B-182E-3BF9-B65B-A7947F2B34EF]@0xffffff7f816f2000->0xffffff7f816f8fff
            dependency: com.apple.iokit.IONetworkingFamily(3.2)[247581D8-DFC5-3AAE-B079-3D13F5514C2B]@0xffffff7f80f37000
            dependency: com.apple.iokit.IOUSBFamily(720.4.4)[0D6EABE6-1CC7-348C-8BE2-AF04F66722BF]@0xffffff7f8110c000

BSD process name corresponding to current thread: configd
Boot args: dart=0 nv_disable=1 -gux_defer_usb2 kext-dev-mode=1 slide=0 

Mac OS version:
14D136

Kernel version:
Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64
Kernel UUID: 4B3A11F4-77AA-3D27-A22D-81A1BC5B504D
__HIB  text base: 0xffffff8000100000
System model name: MacBookPro6,2 (Mac-F22589C8)

System uptime in nanoseconds: 1385277392260
last loaded kext at 970450935869: com.apple.driver.AppleUSBCDCECMData	4.3.3b1 (addr 0xffffff7f82297000, size 28672)
last unloaded kext at 758525790983: com.apple.driver.AppleUSBCDC	4.3.3b1 (addr 0xffffff7f8228c000, size 16384)
loaded kexts:
org.tw.CodecCommander	2.3.4
com.protech.NoSleep	1.3.3
com.joshuawise.kexts.HoRNDIS	6
com.darwin.driver.ACPIBacklight	2.0.3
org.rehabman.voodoo.driver.PS2Trackpad	1.8.15
org.rehabman.voodoo.driver.PS2Keyboard	1.8.15
com.insanelymac.RealtekRTL8100	1.0.0
net.osx86.kexts.GenericUSBXHCI	1.2.7
org.rehabman.driver.FakePCIID	1.1.1
org.rehabman.voodoo.driver.PS2Controller	1.8.15
org.hwsensors.driver.CPUSensors	1655
org.rehabman.driver.AppleSmartBatteryManager	1.55
org.netkas.driver.FakeSMC	1655
com.apple.driver.AppleUSBCDCECMData	4.3.3b1
com.apple.driver.AppleUSBCDCACMData	4.3.3b1
com.apple.driver.AppleUSBCDC	4.3.3b1
com.apple.filesystems.ntfs	3.12
com.apple.driver.AudioAUUC	1.70
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.autofs	3.0
com.apple.iokit.IOBluetoothSerialManager	4.3.4f4
com.apple.driver.AppleHDA	272.18
com.apple.driver.AppleGraphicsDevicePolicy	3.7.7
com.apple.driver.AppleUpstreamUserClient	3.6.1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHWAccess	1
com.apple.driver.AppleIntelHD5000Graphics	10.0.6
com.apple.driver.AppleHV	1
com.apple.driver.AppleLPC	1.7.3
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.driver.AppleIntelFramebufferAzul	10.0.6
com.apple.driver.AppleMCCSControl	1.2.11
com.apple.iokit.SCSITaskUserClient	3.7.5
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0d1
com.apple.BootCache	36
com.apple.driver.XsanFilter	404
com.apple.iokit.IOAHCIBlockStorage	2.7.1
com.apple.driver.AppleUSBHub	705.4.2
com.apple.driver.AppleAHCIPort	3.1.2
com.apple.driver.AppleUSBXHCI	710.4.11
com.apple.driver.AppleUSBEHCI	705.4.14
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleACPIEC	3.1
com.apple.driver.AppleACPIButtons	3.1
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	161
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.kext.triggers	1.0
com.apple.driver.DspFuncLib	272.18
com.apple.kext.OSvKernDSPLib	1.15
com.apple.iokit.IOBluetoothFamily	4.3.4f4
com.apple.iokit.IOSurface	97.4
com.apple.iokit.IOUSBHIDDriver	705.4.0
com.apple.iokit.IOUSBUserClient	705.4.0
com.apple.driver.AppleGraphicsControl	3.10.22
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.driver.AppleHDAController	272.18
com.apple.iokit.IOHDAFamily	272.18
com.apple.iokit.IOAudioFamily	203.3
com.apple.vecLib.kext	1.2.0
com.apple.iokit.IOSerialFamily	11
com.apple.AppleGraphicsDeviceControl	3.10.22
com.apple.iokit.IOAcceleratorFamily2	156.14
com.apple.driver.X86PlatformPlugin	1.0.0
com.apple.driver.IOPlatformPluginFamily	5.9.1d7
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IONDRVSupport	2.4.1
com.apple.driver.AppleSMBusController	1.0.13d1
com.apple.iokit.IOGraphicsFamily	2.4.1
com.apple.driver.AppleUSBComposite	705.4.9
com.apple.iokit.IOSCSIMultimediaCommandsDevice	3.7.5
com.apple.iokit.IOBDStorageFamily	1.7
com.apple.iokit.IODVDStorageFamily	1.7.1
com.apple.iokit.IOCDStorageFamily	1.7.1
com.apple.iokit.IOAHCISerialATAPI	2.6.1
com.apple.iokit.IOSCSIArchitectureModelFamily	3.7.5
com.apple.iokit.IONetworkingFamily	3.2
com.apple.iokit.IOAHCIFamily	2.7.5
com.apple.iokit.IOUSBFamily	720.4.4
com.apple.driver.AppleEFINVRAM	2.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	396
com.apple.iokit.IOStorageFamily	2.0
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	3.1
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookPro6,2, BootROM MBP61.0057.B0C, 2 processors, Intel Core i5, 2.29 GHz, 4 GB, SMC 1.58f17
Graphics: Intel HD Graphics 4400, Intel HD Graphics 4400, Built-In
Memory Module: BANK0/DIMM0, 4 GB, DDR3, 1600 MHz, Hynix Semiconductor, HMT351S6EFR8A-PBN0
Network Service: Moto E, Ethernet, en1
Serial ATA Device: HGST HTS541010A9E680, 1 TB
Serial ATA Device: hp       CDDVDW SU-208CB
USB Device: Hub
USB Device: Moto E
USB Device: HP Truevision HD
USB Device: Lenovo Wireless Optical Mouse N100
Thunderbolt Bus:

jwise, 2015-05-23:
Try upgrading to rel7?

vignesh0025, 2015-05-23:
Yup. Did it. I will report back if any panic occurs.

mplucinski, 2015-09-29:
Happens to me on the latest version too. It's reproducible, in the sense that it happens at least once a day. Here is Apple's error report:

Anonymous UUID:       4D5AE888-E0B8-9DF7-840F-C3C85AD95A59

Tue Sep 29 10:27:40 2015

*** Panic Report ***
panic(cpu 3 caller 0xffffff8002e16df2): Kernel trap at 0xffffff8002e0a7f7, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000008, CR3: 0x000000026119a00b, CR4: 0x00000000000626e0
RAX: 0x00000000000005dc, RBX: 0xffffff801f3c7000, RCX: 0x00000000000005ce, RDX: 0x0000000000000000
RSP: 0xffffff800e7ab9d0, RBP: 0xffffff800e7ab9d0, RSI: 0xffffff8105c5df00, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0xffffff8002f8f218, R10: 0xffffff800e7ab96c, R11: 0xffffff8002f8f1f0
R12: 0xffffff8105c5df00, R13: 0x0000000000000000, R14: 0x0000000000000156, R15: 0xffffff8105c5df00
RFL: 0x0000000000010293, RIP: 0xffffff8002e0a7f7, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000008, Error code: 0x0000000000000000, Fault CPU: 0x3

Backtrace (CPU 3), Frame : Return Address
0xffffff800e7ab680 : 0xffffff8002d2ad21 
0xffffff800e7ab700 : 0xffffff8002e16df2 
0xffffff800e7ab8c0 : 0xffffff8002e33ca3 
0xffffff800e7ab8e0 : 0xffffff8002e0a7f7 
0xffffff800e7ab9d0 : 0xffffff7f85504757 
0xffffff800e7aba20 : 0xffffff7f83923a4a 
0xffffff800e7aba60 : 0xffffff7f83923780 
0xffffff800e7abad0 : 0xffffff7f839239ac 
0xffffff800e7abb10 : 0xffffff8002fa1e1c 
0xffffff800e7abc50 : 0xffffff8002f8c313 
0xffffff800e7abce0 : 0xffffff8002f7fe2b 
0xffffff800e7abd50 : 0xffffff8002f6e770 
0xffffff800e7abdd0 : 0xffffff8002f63ccf 
0xffffff800e7abe40 : 0xffffff80031ec1a5 
0xffffff800e7abef0 : 0xffffff80031ebff2 
0xffffff800e7abf50 : 0xffffff800324b376 
0xffffff800e7abfb0 : 0xffffff8002e344a6 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.2)[150D35F0-A5B1-3C6C-803A-A45B70166285]@0xffffff7f83912000->0xffffff7f8393dfff
         com.joshuawise.kexts.HoRNDIS(6.0)[A615230B-182E-3BF9-B65B-A7947F2B34EF]@0xffffff7f85502000->0xffffff7f85508fff
            dependency: com.apple.iokit.IONetworkingFamily(3.2)[150D35F0-A5B1-3C6C-803A-A45B70166285]@0xffffff7f83912000
            dependency: com.apple.iokit.IOUSBFamily(720.4.4)[4ED60649-760E-3E91-9C58-322700847C7E]@0xffffff7f83951000

BSD process name corresponding to current thread: configd

Mac OS version:
14F27

Kernel version:
Darwin Kernel Version 14.5.0: Wed Jul 29 02:26:53 PDT 2015; root:xnu-2782.40.9~1/RELEASE_X86_64
Kernel UUID: 58F06365-45C7-3CA7-B80D-173AFD1A03C4
Kernel slide:     0x0000000002a00000
Kernel text base: 0xffffff8002c00000
__HIB  text base: 0xffffff8002b00000
System model name: MacBookPro8,1 (Mac-94245B3640C91C81)

System uptime in nanoseconds: 88987241585965
last loaded kext at 155649380342: com.apple.driver.AppleUSBCDCACMControl	4.3.3b1 (addr 0xffffff7f83bfb000, size 20480)
loaded kexts:
org.virtualbox.kext.VBoxNetAdp	4.3.30
org.virtualbox.kext.VBoxNetFlt	4.3.30
org.virtualbox.kext.VBoxUSB	4.3.30
com.avira.kext.FileAccessControl	1.0.0d1
org.virtualbox.kext.VBoxDrv	4.3.30
com.joshuawise.kexts.HoRNDIS	6
com.apple.driver.AppleUSBCDC	4.3.3b1
com.apple.filesystems.msdosfs	1.10
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AppleTyMCEDriver	1.0.2d2
com.apple.driver.AGPM	110.19.6
com.apple.filesystems.autofs	3.0
com.apple.iokit.IOBluetoothSerialManager	4.3.6f3
[... 108 lines cut ...]
com.apple.iokit.IOFireWireFamily	4.5.8
com.apple.iokit.IO80211Family	730.60
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.iokit.IONetworkingFamily	3.2
com.apple.iokit.IOAHCIFamily	2.7.5
com.apple.iokit.IOUSBFamily	720.4.4
com.apple.driver.AppleEFINVRAM	2.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	398
com.apple.iokit.IOStorageFamily	2.0
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	3.1
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.corecrypto	1.0
com.apple.kec.Libm	1
com.apple.kec.pthread	1