jwise/HoRNDIS #42: 10.11

DiegoGiovany, 2015-06-19:
Not working on OSX El Capitain:

Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: HoRNDIS: init: HoRNDIS tethering driver for Snow Leopard+, by Joshua Wise
Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: 006028.794737 IOUSBHostHIDDevice@: IOUSBHostHIDDevice::start: unable to set device idle milliseconds
Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: 006028.794750 IOUSBHostHIDDevice@: IOUSBHostHIDDevice::start: start has failed
Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: 006028.797808 IOUSBHostHIDDevice@: IOUSBHostHIDDevice::start: unable to set device idle milliseconds
Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: 006028.797820 IOUSBHostHIDDevice@: IOUSBHostHIDDevice::start: start has failed
Jun 18 21:35:33 MBPRetiDeuschle kernel[0]: IOHIDDevice::newUserClient failed to get a workloop

jwise, 2015-06-19:
Sorry to hear that.  Unfortunately, I only have one Mac, and I cannot upgrade my 'production' machine to El Cap ... Yosemite broke my machine badly enough, and I don't want to run the risk of breaking it again with El Cap.  So, until El Cap becomes final, I'm afraid I won't be able to provide much support for it.  Sorry.

DiegoGiovany, 2015-06-20:
Thanks !!!! 
You can install 10.11 on a usb drive, like I had done...  I had the same issues with yosemite.... 
I can do some tests and traces if it helps.
Thanks again, you work is awesome!

winsock, 2015-06-20:
I'm willing to help on this, I am running 10.11 for updating some of my Apps. HoRNDIS also works if and only if you start tethering before the kext loads so right now my workflow is. Although I do not get any errors like @DiegoGiovany has in my kernel logs.

    1. Start tethering
    2. kextunload HoRNDIS.kext
    3. kextload HoRNDIS.kext

DiegoGiovany, 2015-06-20:
Nice workaroud @winsock, just tried, and works !!!

jwise, 2015-06-25:
Is this any better in 10.11 b2?

DiegoGiovany, 2015-06-25:
The interface Ultra-Ultra-Smooth now, serious, its almost lagfre, evento with 20 or more windows ...  Safari is not crashing anymore... The only one thing that still not working is the virtualhere drivers, they are causing a Kernel panic...

DiegoGiovany, 2015-06-25:
Nops, still need the kext unload/load workaround... same erros on console...

jwise, 2015-07-24:
Nope... Last time I installed an OS X beta on my laptop, it was version 10.10.0, and I got bitten by it :-) I'll probably hold off until 10.11.1 before I upgrade my production machine.  Sorry.

jwise, 2015-08-03:
Another report with different logs:

kernel[0]: Couldn't alloc class "AppleUSBComposite"
kernel[0]: 006570.648877 SAMSUNG_Android@14100000: AppleUSBDevice::waitForInterfacesGated: timeout waiting for _interfacesMatched
kernel[0]: 006570.648909 SAMSUNG_Android@14100000: AppleUSBDevice::waitForInterfacesGated: timeout waiting for _interfacesMatched

fuantaji, 2015-08-03:
The kextunload/kextload workaround doesn't work for me.

On the side note I've tried compiling HoRNDIS with later SDK versions.
With 10.10 SDK I've hit a wall of invalid signature on kextload.
With 10.11 SDK HoRNDIS sources can't be compiled since IOKit/usb header structure seems to be different now or [more likely] it is missing some files.

winsock, 2015-08-04:
@fuantaji They rewrote the I/O code in 10.11 So the SDK is complete just well needs a ton of work to port. Almost all of the functions use deprecated methods/classes.

LouisCo, 2015-08-28:
Hi,
I should start learning iOS (so Swift and Obj-C) in less than 10 months. Right now, I'm an Android Developer, running El Capitan public beta on my Macbook. If I can help to make this work on next OS X, let me know. Note that I've little knowledge in C and C++ (in addition to Java, Android and web).

N1Xua, 2015-09-11:
Hi ,Winsock/Fuantaji,

Did apple not deliver a ( non-working) RNDIS solution in El Capitan.
our Device that connected as CDC-ECN (Ethernet) class device suddenly connects over RNDIS and  does not show up as network interface, as if the  network part never gets set up

I think that you are better off with a defect about that @ the developer.apple.com in this case.

winsock, 2015-09-11:
I can try to leave a bug report in radar about the oversight of a RNDIS driver. However, who knows when/if apple will actually follow through with that bug report.

winsock, 2015-09-22:
@jwise I'm not too experienced with kernel extension development/debugging. I think I got most of the code ported to 10.11, however I get a KP... Since you know your codebase better than me I thought my port still could be of use to you. 

https://github.com/winsock/HoRNDIS/tree/elcaptian

Regards,

Andrew Querol

winsock, 2015-09-23:
I narrowed down my current kp to getting the input and output data pipes. To continue any further I'll have to setup a slave computer running the debug kernel so I can use lldb and step though the code -.-
https://github.com/winsock/HoRNDIS/commit/c3182ba4dba70c790a1e5823b3abf2afdfe9980a

jwise, 2015-09-24:
Andrew -- thanks for your efforts!  I unfortunately don't have a 10.11 machine (I probably won't update for a while, given my experience with 10.10).

It surprises me that so much changed incompatibly.  I wonder if there's a smaller, compatible, change that can be made?  It would suck to have to maintain two versions of the driver (one for <=10.10, and one for 10.11+); I've avoided that so far, having support all the way back to 10.6 through 10.10.  

Is the crash inside the kernel, or inside the kext?  (i.e., is the crash on that line related to a potentially null endpointDescriptor, or does it have to do with the actual copyPipe operation?)

winsock, 2015-09-24:
From what I read of you keep compiling with the older sdk it'll still work on El Captain as long as it also works in Yosemite. My guess is since they changed so much with 10.11 a bug that was probably hidden or a non issue for 10.10 and below now shows up since the implementation of the older api would've changed. The reason I had to port so much was due to only having the 10.11 sdk. However while porting I noticed and fixed quite a few bugs and questionable code that might fix it with the old sdk as well. @jwise what is the exact build environment you use at the moment? What osx sdk/etc. With any luck back porting some of my general fixes and cleanup might do the trick ;)

winsock, 2015-09-24:
For what it's worth it's my KP was inside the extension. Most likely the page fault was caused by the descriptor being incorrect when I ported it to the new style. Causing the function to return an invalid or null pointer that I access and attempt to copy for the pipe.

winsock, 2015-09-26:
HoRNDIS: init: HoRNDIS tethering driver for Mac OS X, by Joshua Wise
HoRNDIS: init: PTR: I am: <ptr>
HoRNDIS: start: start
HoRNDIS: createWorkLoop: creating workloop
HoRNDIS: openInterfaces: PTR: fCommInterface: <ptr>
HoRNDIS: message: kIOUSBMessagePortHasBeenResumed
HoRNDIS: stop: stop

Hmmm building with the 10.6 SDK now. Made some changes to code that I thought looked funny but still no dice. Also the weird part is that after stop is called HoRNDIS actually retains control of the USB pipe! Whatever port I used will be locked up till reboot -.- @jwise Any idea why these things are happening? With the early stop call and holding the pipe?

saamish, 2015-10-01:
@winsock do there possibility to compile horndis over the 10.11.1 because of changes in IOKit framework:

https://developer.apple.com/library/prerelease/mac/releasenotes/General/APIDiffsMacOSX10_11/Objective-C/Kernel.html

I was upgrading the latest xcode from the apple store but it did not give any help for the compile issues. el capitan contains buch of limited rules for root priviliges. am I right that horndis is needed to port from yosemite to el capitan until it works like a charm.

rdjs, 2015-10-01:
@winsock 's workaround didn't fix for me, is it still valid?

easye, 2015-10-01:
On 10/1/15 14:10, Russell wrote:
> @winsock 's workaround didn't fix for me, is it still valid?

Using kextunload/kextload worked for me: I just upgraded to OX 10.11
when it became generally available today.

I had to unplug/replug the phone a couple times as well as specify a
full absolute path to the kext, but it eventually worked.

-- 
"No, this is not a disentanglement from, but a progressive /knotting-into/

skylerbunny, 2015-10-02:
Another thing that 'made this work' that I just did myself, is a wipe of network preferences:

$ sudo rm -rf Library/Preferences/SystemConfiguration/NetworkInterfaces.plist
$ sudo rm -rf /Library/Preferences/SystemConfiguration/preferences.plist

$ sudo reboot

It appears that it may only have worked for the first attempt after the reboot, to tether. On subsequent attempts it was not 'latching onto' the newly created device, nor creating a new one to work with it. This may be useful information, however.

saamish, 2015-10-02:
i have used command: kextunload -bundle and it seems to work but there is still something weird ongoing. .

are current issues direct relations baed on iokit framework changes? el capitan contains bunch of changes related to root privileges. @jwise what is your opinion do horndis needed to port for el capitan to get it  work as charm because of changes in root rights. i know that "apple specific" (kernel??) extensions are loaded everytime when os x 10.11 is booted but how about 3rd party extensions? do there some of limited rules for 3rd party extension from el capitan onward?

d235j, 2015-10-02:
@winsock you can set up two-machine debugging fairly easily using VMWare. I've done it before and I'm setting it up now to maybe do a little debugging here. http://ho.ax/posts/2012/02/debugging-the-mac-os-x-kernel-with-vmware-and-gdb/ has a decent though slightly dated walkthrough. (It's a bit tricky to turn off SIP in VMWare, but it's possible by adding bios.bootDelay = "5000" to the .vmx file, then selecting the disk rather than the partition in the boot manager while holding Cmd-R, then opening Terminal and running csrutil.)

d235j, 2015-10-02:
@winsock you're getting a null pointer dereference at https://github.com/winsock/HoRNDIS/blob/elcaptian/HoRNDIS.cpp#L299 , with endpointDescriptor being null...

d235j, 2015-10-02:
It might be worthwhile to look at https://github.com/RehabMan/OS-X-BrcmPatchRAM/blob/master/BrcmPatchRAM/USBHostDeviceShim.cpp for some guidance, or even perhaps use that shim directly (it is license-compatible with HoRNDIS).

jwise, 2015-10-02:
That might be useful.  The thing that I find puzzling is that there are sporadic reports of the driver successfully loading, somehow!  I wonder if it would be more productive to try to figure out what the conditions are under which the driver does successfully load, and then try to make those happen, rather than going down this dance of rewriting the driver to support an entirely new API?

jwise, 2015-10-02:
To expand on that some -- I still haven't seen any Apple documentation that says, "if you have a USB driver for the old way of doing things, it's not supported anymore, and you need to rewrite the entire low-level USB layer of your driver".  So I am a little loathe to keep going down that path unless it proves absolutely necessary...

d235j, 2015-10-02:
You have to rewrite it if you want it to compile with Xcode 7 against the 10.11 SDK. They took out the compatibility headers — to compile it using the old API you need a copy of Xcode 6.4 with the 10.10 SDK around.

I do agree that fixing the old-API version should be done, though.

jwise, 2015-10-02:
Yeah, I actually still build against the 10.6 SDK.  Sigh, that is super annoying -- I didn't realize they ripped out the API.  Ok, thanks for the info there.

It does then sound like the old API is not long for this world, but maybe we can squeeze one more major version out of it?

d235j, 2015-10-02:
That shim I linked earlier appears to be set up to allow writing a common codebase that targets both APIs. I'm not sure how well it works in practice, though.

jwise, 2015-10-02:
Yeah, I think that's probably a good fallback option if we have to go that way.  It means keeping two SDKs around, and complicating the build by building three kexts (10.6-10.9 unsigned, 10.10 signed, 10.11 signed), but I guess it could work.  I still am somehow hoping beyond hope that there's some matching change that we can make to the old driver to get it to match more reliably in El Cap.

d235j, 2015-10-02:
I don't see why it shouldn't be possible to fix the old driver to work correctly in El Cap...

winsock, 2015-10-03:
The driver will work in El Cap compiled with 10.6 SDK (I'm compiling with Xcode 7 with manually copied 10.6 sdk) We just need to fix the intermediate issues that remain. @d235j thanks for confirming that, however I don't know if going along with the new API's is the proper way. I actually use Rehabman's shim on my system already (It's not Apple hardware :P) to get USB3 behaving.

d235j, 2015-10-03:
@winsock, it's not working for me. With full debugging enabled, I very frequently get fCommInterface: 0, and the driver bails out. I'm still not sure why...
The device I'm testing with is a BeagleBone Black, which supports standard RNDIS.

winsock, 2015-10-03:
@d235j What I meant is that it "should" work after we find the weird intermittent issue. I was stating that El Cap still properly loads kext's linked agains the old apis

d235j, 2015-10-03:
Ahh, right. Yeah, I'm aware that the old APIs are still binary-compatible.
Any ideas on what may be causing this intermittent issue?

winsock, 2015-10-03:
I thought originally some odd memory management errors were to blame, but after fixing the ones I found when making the 10.11sdk port it still erred out 😓

winsock, 2015-10-03:
I almost wonder if it would be advantageous to make a CDC Controller plugin for RNDIS making use of the existing apple code.

saamish, 2015-10-03:
I compiled extension over the 10.11.1 by copying needed headers into extra folder etc. I used Xcode 7.1 beta release too. This is my very first time what I have done anything with OS X so maybe I did something wrong because of extensions is compiled successfully ;)  But seems that it still behaving as previous 7 release downloaded from @jwise github.

winsock, 2015-10-03:
I can compile alright without regressions so you're did it alright, still doesn't fix the issue at hand. I'm still bug hunting @saamish I wish I could see their git diff files for their USB pipeline xD

d235j, 2015-10-03:
@winsock look at my PR — adding those values to the plist makes it work reliably, the first time you plug in. It does not reinitialize correctly, though.

saamish, 2015-10-04:
@winsock ur right and I wish too :)

saamish, 2015-10-05:
I have faced bunch of other errors printed to kernel log when horndis driver is unplugged. Seems that some of errors  were existed from 2012 already. Example 

1) SIOCGINFO_IN6 failed, device not configured. 
2) nw_interface_create_with_name_netutil_ifname_toi_ifindex_(enX) failed.
3) mdnsresponder getextendflags SIOCGIFLAGS failed errno = 6  ( related to point 1)

etc. so I think real issues were inside of basic os x components

grungebob, 2015-10-05:
Hi,
I am sorry, but I'm completely new to this, and the wifi card or wifi cable has broken in my Macbook (it's a late 2008/ early 2009 and there was a design flaw to that system ugh). 

I have downloaded HoRNDIS and my phone allows tethering, but the issue is that I don't understand how to perform the workaround. Is that something that I need to input into terminal?

Any help is very appreciated.

Thanks

wcontello, 2015-10-06:
I can confirm that HoRNDIS is not working on 10.11. I was also using with the Beaglebone Black.

saamish, 2015-10-08:
@grungebob yes you have to run those scripts in terminal and you need to install driver first.

$sudo kextload /Library/Extension/HoRNDIS.kext --> device should be recognized 
$sudo kextunload /System/Library/Extension/HoRNDIS.kext --> when device is unplugged

wcontello, 2015-10-09:
I plug in the Beaglebone Black, then issue the cli command:
Proton:~ wayne$ sudo kextload /Library/Extension/HoRNDIS.kext

then I get:
/Library/Extension/HoRNDIS.kext failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).

I have installed the driver from the package:
HoRNDIS-rel7.pkg

Log entries from the console:
Oct  9 15:30:52 Proton sudo[8788]:    wayne : TTY=ttys011 ; PWD=/Users/wayne ; USER=root ; COMMAND=/sbin/kextload /Library/Extension/HoRNDIS.kext
Oct  9 15:30:52 Proton com.apple.kextd[55]: Can't open CFBundle for /Library/Extension/HoRNDIS.kext.
Oct  9 15:30:52 --- last message repeated 1 time ---
Oct  9 15:30:52 Proton com.apple.kextd[55]: Error: Kext /Library/Extension/HoRNDIS.kext - not found/unable to create.
Oct  9 15:30:55 Proton WindowServer[242]: send_datagram_available_ping: pid 267 failed to act on a ping it dequeued before timing out.

any thoughts?

nordbit, 2015-10-12:
/System/Library/Extension/HoRNDIS.kext 

not

/Library/Extension/HoRNDIS.kext

jwise, 2015-10-12:
@nordbit: Nope; on 10.10+, you need /Library/Extensions. But @wcontello: /Library/Extensions, not /Library/Extension.

In other news, I did upgrade my laptop to 10.11 the other day.  So maybe sometime in the next few months I will give this a go :-)

crodjer, 2015-10-12:
`kextload /System/Library/Extensions/HoRNDIS.kext/` seems to not work. `kextutil` says that the extension needs to be code signed:

```
└─(Extensions)-> sudo kextutil HoRNDIS.kext/
Diagnostics for /System/Library/Extensions/HoRNDIS.kext:
Code Signing Failure: not code signed
Untrusted kexts are not allowed
ERROR: invalid signature for com.joshuawise.kexts.HoRNDIS, will not load
```

nordbit, 2015-10-14:
@jwise :
```
 So maybe sometime in the next few months I will give this a go :-)
```
You mean next few days, i hope? :-)

davidcastellani, 2015-10-17:
The below worked for me on 10.11.0

1. Start tethering
2. kextunload /Library/Extensions/HoRNDIS.kext
3. kextload /Library/Extensions/HoRNDIS.kext

LouisCAD, 2015-10-17:
@davidcastellani So if you put it in a sh file that you wrap in an app, you may just have to open the app after starting tethering, right?

davidcastellani, 2015-10-17:
You could do that, though you would have to run that particular app whenever you plugged in your phone.

I created a file on my Macbook Pro's desktop called usb_tether.command with the contents below. 

#!/bin/bash might not be needed, I was originally executing the below via command line.

````
#!/bin/bash
sudo -v
sudo kextunload /Library/Extensions/HoRNDIS.kext
sudo kextload /Library/Extensions/HoRNDIS.kext
````

I just turn on usb tether on my phone, plug in the usb cable, then double click usb_tether.command and it seems to work. Try it and report back.

jwise, 2015-10-18:
Ok, I took a crack at it today, and I ran into the grief that apparently my machine doesn't understand how to sign binaries anymore, with the message: 

```
joshua@bruges:~/projects/horndis/HoRNDIS$ codesign --force -s "54GTJ2AU36" build/signed/HoRNDIS.kext
build/signed/HoRNDIS.kext: object file format unrecognized, invalid, or unsuitable
```

I'd try building with a newer SDK, but the newer SDKs don't support 10.6, which I would like to maintain compatibility with.  I guess I could have two xcodeproj files (one for 10.6-10.9, one for 10.10+), butthis seems particularly awful.  I have no idea what makes the object file so unsuitable... I guess I get to go build codesign from source and instrument it.  Thanks, Apple!

jwise, 2015-10-18:
`codesign` cannot really be built from source, because Mac OS X is a dumpster fire.  I grabbed an old unsigned kext and tried to sign it with a current `codesign`, assuming that there was something broken in the build, because Mac OS X is a dumpster fire.  It turns out that it wasn't the fault of the build process, but instead a regression of some form in `codesign` (or, more likely, the `Security` framework`), because Mac OS X is a dumpster fire.  I had a few technical support incidents left from when I bought a developer program membership, but I don't anymore because my developer program membership expired, and anyway, all they will tell me to do is [file a radar, which is the same as just sort of laughing at you](http://www.dourish.com/goodies/see-figure-1.html), because Mac OS X is a dumpster fire.  So this means that in order to build a binary that will run on Mac OS X versions 10.6 through 10.11, I need to be running a version of Mac OS X no later than 10.10, **because Mac OS X is a dumpster fire**.

Can someone remind me why we still do the unpaid work -- even, really, pay $99 a year for the privilege of -- writing drivers that Apple can't be bothered to, when they actively make our lives more difficult at every step in the process?

notthetup, 2015-10-18:
*hugs*! I feel you @jwise. Keeping compatibility with OSX updated has always been mess. We appreciate your efforts!

[411 more replies]

commit 920d799b (2016-10-16): HoRNDIS: rel8pre2 becomes rel8 (fixes #42)
closed 2016-10-16 by jwise