jwise/HoRNDIS #44: Horndis spamming console and connection stop working

pierre-lagarde, 2015-07-22:
Hi, 

First, thanks a lot for the great work, been using your extension for two years now and it's a great tool I can't live without.
Though, since I use it on my macBookPro, I'm having the same behavior that is described on this closed file : 
https://github.com/jwise/HoRNDIS/issues/40
(same message repeated n times for instance "22/07/2015 11:47:28,000 kernel[0]: HoRNDIS: outputPacket: waiting for buffer..." and also "22/07/2015 11:47:28,000 kernel[0]: HoRNDIS: outputPacket: timed out waiting for buffer")
but with a regular macBook Pro with regular Mavericks 10.9.5.
Plus that each time the console start to be spammed, the connection is timing out and I have to stop and start again the usb tethering on my phone to get my connection back to life. Some websites are presumably more likely to cause the issue than others but in fact it can happen while loading any website if I surf quickly for instance.
It makes the tool nearly unusable.
My phone is a Zenfone Asus under Lollipop 5.0.1
I added a list of possibly useful extensions and information at the end of this message, but tell me if you need any more information (I'm not used to github reporting, so just tell me what is the most accurate to do and report and I'll do) 
And maybe I missed something in your manuals, so just tell me and I will apologize quickly ;).

Best regards and keep up.

>> uname -a
Darwin macBookPro 13.4.0 Darwin Kernel Version 13.4.0: Wed Mar 18 16:20:14 PDT 2015; root:xnu-2422.115.14~1/RELEASE_X86_64 x86_64

>> sudo kextstat|grep -i rndis
 133    0 0xffffff7f822e6000 0x7000     0x7000     com.joshuawise.kexts.HoRNDIS (6) <36 34 5 4 3 1>

>> sudo kextstat|grep -i usb
   34   17 0xffffff7f80d0b000 0x61000    0x61000    com.apple.iokit.IOUSBFamily (686.4.1) <11 7 5 4 3 1>
   35    0 0xffffff7f80e76000 0x1d000    0x1d000    com.apple.driver.AppleUSBXHCI (683.4.0) <34 11 7 5 4 3 1>
   46    1 0xffffff7f80d8c000 0x3000     0x3000     com.apple.iokit.IOUSBUserClient (660.4.2) <34 7 5 4 3 1>
   47    0 0xffffff7f80ed4000 0x18000    0x18000    com.apple.driver.AppleUSBHub (683.4.0) <34 5 4 3 1>
   55    1 0xffffff7f80e6f000 0x7000     0x7000     com.apple.driver.AppleUSBComposite (656.4.1) <34 4 3 1>
   56    0 0xffffff7f80ecd000 0x7000     0x7000     com.apple.driver.AppleUSBMergeNub (650.4.0) <55 34 4 3 1>
   57    3 0xffffff7f80e66000 0x9000     0x9000     com.apple.iokit.IOUSBHIDDriver (660.4.0) <34 29 5 4 3 1>
   58    0 0xffffff7f81ae6000 0x4000     0x4000     com.apple.driver.AppleUSBTCKeyboard (240.2) <57 34 29 7 6 5 4 3 1>
   60    0 0xffffff7f81af1000 0x13000    0x13000    com.apple.driver.AppleUSBMultitouch (240.10) <57 34 29 6 5 4 3 1>
   65    1 0xffffff7f80d78000 0x11000    0x11000    com.apple.iokit.IOUSBMassStorageClass (3.6.0) <64 34 15 5 4 3 1>
   67    0 0xffffff7f81c77000 0x8000     0x8000     com.apple.driver.AppleUSBCardReader (3.4.1) <66 65 64 34 15 5 4 3 1>
   69    0 0xffffff7f81aec000 0x3000     0x3000     com.apple.driver.AppleUSBTCButtons (240.2) <57 34 29 7 6 5 4 3 1>
   96    1 0xffffff7f81299000 0x1d000    0x1d000    com.apple.iokit.IOBluetoothHostControllerUSBTransport (4.2.7f5) <34 11 10 7 5 4 3 1>
   97    0 0xffffff7f812b6000 0x7000     0x7000     com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport (4.2.7f5) <96 34 11 10 7 5 4 3>
  125    0 0xffffff7f82391000 0x8000     0x8000     org.virtualbox.kext.VBoxUSB (4.3.26) <124 46 34 7 5 4 3 1>

yaroslavvb, 2015-08-20:
I'm having the same problem. Horndis spams >500 message per second in my Console and makes it spinball. Also, I suspect it causes my computer to crash, I've had 2 crashes in the last hour, and in both cases "HoRNDIS: outputPacket: waiting for buffer..." was the last message in my /var/log/system.log and /var/log/debug.log

I've started experiencing this problem after upgrading my MacOS to 10.10.4 from Mavericks and upgrading my Nexus 6 to 5.1 from 5.0

cpboyd, 2016-02-05:
I'm getting the repeated
```
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: waiting for buffer...
HoRNDIS: outputPacket: timed out waiting for buffer
```

On Yosemite 10.10.5 with both rel7 and rel8pre1

cpboyd, 2016-02-05:
I've started a ping as the connection starts.

I got ~1.5ms ping to the device for the first 3 times.
The 4th (and final) ping took 199.739ms

Papipo, 2016-11-26:
I don't know if this is my problem, but I have the same logs.

HoRNDIS and tethering worked flawlessly the first time I installed the drivers, but never again. In my case I get a self-assigned IP in the 169.X range, which won't work.

moof2k, 2017-01-23:
I'm seeing this as well. Worked once, then never again. Spams "kernel[0]: HoRNDIS: outputPacket: timed out waiting for buffer" to the console for a few seconds, then nothing.

    Darwin .. 15.6.0 Darwin Kernel Version 15.6.0: Wed Nov  2 20:30:56 PDT 2016; root:xnu-3248.60.11.1.2~2/RELEASE_X86_64 x86_64

Tried reloading the driver, no luck

    $ sudo kextunload -b com.joshuawise.kexts.HoRNDIS
    $ sudo kextload -b com.joshuawise.kexts.HoRNDIS