jwise/HoRNDIS #45: Not recognizing connection on OS X 10.10.3

snowmaker, 2015-09-07:
Installed successfully (according to wizard), but when I connect my Android device and enable USB tethering, the mac simply doesn't do anything.  No new connection appears in Network / connections.  

I'm using  OS X 10.10.3  (macbook air) and a Samsung S6.  Any suggestions on how to debug?

Anonymouse3333, 2015-09-29:
@snowmaker @jclif Has your challenge been overcome? I encountered the same situation. Yosemite 10.10.3 required a restart for my MBA '15. After that, I tried an older Samsung Android phone, which worked instantly. So I reconnected the Alcatel phone again, this time realizing I had selected "Linux/Mac OS (EEM)" standard under the USB tethering type option of Tethering & portable hotspot settings. What I did was change over to the "Windows (RNDIS)" standard, and presto! Good luck. Many "thank yous" to @jwise J. Wise for this great driver.
[image]
[image]

snowmaker, 2015-09-29:
Darn, I don't have the "USB Tethering type" option that @Anonymouse3333 is talking about on my phone (Samsung S6).  It just isn't there in settings the way it is for him.  So, not a solution for this phone.

rodneyrod, 2015-10-16:
Is there any way to solve this issue?