jwise/HoRNDIS #58: HoRNDIS needs a new maintainer

jwise, 2016-10-16:
After a three year tenure of maintaining HoRNDIS, it's become clear that I no longer have the time or energy to give HoRNDIS the attention that it deserves -- after all, I only managed to squeeze out a 10.11 release some months after 10.**12** was released.  The burden of doing user support and responding to people for whom the driver does not work with technical advice (or, at the very least, with a modicum of compassion...) has put me down the road to open-source burnout, and I need to take a break.

So, I'm looking for someone who is willing to take over the torch of maintainership for HoRNDIS.  Do you know someone who would like an opportunity to use their skills to contribute to the community?  Are you that person?  Please feel free to send me an e-mail, or comment here.

eldavido, 2017-08-21:
I might be interested in this. Can we chat either via email or on the phone?

I mostly do userspace C# work these days, but I wouldn't mind picking up some lower-level work. I did some Linux kernel dev back in college (computer eng) and would like to dip my toe back into it.

jwise, 2017-08-22:
Hi @eldavido.  Glad to hear from you!  Feel free to send me some mail and we'll talk some.

eldavido, 2017-08-22:
I see you live in Mountain View. I'm happy to just get in the car and have a face-to-face sometime as I'm up in Oakland. I'm actually getting married on Saturday but perhaps we could chat in about 3 weeks? I'll be back in the US on September 12th, let's plan on something that week.

asiderop, 2018-03-09:
@jwise, Still looking for a new maintainer?

jwise, 2018-03-10:
@asiderop Yup -- I haven't touched it in a long time (and, in fact, I haven't even upgraded my Mac to High Sierra!).

radimidia, 2018-04-04:
Friends on which version of OSX has HoRNDIS stopped? I installed it on OSX 10.12 (Sierra) and it did not work! Hugs

vishal733, 2018-04-04:
And for me. I installed it on OSX 10.13.3 (High Sierra). Works like a
breeze. Love it!

radimidia, 2018-04-04:
Friend vishal733 which cell phone did you use? I'm using the Galaxy S7 and Sierra. Actually installed everything right, but does not appear in the options of Network of the Mac.

jwise, 2018-04-04:
(removed off-topic posts)

@radimidia Please do not repeatedly cross post problems across multiple issue: post in only one issue to track a single problem.  Thanks.

vishal733, 2018-04-04:
I'm using the OnePlus5, with High Sierra on Macbook Pro (early 2015 model).

All I had to do was run these two commands, and it just worked:

brew cask install horndis
sudo kextload /Library/Extensions/HoRNDIS.kext

mikhailai, 2018-06-18:
Hi Joshua,

I would like to take over the project. I've been hacking around, and migrated the code to the new IOUSBHostInterface/IOUSBHostDevice interfaces provided my MacOS El Capitan and later. It works fairly well on my Sierra Mac (testing on High Sierra and beta Mojave is still a TODO). I'd appreciate the opportunity to put it where other people can find and use.

On a side note, are people still interested/care about having this tethering driver nowdays?

mikhailai, 2018-06-18:
If you're interested to proceed, it would be nice for us to get on call some time.

mikhailai, 2018-06-22:
Hi @jwise, I'm interested in taking over the maintenance of HoRNDIS (see my posts above).

mikhailai, 2018-06-23:
> Hi @mikhailai <https://github.com/mikhailai> -- sorry I forgot to respond
> earlier, and thank you so much for reminding me! Send me some e-mail, and
> we can set up a time to talk or chat.

Hi Joshua,

Thanks for getting back. Here is my email - miakhiae@gmail.com

I'm currently visiting Spain - till June 28-th, so it may be tricky to find
time to chat till then. I would actually like to get in touch sooner, so we
could try Tuesday (26-th) around 1pm PST - let me know if that would work
for you (I'm assuming you're in California).

After 28-th, I'll be available mornings till around 12pm PST, or late at
night: after 9.30pm PST. Please let me known when it's convenient for you.

 - Mikhail

jwise, 2018-07-09:
@mikhailai has graciously volunteered to maintain HoRNDIS going forward, and has put together the latest release -- 9.0.  Thanks so much, Mikhail!  I'll leave this bug open until I manage to transition the web site to GitHub Pages.

radimidia, 2018-07-09:
Great initiative! Congratulations! We will be your beta tester! As soon as it is available we will evaluate! hug

Em 9 de jul de 2018, à(s) 18:13, Joshua Wise <notifications@github.com> escreveu:

> has graciously volunteered to maintain HoRNDIS going forward, and has put together the latest release -- 9.0. Thanks so much, Mikhail! I'll leave this bug open until I manage to transition the web site to GitHub Pages.
>

CoVoCre, 2019-02-07:
Thank you so much for this awesome driver ! It works like a charm, and i can assure you I really care about this driver @mikhailai 😉I love android and macOs, and there are a lot of community devs doing the heavy lifting of making the little adjustments that in the end make the products way better.
I'd like to donate one or two dollars, and maybe other would like as well, not knowing enough about code to help with the dev work 😉 Have you guys though about an open collective or even just a paypal link for donations ? 😊

mikhailai, 2019-02-07:
@CoVoCre, thanks for your kind words. I don't know about Joshua (maybe you can ask him directly), but I don't care much about donations - so don't feel like setting up the link ;). The best support y'all can provide is help in debugging the driver whenever issues arise (and hopefully that would not happen ;)).

jwise, 2019-02-07:
Yep, my position had always been that people who wish to donate would make me very happy indeed by sending some money to a cause they cared about :-) I particularly recommend [Second Harvest Food Bank](https://www.shfb.org), which serves my local community.

MuntashirAkon, 2020-10-19:
It appears that we need a new maintainer again.