jwise/HoRNDIS #59: Sierra Support?

oxoocoffee, 2016-10-19:
I want to report that installing driver under sierra works. It shows in System Preferences -> Network as 
"Multifunction Composite Gadget" but when I try to share connection to my Udoo Neo Full board it does not work. My sharing is Wifi to "Multifunction Composite Gadget".

Here is output from Console filtered on NDIS with sharing disabled

default	14:11:11.148987 -0500	kernel	HoRNDIS: rndisSetPacketFilter: SET not successful?
default	14:11:11.149108 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsTerminated
default	14:11:25.557729 -0500	kernel	HoRNDIS: init: HoRNDIS tethering driver for Mac OS X, by Joshua Wise (rel8 final)
default	14:11:25.557748 -0500	kernel	HoRNDIS: probe: probe: came in with a score of 60000
default	14:11:25.557904 -0500	kernel	HoRNDIS: message: unknown message type e000401f
default	14:11:25.557944 -0500	kernel	HoRNDIS: message: unknown message type e0000230
default	14:11:25.558182 -0500	kernel	HoRNDIS: probe: probe: looks like we're good (2/2/255)
default	14:11:25.560639 -0500	kernel	HoRNDIS: message: unknown message type e0004022
default	14:11:25.590748 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsAttemptingOpen
default	14:11:25.591291 -0500	kernel	HoRNDIS: openInterfaces: data interface: okay, I got one, and it was a 0x0a/0x00/0x00
default	14:11:25.591299 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsAttemptingOpen
default	14:11:25.592237 -0500	kernel	HoRNDIS: rndisInit: their MTU 1486
default	14:11:25.592580 -0500	kernel	HoRNDIS: setMaxTransferUnit: Excuse me, but I said you could have an MTU of 1486, and you just tried to set an MTU of 1500.  Good try, buddy.
default	14:11:25.592596 -0500	kernel	HoRNDIS: init: starting up with MTU 1486

One thing is that I can ssh to my Udoo Neo board from my mac just fine. As soon sharing is enabled I can't ssh and my en9 interface on my Mac is never activated.

Here is output from Mac Console app after:
1. sharing was enabled
2. On Udoo board en9 was set to dhcp on boot

default	14:29:18.285758 -0500	kernel	HoRNDIS: rndisCommand: unsupported: RNDIS_MSG_INDICATE
default	14:29:42.779514 -0500	kernel	HoRNDIS: enable: setting up the pipes failed
default	14:29:43.881392 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsTerminated
default	14:29:56.845687 -0500	kernel	HoRNDIS: init: HoRNDIS tethering driver for Mac OS X, by Joshua Wise (rel8 final)
default	14:29:56.845706 -0500	kernel	HoRNDIS: probe: probe: came in with a score of 60000
default	14:29:56.845971 -0500	kernel	HoRNDIS: message: unknown message type e000401f
default	14:29:56.846010 -0500	kernel	HoRNDIS: message: unknown message type e0000230
default	14:29:56.846140 -0500	kernel	HoRNDIS: probe: probe: looks like we're good (2/2/255)
default	14:29:56.849304 -0500	kernel	HoRNDIS: message: unknown message type e0004022
default	14:29:56.880039 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsAttemptingOpen
default	14:29:56.880438 -0500	kernel	HoRNDIS: openInterfaces: data interface: okay, I got one, and it was a 0x0a/0x00/0x00
default	14:29:56.880447 -0500	kernel	HoRNDIS: message: kIOMessageServiceIsAttemptingOpen
default	14:29:56.881424 -0500	kernel	HoRNDIS: rndisInit: their MTU 1486
default	14:29:56.881744 -0500	kernel	HoRNDIS: setMaxTransferUnit: Excuse me, but I said you could have an MTU of 1486, and you just tried to set an MTU of 1500.  Good try, buddy.
default	14:29:56.881764 -0500	kernel	HoRNDIS: init: starting up with MTU 1486
default	14:30:14.457844 -0500	kernel	HoRNDIS: rndisCommand: unsupported: RNDIS_MSG_INDICATE
default	14:30:14.471638 -0500	kernel	HoRNDIS: receivePacket: non-PACKET over data channel? (msg_type 00000000)

ritikrishu, 2016-12-13:
same here, nothing changed in System Preferences -> Network, after installing latest binary( HoRNDIS-rel8.pkg). 
Mobile- Samsung S7 edge
OS- Sierra

631068264, 2016-12-22:
[image]
**Support** !!!!  Mobile —— Huawei
You should 
 [uninstall HoRNDIS](https://github.com/jwise/HoRNDIS#uninstallation),
install again, 
 put off the usb , 
connect the usb to your Mac again
set the mobile share the network with PC
DONE !!

Windywind, 2017-01-02:
Same, nothing turns up in _System Preferences -> Network_

Mobile: LG Nexus 5 (D820)
Operating System: macOS Sierra
MacBook Model: Macbook (Retina, 12-inch, Early 2016)

Windywind, 2017-01-02:
Well...After some times of uninstall and reinstall, and rebooting, it works...

Firstly, I saw a _Nexus 5_ in _System Preferences -> Network_, (but I have already uninstalled this HoRNDIS) then I tried to install again. And plug on my mobile, it just works. (this post is sent via tethering network.)

But I cannot tell how & why :P

GLedoussal, 2017-01-11:
Could'nt make it work until I installed ADB. It seems to be working fine now.

NFMynster, 2017-01-21:
I've got it working just by installing and rebooting. Works outta the box for me on 10.12.2 as well.

Manouchehri, 2017-02-12:
HoRNDIS works on 10.12.4 for me.

```bash
brew cask install horndis
sudo kextload /Library/Extensions/HoRNDIS.kext
```

sugitach, 2017-03-05:
I got connection after install ADB, too.
1. install HoRNDIS -> not works
2. reboot mac -> not works
3. install ADB (Android Studio) -> not works
4. reboot mac, and reconnect USB -> It works !!

I'm using in macOS 10.12.3 sierra, motoZ android 7.0.

camslice, 2017-04-11:
Worked after a reboot on 10.12.4

(I already had ADB installed so can't verify if that was part of the problem)

cxcorp, 2017-08-19:
Works on 10.12.6 without adb. Probably had to reboot, but `sudo kextload /Library/Extensions/HoRNDIS.kext` worked.

ecbrodie, 2017-12-02:
I just ran into these instructions and they work AWESOME for me (10.12.6).

It seems like it has been useful for multiple people, so I'm going to PR in a README change to mention these steps.