jwise/HoRNDIS #60: invalid kext signature - does not load on 10.11.5

tmcsys, 2016-10-22:
HoRNDIS-rel8.pkg  OS X 10.11.5 (15F34)

Might this have something to do with "enhanced" security in El Capitan.

<code>
10/22/16 7:03:56.129 PM sudo[32617]:      tom : TTY=ttys000 ; PWD=/Users/tom/Downloads ; USER=root ; COMMAND=/sbin/kextload /System/Library/Extensions/HoRNDIS.kext
10/22/16 7:03:56.461 PM com.apple.kextd[51]: Untrusted kexts are not allowed
10/22/16 7:03:56.461 PM com.apple.kextd[51]: ERROR: invalid signature for com.joshuawise.kexts.HoRNDIS, will not load
</code>

jwise, 2016-10-22:
`/System/Library/Extensions` is where unsigned extensions go; signed versions go in `/Library/Extensions`, and on 10.10+ you should kextload that version instead.

closed 2016-10-22 by jwise

tmcsys, 2016-10-22:
I did not restart my system after installing the Rel8 package. I had some stale network connections from previous versions which I deleted and did not see a new network connection until I manually loaded the kext from the correct location:
<code>
sudo kextload /Library/Extensions/HoRNDIS.kext 
</code>
The new connection came up "Connected", and I am posting this reply through the HoRNDIS USB connection.
Truly A thing of beauty!
Thank you for the very fast response.