jwise/HoRNDIS #66: Frequent kernel panic on Sierra 12.12.4

junaedahmed, 2017-04-27:
Kernel panic occurs on Sierra 12.12.4 sometime if phone cable is disconnected suddenly during connection.

its very frustating 

Pls see into it

MatchesMalonee, 2017-05-23:
I've been facing this issue too.

MatchesMalonee, 2017-05-23:
Have you tried reinstalling the kext?

florianheller, 2018-08-30:
Hi have issues with kernel panics on high sierra (10.13.6). I'm using the driver to connect beaglebones to my mac, and everytime I unplug the USB connection or restart the beaglebone, it crashes my mac. 
Here's one of the crash logs:

```
Tue Aug 28 10:03:58 2018

*** Panic Report ***
panic(cpu 2 caller 0xffffff801678776f): Kernel trap at 0xffffff7f972b4f24, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x000070000a54e458, CR3: 0x0000000154059056, CR4: 0x00000000003627e0
RAX: 0xffffff7f972b4f1a, RBX: 0xffffff8056561090, RCX: 0x0000000000000008, RDX: 0xffffff805978f000
RSP: 0xffffff922ba2bba0, RBP: 0xffffff922ba2bba0, RSI: 0xc0ffee0923f1cbcc, RDI: 0xffffff805a774c80
R8:  0x00000000facefeed, R9:  0x0000000000000000, R10: 0xffffff922ba2ba90, R11: 0xffffff7f976ee810
R12: 0xffffff80688a4c80, R13: 0x0000000000000000, R14: 0xffffff804aef6000, R15: 0xffffff805978f000
RFL: 0x0000000000010246, RIP: 0xffffff7f972b4f24, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x000070000a54e458, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff801654d290 : 0xffffff801666c1c6 
0xffffff801654d2e0 : 0xffffff8016795274 
0xffffff801654d320 : 0xffffff8016787544 
0xffffff801654d390 : 0xffffff801661e1e0 
0xffffff801654d3b0 : 0xffffff801666bc3c 
0xffffff801654d4e0 : 0xffffff801666b9fc 
0xffffff801654d540 : 0xffffff801678776f 
0xffffff801654d6b0 : 0xffffff801661e1e0 
0xffffff801654d6d0 : 0xffffff7f972b4f24 
0xffffff922ba2bba0 : 0xffffff7f972ad267 
0xffffff922ba2bbe0 : 0xffffff8016c3a9e2 
0xffffff922ba2bc30 : 0xffffff8016c3a9ce 
0xffffff922ba2bc80 : 0xffffff8016c31fa2 
0xffffff922ba2bcc0 : 0xffffff8016c18453 
0xffffff922ba2bd00 : 0xffffff8016c657a3 
0xffffff922ba2bd30 : 0xffffff8016cc8b9b 
0xffffff922ba2bd90 : 0xffffff8016744107 
0xffffff922ba2bdc0 : 0xffffff80166716e0 
0xffffff922ba2be10 : 0xffffff801664ea3d 
0xffffff922ba2be60 : 0xffffff801666154b 
0xffffff922ba2bef0 : 0xffffff801677171d 
0xffffff922ba2bfa0 : 0xffffff801661e9e6 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[5A99D802-B477-3D83-9F76-304605479B9F]@0xffffff7f972a2000->0xffffff7f972d1fff

BSD process name corresponding to current thread: UserEventAgent

Mac OS version:
17G65

Kernel version:
Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64
Kernel UUID: 1AE5ACFD-3B6F-3D74-AD52-31F1430DBC6F
Kernel slide:     0x0000000016400000
Kernel text base: 0xffffff8016600000
__HIB  text base: 0xffffff8016500000
System model name: MacBookPro12,1 (Mac-E43C1C25D4880AD6)

System uptime in nanoseconds: 413672746354247
last loaded kext at 412524315153030: com.joshuawise.kexts.HoRNDIS	9.2 (addr 0xffffff7f9c285000, size 36864)
last unloaded kext at 61455408784377: com.joshuawise.kexts.HoRNDIS	6 (addr 0xffffff7f9c1a5000, size 28672)
loaded kexts:
com.joshuawise.kexts.HoRNDIS	9.2
com.CalDigit.ThunderboltStationChargingSupport	1.1
com.jmicron.driver.jmPeripheralDevice	2.0.2
com.silabs.driver.CP210xVCPDriver	5.0.5
org.virtualbox.kext.VBoxNetAdp	5.2.6
org.virtualbox.kext.VBoxNetFlt	5.2.6
org.virtualbox.kext.VBoxUSB	5.2.6
org.virtualbox.kext.VBoxDrv	5.2.6
com.apple.filesystems.msdosfs	1.10
com.apple.iokit.SCSITaskUserClient	404.30.2
com.apple.iokit.IOBluetoothUSBDFU	6.0.7f10
[... 182 lines cut ...]
Network Service: iPad, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
PCI Card: ethernet, Ethernet Controller, Thunderbolt@194,0,0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@193,0,0
PCI Card: pci1b21,612, AHCI Controller, Thunderbolt@192,0,0
Thunderbolt Device: Thunderbolt Station 2, CalDigit, Inc., 3, 25.1
Model: MacBookPro12,1, BootROM MBP121.0177.B00, 2 processors, Intel Core i5, 2.9 GHz, 16 GB, SMC 2.28f7
Graphics: Intel Iris Graphics 6100, Intel Iris Graphics 6100, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445424533303445422D45474346202020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445424533303445422D45474346202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.37.31.1a9)
Bluetooth: Version 6.0.7f10, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en5
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: 4-Port USB 3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: Bluetooth USB Host Controller
USB Device: 4-Port USB 2.0 Hub
USB Device: Keyboard Hub
USB Device: USB Optical Mouse
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

mikhailai, 2018-08-30:
Hi Florian. Thanks for reporting - I thought the crashes were fixed in HoRNDIS 9.2. One thing I find suspicious is the line:

last unloaded kext at 61455408784377: com.joshuawise.kexts.HoRNDIS	6 (addr 0xffffff7f9c1a5000, size 28672)

The "com.joshuawise.kexts.HoRNDIS	6" indicates the presence of an older version of HoRNDIS driver (rel 8 or earlier). Can you please check if the following folder is present:
/System/Library/Extensions/HoRNDIS.kext/
And if so, remove it.

florianheller, 2018-08-31:
Hi, thanks for the quick reply.
neither /System/Library/Extensions, nor /Library/StagedExtensions/... contain a HoRNDIS.kext but it seems my problems are gone. My hunch is that I got the old version from an outdated Beaglebone image that was delivered with some hardware DK. Sorry for the confusion.