jwise/HoRNDIS #67: DHCP not work under 10.12.5(Sierra) and Android 7.1.2(Lineage OS)

xShiftx, 2017-05-27:
Network preferences can't establish connection using DHCP service in auto configuration.
When used manual address connection indicator is green but mac os is not connected to internet.
Sometimes establishing connection success and have an internet access but most attempts failure.

Why you using two drivers for S/L/E and /L/E. Please merge kexts in one that can be placed in Clover kext folders instead of installation in system. It better for updates and reinstalling.

xShiftx, 2017-05-27:
So. If established manual IP address and manual set router as android device IP command 

$ ipconfig getpacket en2

shows empty string. Seems connection not works.

xShiftx, 2017-05-27:
Sorry. The problem was solved when I was disable Orbot service on my Android phone. It's very strange but with enabled Orbot service tethering would't work. Thanx for solution ...

xShiftx, 2017-05-27:
Test single kext from /S/L/E in Clover kexts folder. It works fine. Solved.

xShiftx, 2017-06-05:
So. After week of tests I was found that kext can't determine DHCP auto IP if device plugged in Mac OS after system is booted. When I plug Android before loading and set USB modem mode before Mac OS boot it works fine. Otherwise it not works.

xShiftx, 2017-06-06:
So. After disabling Clover entry `DeleteUnused_400000` HoRNDIS.kext stops to unloading and after unpluging or plunging Android phone via USB it works perfect(have Internet connection). Please close ticket.

xShiftx, 2017-06-08:
Steel not working after sleep ):

mikhailai, 2018-07-31:
Is this issue still relevant?

neoncloud, 2021-11-16:
> So. After week of tests I was found that kext can't determine DHCP auto IP if device plugged in Mac OS after system is booted. When I plug Android before loading and set USB modem mode before Mac OS boot it works fine. Otherwise it not works.

Same issue, Android 11 + mac os 10.15.7