jwise/HoRNDIS #69: MHS291L

thorr18, 2017-06-16:
I heard HoRNDIS has been used for this model of Jetpack in the past, but mine doesn't work. 
Is HoRNDIS only supposed to work for Android phones? 
I have two MHS291LVW devices and when I set them to USB tether, I only get a blank disk window. 
I removed an old version I don't remember using, rebooted, and installed with the newest binary. 
As far as I can tell it's installed. I could mail a Jetpack for use by developer if that would help.
### OS version
> BsMBP:~ bem$ sw_vers
> ProductName:	Mac OS X
> ProductVersion:	10.12.5
> BuildVersion:	16F73
> BsMBP:~ bem$ 
### Files are in place:
> BsMBP:~ bem$ sudo find -x / -name "HoRNDIS"
> Password:
> /Library/Extensions/HoRNDIS.kext/Contents/MacOS/HoRNDIS
> /System/Library/Extensions/HoRNDIS.kext/Contents/MacOS/HoRNDIS
> /Users/bem/.Trash/HoRNDIS.kext/Contents/MacOS/HoRNDIS
> /Users/bem/.Trash/HoRNDIS.kext 3.29.18 PM/Contents/MacOS/HoRNDIS
> BsMBP:~ bem$ 

No new network interface appears when connecting it. This is unlike what I see when I connect an Android phone, but this phone seems to work with or without HoRNDIS. 
### Network devices after plugging it in: 
> BsMBP:~ bem$ networksetup -listallhardwareports
> 
> Hardware Port: Nexus 6
> Device: en5
> Ethernet Address: 02:68:14:09:7f:07
> 
> Hardware Port: Wi-Fi
> Device: en0
> Ethernet Address: 3c:15:c2:c9:f2:da
> 
> Hardware Port: Bluetooth PAN
> Device: en3
> Ethernet Address: 3c:15:c2:c9:f2:db
> 
> Hardware Port: Thunderbolt 1
> Device: en1
> Ethernet Address: 72:00:03:5a:1c:70
> 
> Hardware Port: Thunderbolt 2
> Device: en2
> Ethernet Address: 72:00:03:5a:1c:71
> 
> Hardware Port: Thunderbolt Bridge
> Device: bridge0
> Ethernet Address: 72:00:03:5a:1c:70
> 
### Screenshot: 
[image]
Info appreciated.

mikhailai, 2018-07-31:
Hi @ thorr18. Are you still interested in resolving this issue?

thorr18, 2018-07-31:
Thanks for checking but I'm not sure that I still have time for duplicating the problem or testing a solution right now. Also, I'm on High Sierra now and there may need to be fixes for that first.

mikhailai, 2018-07-31:
Sounds good. Does not make sense to spend time on something you don't need right now.
I'll close the issue, and if you need it in the future, you can open the new issue or re-open this one.

closed 2018-07-31 by mikhailai