jwise/HoRNDIS #70: What is the license?

ryandesign, 2017-07-28:
Please clarify the license...

The homepage says:

> The source is available on GitHub; it is licensed under the GNU General Purpose License, version 3

and the COPYING.rtf file is GPL 3. (Note the P stands for Public, not Purpose.) However the files HoRNDIS.cpp and HoRNDIS.h say:

> This program is free software; you can redistribute it and/or modify
> it under the terms of the GNU General Public License as published by
> the Free Software Foundation; either version 2 of the License, or
> (at your option) any later version."

If the source is really licensed under GPL 2 or later, that's what the homepage should say, and the COPYING file should reflect version 2 of the license.

mikhailai, 2018-07-31:
Hi. Are you still interested to know the answer? I could clarify it with the original author, but I'm curious why it makes a difference for you.

mikhailai, 2018-08-28:
No response from author, closing.

closed 2018-08-28 by mikhailai