jwise/HoRNDIS #72: Problems with High Sierra (10.13)?

cmroanirgo, 2017-10-03:
Hi,
I've been using rel7 on older versions of OSX 10.8+ for a few years while without any issues. However, when I upgraded to 10.13 USB tethering stopped working, so I upgraded to rel8 (& rebooted). It's *kinda* working, but not well. Here's what I see:

1. in Network Preferences, the device connects and DHCP works and I get a local IP address correctly.
2. At this point I can ping internet locations, but DNS was failing.
3. Workaround: add 8.8.8.8 as the DNS resolver and things seem to be working well.

So, I'm not sure if this is just my machine or not, but thought I'd add this to the mix.

iGlaedr, 2017-10-06:
Hi 
I have a Macbook Pro (15-inch, 2017) and I it doesn't detect my one plus 3t when I enable USB tethering.
Can anyone help me?

Thanks
iGlaedr

iGlaedr, 2017-10-06:
I Solved it my self, don't have your phone plugged in when your installing the Software.
Reboot an then it works.

Quezler, 2017-10-13:
@iGlaedr 's solution worked for me

kaelte, 2017-10-27:
```
Fairphone 2
OS: Fairphone Open 17.10.2
iMac: macOS 10.13 High Sierra
```

First I have tried to built from source:

```
$ git checkout rel8
HEAD is now at 920d799... HoRNDIS: rel8pre2 becomes rel8 (fixes #42)

$ xcodebuild  -project HoRNDIS.xcodeproj/
=== BUILD TARGET HoRNDIS OF PROJECT HoRNDIS WITH THE DEFAULT CONFIGURATION (Release) ===

Check dependencies
Code Signing Error: No certificate matching 'Developer ID Application: Joshua Wise (54GTJ2AU36)' found:  Select a different signing certificate for CODE_SIGN_IDENTITY, a team that matches your selected certificate, or switch to automatic provisioning.
error: There is no SDK with the name or path '/Users/klaus/GIT/HoRNDIS/macosx10.6'

** BUILD FAILED **

The following build commands failed:
	Check dependencies
(1 failure)
```

So I have downloaded the binary package from [http://joshuawise.com/horndis](http://joshuawise.com/horndis). The installation went without a flaw and after rebooting the iMac I can use my phone for USB tethering.

Thanks !

ashk-sonkar, 2017-12-20:
rel8 isn't working for high sierra.It installs but doesn't detect my moto g3. please help

ashk-sonkar, 2017-12-20:
sorry, solved it. It was just a permission issue.

sad-story, 2018-03-20:
ashk-sonkar - Glad you solved the permission issue.

 Care to share with everyone else how you solved it?

Nowaker, 2018-04-02:
I can confirm Horndis works on High Sierra. Here's a quick guide:

1. `brew cask install horndis`
   - You might have to open System Preferences -> Security & Privacy -> General and whitelist the installer to run.
2. Reboot OSX, press and hold CMD+R until Recovery Mode boots, then open Terminal.
3. `csrutil disable`
4. `reboot`
5. `sudo kextload /System/Library/Extensions/HoRNDIS.kext`
   - This caused Android to show up in my System Preferences -> Network list. 
[image]

radimidia, 2018-04-03:
Friends, with this program is it possible to connect Galaxy S7 on Mac via USB cable? Are you version for OSX 10.12 (Sierra)? Thank you

Nowaker, 2018-04-03:
@radimidia Please try and let us know. S7 + High Sierra works for me.

sad-story, 2018-04-03:
so I go this error:

MacBook-Pro:~ username$ sudo kextload /System/Library/Extensions/HoRNDIS.kextsudo kextload /System/Library/Extensions/HoRNDIS.kext
/System/Library/Extensions/HoRNDIS.kextsudo failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).

radimidia, 2018-04-04:
Friends did not work out. I did the Nowaker procedure. I did the process on the terminal with the "csrutil disable" command line. See in the Photos!

[image]
[image]
[image]

radimidia, 2018-04-04:
Dear Sad-Story, should this step with the command line be performed after the process described by Nowaker?Thank you

jaypeek1, 2018-04-18:
Hello,

It did not work for my Samsung Galaxy Grand Prime.

"brew" was not recognized by my Macbook in Github.
[image]

radimidia, 2018-04-18:
What version of android?

jaypeek1, 2018-04-19:
Android version 5.1.1

jaypeek1, 2018-04-19:
> What operating system version runs on your MacBook Pro?
> I tested on various brands of cell phones (LG, Motorola), but only Samsung
> did not work.
> Missing test on Samsung S8 with Android 8.
> Throughout the day I will pass the list of cell phones. Let's research

[image: --]

Jay Peek
[image: http://]about.me/jaypeek
<http://about.me/jaypeek?promo=email_sig>

[image: --]
(415) 760-3571

radimidia, 2018-04-19:
TEST TETHERING USB FOR MAC (OSX 10.12.6 Sierra):

-MOTO G3 - Android 6.0 ok

-MOTO G2 - Android 7.1 ok

-MOTO G2 1069 - Android 6.0 ok

-MOTO G5 - Android 7: no

-MOTO G5 Plus - Android 7: ok

-MOTO E 4 Plus - Android 7.1.1: ok

-LG K10 / M250 - Android  7.0: ok
-SAMSUNG ON7 - Android 6.0.1 model SM-G600FY: no
-SAMSUNG S7 - Android 7.0: no

jaypeek1, 2018-04-21:
Radimidia,

Ok. Will update.

Ty.

Sincerely,

JWP

Sivajakn, 2018-04-23:
hi, I am a new Mac user and I followed Nowaker and sad-story's steps and my android usb tethering was successful. but the next day when I was trying to tether it was not working. so I did the whole procedure again and it worked for a day. the next day it doesn't work. help me pls. do I have to to the whole procedure everyday :( ?

radimidia, 2018-04-23:
Please answer the following questions:

-What is the make and model of your smartphone?
-What version of your Android?
-What version of your MAC?

Thank you!

Sivajakn, 2018-04-23:
My phone is redmi5a running on android 7 nougat and am having MacBook air
with os high Sierra 10.13.2

Thank you

> hi, I am a new Mac user and I followed Nowaker and sad-story's steps and
> my android usb tethering was successful. but the next day when I was
trying
> to tether it was not working. so I did the whole procedure again and it
> worked for a day. the next day it doesn't work. help me pls. do I have to
> to the whole procedure everyday :( ?
>
> <https://github.com/jwise/HoRNDIS/issues/72#issuecomment-383514355>, or
mute
> the thread
rSzMS33pY9Jo9WeJks5trZ_8gaJpZM4Prly5>

—
<https://github.com/jwise/HoRNDIS/issues/72#issuecomment-383533304>, or mute
the thread

michelledoyle1, 2018-04-25:
Followed the above tutorials to no avail. My mac is a late 2013 macbook pro running High Sierra 10.13.4 and my phone is a samsung s7.

radimidia, 2018-04-25:
Funcionou? Meu Macbook Pro 2015 rodando Sierra não funcionou com S7 Android
7!

radimidia, 2018-04-25:
It worked? My Macbook Pro 2015 running Sierra did not work with S7 Android
7!

2018-04-25 9:06 GMT-03:00 Radimídia <radimidia@gmail.com>:

timblaktu, 2018-04-26:
Latest HoRNDIS stopped working with recent updates on Macbook Air OSX 10.13.4 (17E202) with Galaxy S7 G930F, Android 7.0 kernel 3.18.14-12365438, Android security patch level March 1 2018.

These devices were working together a couple months ago, definitely before system updates were applied to both via standard Apple App Store/Samsung OTA os updates.

I see the "SAMSUNG Android" interface in Network Preferences on the Mac, but it never becomes active when I connect the S7. Similarly, on the S7's tethering settings page, "USB tethering" remains grayed out even when I connect. 

I have tried multiple cables, and the same phone tethers fine to a Dell XPS 9550 with same cables.

uwehale, 2018-05-13:
Nowaker steps worked with Motorola Moto phone.  
The 
`sudo kextload /System/Library/Extensions/HoRNDIS.kext`
command has to be run each time.

Is there a way to run this without disabling csrutil?

16110376, 2018-06-13:
Hi, High Sierra can't detect USB Tethering, my Phone Asus Zenfone 3 Android 8.0

maricn, 2018-06-23:
I've used @Nowaker's suggestion, except when I did `sudo kextload /System/Library/Extensions/HoRNDIS.kext` it reported an error saying that it's already loaded.

And my phone didn't come up still.

So, if you're getting the same error, you should first do
• `sudo kextunload /System/Library/Extensions/HoRNDIS.kext`
And then
• `sudo kextload /System/Library/Extensions/HoRNDIS.kext`

After doing that, my phone appeared in Network preferences as Ethernet 2.

Sony Xperia XZ working on High Sierra (10.13.5) hackintosh.

swisspra, 2018-07-27:
Thanks to @Nowaker, It now works for me :)
Here is the suggestion:
My Devices: MBP Mid-2014 10.13.6 High Sierra| Xiaomi Redmi5

-  brew cask install horndis
        - You might have to open System Preferences -> Security & Privacy -> General and whitelist the 
           installer to run.
- Reboot OSX, press and hold CMD+R until Recovery Mode boots, then open Terminal.

          "csrutil disable"

          "reboot"

          "sudo kextload /System/Library/Extensions/HoRNDIS.kext"
*********** I couldn't ran this command because horndis.kext is in another location************
          I have to do this instead

          "sudo kextload /Library/Extensions/HoRNDIS.kext"

After you phone shows up in the setting you can go back and reenable csrutil

mikhailai, 2018-07-27:
Hi, y'all. I'm a new HoRNDIS maintainer, and have done a lot of code changes, that have been released as HoRNDIS 9.0 and 9.1.

This "csrutil" magic is totally unnecessary now. All you need to install HoRNDIS is to download the latest release from github page and run the package installer normally, or just use "brew cask install horndis".
IMPORTANT: if it pops up the message saying that kernel extensions is blocked, please do follow the prompt to go to the security settings and press "Allow".

That's it! All the later steps with "CMD+R" are NOT needed. If things don't work, just do a normal reboot. If that does not work, please let me know: this needs to be investigated.

swisspra, 2018-07-27:
@mikhailai 
For me, the normal installation didn't work [both installer and homebrew]
After I read your post, I tried to reinstall it without disable csrutil. and yes it is work!!
However, both methods put the HoRNDIS.kext to /Library/Extensions/HoRNDIS.kext but not
/System/Library/Extensions/HoRNDIS.kext. so that my mac doesn't load this extension.
I still have to run this command "sudo kextload /Library/Extensions/HoRNDIS.kext"  every time I connected my phone to make the extension work.

The solution for me is to copy the HoRNDIS.kext to /System/Library/Extensions/ manually.
Now it is work flawlessly.

radimidia, 2018-07-27:
Master @mikhailai, works with samsung galaxy s7? Thank you!

Em 27 de jul de 2018, à(s) 05:15, swisspra <notifications@github.com> escreveu:

> @mikhailai

mikhailai, 2018-07-27:
Hi @swisspra. I'm very surprised that you had to move HoRNDIS to "/System/Library/Extensions". As far as I know, the "/Library/Extensions" is the preferred location for installing the 3-rd party drivers, and MacOS should load the drivers from both locations. My understanding was that "/System/Library/Extensions" was needed for the very old versions of MacOS, and that's why I removed installation into it in "9.0+".

Are you sure that rebooting the Mac does not allow the driver to load from "/Library/Extensions"? On my systems, "/Library/Extensions" works flawlessly, and I'm not aware of anyone else having this problem with 9.0+ drivers. BTW, one thing to try would be to run "sudo touch /Library/Extensions".

808b, 2018-09-04:
HI, i have galaxy note 4 with android 4 and my mac is running on 10.13.6. It was working on el captain but i'm not sure if it was HoRNDIS. I try several times to instal the pkg (HoRNDIS-9.2) and to run some texts on terminal but nothing worked after rebooting. any help with easy steps?

mikhailai, 2018-09-04:
Hi @808b, please check if this applies to you:
https://github.com/jwise/HoRNDIS/issues/78
There may be a problem due to Samsung software.

Also, you can try the command and see if tethering works after running that:
sudo kextload /Library/Extensions/HoRNDIS.kext

808b, 2018-09-04:
thanks mikhailai. this is what i get after running the command:
/Library/Extensions/HoRNDIS.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).

808b, 2018-09-04:
I was asked but in System Preferences -> Security and Privacy it's not possible to allow the app. i have the sentence " the software loading from joshua wise was blocked. 

Kext rejected due to system policy: <OSKext 0x7fcc70005140 [0x7fffadad6af0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/HoRNDIS.kext/", ID = "com.joshuawise.kexts.HoRNDIS" }
Kext rejected due to system policy: <OSKext 0x7fcc70005140 [0x7fffadad6af0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/HoRNDIS.kext/", ID = "com.joshuawise.kexts.HoRNDIS" }
Diagnostics for /Library/Extensions/HoRNDIS.kext:

808b, 2018-09-04:
this is what i get with the terminal:
Kext rejected due to system policy: <OSKext 0x7fcc70005140 [0x7fffadad6af0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/HoRNDIS.kext/", ID = "com.joshuawise.kexts.HoRNDIS" }
Kext rejected due to system policy: <OSKext 0x7fcc70005140 [0x7fffadad6af0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/HoRNDIS.kext/", ID = "com.joshuawise.kexts.HoRNDIS" }
Diagnostics for /Library/Extensions/HoRNDIS.kext:

mikhailai, 2018-09-04:
@808b, basically you need to find a way to approve the HoRNDIS kext - the MacOS HighSierra will not let you use it without this approval. The MacOS Sierra and earlier were not that strict: that's why it worked on El Capitan.

So, again, please make sure in the "Security and Privacy" tab you have a radio button set to allow apps downloaded from "App Store and identified developers", and then try clicking "Allow" button for HoRNDIS (you may need to try an install step before that").

There is also another "hacky" way to get that approval, but it would be nice to try a "good" way first.

mikhailai, 2018-09-04:
> i have the sentence " the software loading from joshua wise was blocked.
That's why it is not working.

iosonosempreio, 2018-09-16:
Hello,
am I the only one who doesn't see the extension in the folder `/System/Library/Extensions`?
Is not working btw.

iosonosempreio, 2018-09-18:
Hello!
I'm on a MacBook Pro retina mid 2012 with HighSierra 10.13.6 and I try to connect a LG Nexus 5.
I did allowed it in the security settings.

mikhailai, 2018-11-11:
@dudymas, thanks for posting this comment: this info will be helpful to people. I copied it into a separate issue, to make it easier for the others to find.

@iosonosempreio, is the problem still relevant? We can try to debug it now.

joelpurra, 2018-11-30:
Can confirm that HoRNDIS works well on macOS Mojave 10.14.1 together with (a very outdated) Android CyanogenMod 4.1.2. Writing this comment over the tethered connection -- thanks =)

Only did `brew cask install horndis` and allowed the kext in the computer's security preferences. Had the phone plugged in during installation, and enabled USB tethering in the phone settings. Disconnected the computer from the wifi used to download the software. Didn't have to do anything else -- was tethered and back online right away!

mikhailai, 2018-11-30:
If it works, it should work well. The only thing that may be different with CyanogenMod are USB interface/descriptors, and if HoRNDIS could recognize these, then it should work fine. Mojave is also supported.

You may want to double-check the connection speed, though I'm not expecting to see any issues there.

rmasclef, 2019-06-09:
Hi there, just a quick message to explain a mistake I made while trying to activate USB tethering on my mac using HoRNDIS.

So I installed it using [https://joshuawise.com/horndis#available_versions](the latest installation package here ... and I could not get it to work.

Turns out the problem was that I did not allow `Joshua Wise` app in `System Preferences -> Security & Privacy -> General`

You should see a message like this (with `Joshua Wise` instead of `Fumihiko Takayama`:

[image]

As soon as I allowed this app it worked well.

I hope this can help some of you who have installation issues 👍

kevinhq, 2020-01-01:
Confirmed this is working on Mojave with Samsung M2. But I need to install it through homebrew as specified on this page: https://github.com/jwise/HoRNDIS

Thanks, for such helpful software!

cruzzer, 2020-03-12:
> Hi there, just a quick message to explain a mistake I made while trying to activate USB tethering on my mac using HoRNDIS.
>
> So I installed it using [...]

Thanks @rmasclef. I've sent an email to the developer with these details and hope he adds a note to the main site (https://joshuawise.com/horndis) to reflect this.

It would of course be great if this could be achieved within the installer, but as a workaround this is pretty painless.

Whatagit, 2020-04-06:
Hi Ivve tried installing on macOS Sierra 10.12.1 and restarted, also did it without the phone connected but nothing at all happens when I reboot?

davidkagoma, 2020-05-20:
> I can confirm Horndis works with Samsung S7 on High Sierra. Here's a quick guide:
> 
> 1. `brew cask install horndis`
>    
>    * You might have to open System Preferences -> Security & Privacy -> General and whitelist the installer to run.
> 2. Reboot OSX, press and hold CMD+R until Recovery Mode boots, then open Terminal.
> 3. `csrutil disable`
> 4. `reboot`
> 5. `sudo kextload /System/Library/Extensions/HoRNDIS.kext`
>    
>    * This caused Android to show up in my System Preferences -> Network list.
>      [image]

This is working. But soon as I enable csrutil, then I'm unable to use usb tethering. 
@mikhailai , any idea on how I can deal with this ? I've used ver 9.2 from github. Im running MacOS 10.15.4 , Samsung A71.

Thanks.

Nowaker, 2020-05-20:
You need to keep it at `csrutil disable`. Enabling it back will render the kext inoperable.

davidkagoma, 2020-05-20:
@Nowaker thanks. Any potential security issues that I should be wary off, with csrutil disabled ?