jwise/HoRNDIS #73: Building on Sierra Fails

4ydx, 2017-10-30:
I get the following error:

'IOKit/usb/IOUSBBus.h' file not found

Are the deploy packages not being built on a machine running Sierra or is there something I need to change/install?

tmagik, 2018-05-03:
The API was removed..

https://developer.apple.com/library/content/releasenotes/General/APIDiffsMacOSX10_11/Objective-C/Kernel.html

Removed IOUSBBus
Removed IOUSBBus::getMetaClass()

seems like it would have to be re-implemented using some new api??

d235j, 2018-05-03:
yeah, Apple rewrote the USB stack in 10.11 (and left in place a compatibility layer for which headers are not available, but for which the old headers would work).

If you want to use the new API, look at the additions on that page.

tmagik, 2018-05-03:
Is there any sort of implementation/migration guide for the USB stack rewrite? … If I had the source code I wouldn’t need to ask stupid questions on how to use equipment I paid a lot of money for :-/

d235j, 2018-05-03:
There's documentation for the new API (IOUSBHostDevice, IOUSBHostInterface, and the rest), but I don't know of any migration guide, and source code for this stack is not available.

It might be worth putting in a radar ticket, but I don't know if that will lead to any responses, unfortunately.

jwise, 2018-05-03:
Yep. I filed a developer technical support request for it two or so years ago, and the answer was "we don't have any sample code for that, good luck, sucker". But if you have a patch set that actually brings us up to the latest APIs, that would be a good thing, I think.

mikhailai, 2018-07-31:
The latest version of "master" builds fine on Sierra or High Sierra. For all who are interested:
* The new API is in the following headers:

#include <IOKit/usb/IOUSBHostDevice.h>
#include <IOKit/usb/IOUSBHostInterface.h>
#include <IOKit/usb/IOUSBHostFamily.h>
#include <IOKit/usb/IOUSBHostPipe.h>

* They are located in:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/IOKit/usb/

* The best source of documentation are the comments in the headers themselves. They also specify "a kind" of a migration guide in the form:

 <code>virtual UInt16 IOUSBDevice::GetbcdUSB();</code><br>
 Replacement: <code>USBToHost16(getDeviceDescriptor()->bcdUSB);</code>

Note, the HoRNDIS itself was updated for the new API, but if you're up to hacking up something else, then have fun!

brandonros, 2019-11-17:
These files work for Mac OS X, but what about trying to import/include + link against IOKit for iPhoneOS?