jwise/HoRNDIS #76: Kernel panic on 10.12

petergus, 2017-12-25:
Experienced a couple of times today after using without any issues for a couple of months.

Anonymous UUID:       4B8A5A9C-D186-E607-3000-BB0A423393FB

Mon Dec 25 15:53:14 2017

*** Panic Report ***
panic(cpu 1 caller 0xffffff801c1fe39d): Kernel trap at 0xffffff801c096c37, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000008, CR3: 0x00000000015bc017, CR4: 0x00000000001627e0
RAX: 0x000000000000060e, RBX: 0x0000000000000000, RCX: 0x0000000000000600, RDX: 0x0000000000000000
RSP: 0xffffff914ba5b9d0, RBP: 0xffffff914ba5b9d0, RSI: 0xffffff811ee40000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000001fd0003, R11: 0x0000000002000001
R12: 0xffffff811ee40000, R13: 0x0000000000000156, R14: 0xffffff811ee40000, R15: 0xffffff8035902800
RFL: 0x0000000000010297, RIP: 0xffffff801c096c37, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000008, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 0

Backtrace (CPU 1), Frame : Return Address
0xffffff914ba5b660 : 0xffffff801c0e953c 
0xffffff914ba5b6e0 : 0xffffff801c1fe39d 
0xffffff914ba5b8c0 : 0xffffff801c09a593 
0xffffff914ba5b8e0 : 0xffffff801c096c37 
0xffffff914ba5b9d0 : 0xffffff7f9f2d8660 
0xffffff914ba5ba30 : 0xffffff7f9cbc1a50 
0xffffff914ba5ba70 : 0xffffff7f9cbc1773 
0xffffff914ba5bae0 : 0xffffff7f9cbc19b2 
0xffffff914ba5bb20 : 0xffffff801c38080a 
0xffffff914ba5bc50 : 0xffffff801c362aea 
0xffffff914ba5bce0 : 0xffffff801c3539a7 
0xffffff914ba5bd50 : 0xffffff801c348f70 
0xffffff914ba5bdd0 : 0xffffff801c33a4fd 
0xffffff914ba5be50 : 0xffffff801c59bb27 
0xffffff914ba5bef0 : 0xffffff801c59b98b 
0xffffff914ba5bf50 : 0xffffff801c6240f5 
0xffffff914ba5bfb0 : 0xffffff801c09ad96 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.2)[6326DB88-5330-3F0C-91F6-D478AB5E7503]@0xffffff7f9cbb0000->0xffffff7f9cbdcfff
         com.joshuawise.kexts.HoRNDIS(6.0)[2FF769BE-1667-3CE2-A8AE-C9D86BAF30D4]@0xffffff7f9f2d6000->0xffffff7f9f2dcfff
            dependency: com.apple.iokit.IONetworkingFamily(3.2)[6326DB88-5330-3F0C-91F6-D478AB5E7503]@0xffffff7f9cbb0000
            dependency: com.apple.iokit.IOUSBFamily(900.4.1)[1B35F2B8-7EB1-3F3A-86DD-35F07996E335]@0xffffff7f9caf8000

BSD process name corresponding to current thread: configd

Mac OS version:
16G29

Kernel version:
Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64
Kernel UUID: D3314D98-5D40-3CD8-98A4-F1DD46C20E03
Kernel slide:     0x000000001be00000
Kernel text base: 0xffffff801c000000
__HIB  text base: 0xffffff801bf00000
System model name: MacBookAir6,2 (Mac-7DF21CB3ED6977E5)

System uptime in nanoseconds: 12849857966163
last loaded kext at 12733523918384: com.apple.iokit.SCSITaskUserClient	394.50.1 (addr 0xffffff7f9f318000, size 36864)
last unloaded kext at 11680654195339: com.apple.driver.usb.AppleUSBHostCompositeDevice	1.1 (addr 0xffffff7f9f25b000, size 28672)
loaded kexts:
com.devguru.driver.SamsungACMData	1.4.31
com.MBB.driver.MBBActivateDriver	6.00.05
com.joshuawise.kexts.HoRNDIS	6
org.virtualbox.kext.VBoxNetAdp	5.1.6
org.virtualbox.kext.VBoxNetFlt	5.1.6
org.virtualbox.kext.VBoxUSB	5.1.6
org.virtualbox.kext.VBoxDrv	5.1.6
ch.tripmode.TripModeNKE	1.0.1
com.radiosilenceapp.nke.PrivateEye	1.1
com.Cycling74.driver.Soundflower	2
com.apple.iokit.SCSITaskUserClient	394.50.1
com.apple.driver.TestDeviceVCOMACMData	1.0.0
com.apple.filesystems.autofs	3.0
[... 110 lines cut ...]
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.50.16
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
com.apple.kec.Libm	1
Model: MacBookAir6,2, BootROM MBA61.0099.B53, 2 processors, Intel Core i7, 1,7 GHz, 8 GB, SMC 2.13f15
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.171.130.1a1)
Bluetooth: Version 5.0.5f1, 3 services, 17 devices, 1 incoming serial ports
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.6

mikhailai, 2018-07-23:
Hi,
If you're still using the HoRNDIS driver, you may want to try the new 9.1 release.

mikhailai, 2018-07-31:
No response from the author. Assuming it's no longer a problem and closing the issue. Debugging this would not be of too much use, because the new 9.1 would work differently from the rel8 that crashed here.

closed 2018-07-31 by mikhailai