jwise/HoRNDIS #79: rel8 killed High Sierra

dicksonrg, 2018-05-30:
I loaded Horndis for the first time (rel8 final) on my 2011 MacBook Pro running High Sierra (10.13.4 I think). 

When I went to restart, the computer only gets to the grey screen. Progress bar will show complete (after an hour), but never any further.

Can't boot in safe mode. Booting in verbose mode shows recurring Horndis activity:

- probe: came in with a score of 60000
- unknown message type e000401f
- unknown message type e0000230
- probe: this composite device is not for us

Can you help me recover my computer?

jwise, 2018-05-30:
Ooo that is super surprising.  I usually try not to support HoRNDIS these days -- it is unmaintained at this point -- but for a broken machine I will try to help out :-)

I don't know what caused this (do you have any USB devices attached?), and I haven't heard of it elsewhere.  But for recovery, probably the best thing to do is to boot into single user mode (hold command-S as the machine boots), and run something like `mv /Library/Extensions/HoRNDIS.kext /tmp` (be very careful when you type that!  do not add spaces anywhere other than where I have them, or else you can end up in quite big trouble!), then run `reboot` . If that doesn't do it, you may have to do some magic involving `kextcache`, but try that first...

dicksonrg, 2018-05-30:
Thank you very much for your help!

Before I saw your response, I reloaded OS 10.13.4. My computer still wouldn't boot normally, but I was able to boot in safe mode, which gave me access to the system/library/extensions folder. I deleted the horndis file, along with another driver that I had loaded a few hours before my computer crashed (a beta driver for OS 10.13 from TP-LINK for their USB wifi adaptor). After clearing those from the trash, my MacBook booted normally in record time :)

Thanks again for your assistance, and for all that you have done to serve the Mac community. I hope you are thriving in what you are doing.

closed 2018-05-30 by dicksonrg