jwise/HoRNDIS #80: unknown message type e000401f / e0000230

edamazzio, 2018-06-03:
Hi.

So I've tried to install HoRNDIS in a fresh install of High Sierra, as well as an old one (different computer) but I keep getting these messages and no interface is created. It used to work on Sierra 10.12

I need HoRNDIS because I don't have any other way to connect to the internet without spending some money buying a PCI card or USB adapter (which I only have two USB ports so I wouldn't like to sacrifice one)

Log is here: 

default	12:26:10.507522 -0600	kernel	HoRNDIS: init: HoRNDIS tethering driver for Mac OS X, by Joshua Wise (rel8 final)
default	12:26:10.507536 -0600	kernel	HoRNDIS: probe: probe: came in with a score of 60000
default	12:26:10.507628 -0600	kernel	HoRNDIS: message: unknown message type e000401f
default	12:26:10.507646 -0600	kernel	HoRNDIS: message: unknown message type e0000230
default	12:26:10.508953 -0600	kernel	HoRNDIS: probe: probe: this composite device is not for us

I've tried sudo kextunload / kextload without positive results. 

Also tried to install without having my phone connected. 

Any help would be greatly appreciated. Thanks

edamazzio, 2018-07-23:
I'm using Oneplus 5. 

Also, I could make it run by copying preferences.plist and NetworkInterfaces.plist from another Mac (also running HighSierra). That computer had Sierra, also ran HoRNDIS, and was upgraded to HighSierra, and HoRNDIS does work on it. 

I actually don't know how that worked. 

Also, I have to constantly unload and load HoRNDIS.kext when I plug or unplug a device, or if USB thetering is disabled on a device, but I think you are already aware of that.

BTW thanks for the new release and the support.

mikhailai, 2018-07-23:
Do these things also happen with 9.1 release? Could you please install the 9.1, reboot the Mac, and try tethering again.
I've changed a lot of code for that release, so was expecting the issues you mentioning to go away. If that's still happening, would you mind attaching the log messages, just like you previously did for rel8.

Thanks!

edamazzio, 2018-07-23:
I haven't installed 9.1 _yet_. I will install it later, as I rely entirely on HoRNDIS right now. 

Cheers!