jwise/HoRNDIS #88: Can't get to working on 10.13.5 and Nokia 7 plus

Riimu, 2018-07-25:
I've used the tethering successfully on previous intallations of os x with different android phones, but now that I'm trying it with my latest setup, I can't seem to get the tethering detected in any way. Nothing appears in the network connections and obviously there is no internet connection. 

My setup:
* Macbook Pro (Retina, 13-inch, Early 2015)
* MacOS High Sierra version 10.13.5
* Nokia 7 plus, Android 8.1

Using the latest driver release, i.e. 9.1. Tied installing via package and homebrew and rebooting the laptop a lot, but to no avail.

Any ideas what to try or how to debug?

Disabling system integrity protection is a no go.

mikhailai, 2018-07-25:
Hi Riikka,
Thanks for reporting the issue. I would need your help in debugging this. First, when you installed the HoRNDIS, it should have asked you to allow the kext under "Security & Privacy", General tab. Have you done this? Also, please double-check that "Allow apps downloaded from:" is set to "App Store and identified developers".

If things are good there, here is the next step. If you have an access to a Linux machine, can you connect the phone to it, enable USB tethering, run "lsusb -v", and send me the result.
If you don't have a Linux machine, you can get "lsusb" from Macports, or I can send you the "lsusb" binary I built on my mac, or instructions how to build it yourself (I wish it was in Homebrew, but Homebrew guys are too slow processing my pull request).

d235j, 2018-07-25:
@mikhailai is it possible to get the needed info on a Mac with `ioreg -l`?

mikhailai, 2018-07-25:
Looks like you can, so maybe we could try that ... AFAIK, the difference is that 'lsusb' gives you the USB descriptors that the device provides, while 'ioreg -l' gives you the MacOS iokit objects that were created based on the descriptors. Therefore, the following information would be missing:
 - Endpoints
 - Interfaces from different configurations, if there is more than one (that is unlikely).
 - Interfaces if the device was not a USB composite device, and no one called "SetConfiguration" on it (this one I'm not sure - don't have any way to check).
Plus, the "ioreg -l" output is very hard to parse, so I'd really prefer to see the "lsusb -v" one.

jwise, 2018-07-25:
The best way to do this used to be USB Prober, which they packaged with OS X, but eventually they kicked it out to the IOUSBFamily Debug Kit ... and now they don't distribute IOUSBFamily Debug Kits for modern OS X.  
http://touch-base.com/downloads/MacProber/USB_Prober_Mac_OS_X_10.7.3.zip should still work, though.

Riimu, 2018-07-26:
> First, when you installed the HoRNDIS, it should have asked you to allow the kext under "Security & Privacy", General tab. Have you done this? Also, please double-check that "Allow apps downloaded from:" is set to "App Store and identified developers".

Yeah, I allowed it through the security settings and apps are allowed from identified developers.

I ran `lsusb -v` gotten from macports (found it in "usbutils" port, fyi) and that `ioreg -l` command. I've attached the outputs from those two.

* [lsusb.txt](https://github.com/jwise/HoRNDIS/files/2231442/lsusb.txt)
* [ioreg.txt](https://github.com/jwise/HoRNDIS/files/2231444/ioreg.txt)

mikhailai, 2018-07-26:
Thanks for the USB logs. I see the problem now: Nokia is using different USB interface specifications that other Android phones. I've fixed it, so I'll send the build you can test soon.

mikhailai, 2018-07-27:
Hi Riikka,
Please try this build and let me know how well that works with your setup. If things are all good, we'll do a 9.2 release based on that.
https://joshuawise.com/downloads/HoRNDIS-9.2d1.pkg

Riimu, 2018-07-27:
I can happily report that I posted this via the tethered connection. So with those changes, the tethering seems to work now!

Thank you.

mikhailai, 2018-07-28:
Thanks for the report. We'll include these changes in the next release. In the mean time, feel free to keep on using the 9.2d1 build: it should be the same quality as the official release.

FranzFelberer, 2018-07-29:
This fixed also my problem with the Nokia 8, thanks.

mikhailai, 2018-08-02:
The HoRNDIS 9.2 was released. Closing.

closed 2018-08-02 by mikhailai