jwise/HoRNDIS #92: HoRNDIS only works on first boot

theronic, 2018-12-06:
The HoRNDIS driver only seems to work on initial boot. When I plug in a Raspberry Pi Zero or Raspberry Pi 3B+, it shows up as a Ethernet gadget that gets an address assigned via DHCP and I can SSH into it. But if I unplug it and plug it back in, the *RNDIS Ethernet gadget remains disconnected.

Running `kextstat | grep DIS` shows:

```
 ~  sudo kextload /Library/Extensions/HoRNDIS.kextss                                                                                               
 ~  kextstat | grep DIS                                                                                                                            
  189    0 0xffffff7f83840000 0x6000     0x6000     com.joshuawise.kexts.HoRNDIS (9.2) 10ED3B68-FBFE-326E-B848-DC8082240F4A <22 15 7 5 4 3 1>
```

I have tried manually unloading and reloading the HoRNDIS driver like so:

```
sudo kextunload /Library/Extensions/HoRNDIS.kext
sudo kextload /Library/Extensions/HoRNDIS.kext
```

But this does not seem to help. Is there a way to confirm that OSX is indeed using the correct driver when plugging in an RPi?

mikhailai, 2018-12-09:
Hi Petrus,

I cannot reproduce the issue. Given that I don't currently have RPi, I was trying to reproduce it on Pocket Beagle board, and everything works fine.
So, I have a couple of questions, in order to proceed:
1. What MacOS version are you using?
2. Does the RPi gets rebooted when you unplug it and plug it back in?

The #2 is important: if the RPi stays powered on, it may be the RPi that is refusing to reconnect, not the HoRNDIS.

theronic, 2018-12-10:
Hi @mikhailai :) Thanks for getting back to me. I'm running Mac OS High Sierra 10.13.6. Yes, I'm powering the RPi Zero via the USB data port, so it is rebooting. I tried uninstalling HoRNDIS and reinstalling.

When I run `system_profiler SPUSBDataType`, I get:

 ```
       RNDIS/Ethernet Gadget:

          Product ID: 0xa4a2
          Vendor ID: 0x0525  (PLX Technology, Inc.)
          Version: 4.14
          Speed: Up to 480 Mb/sec
          Manufacturer: Linux 4.14.79+ with 20980000.usb
          Location ID: 0x14200000 / 13
          Current Available (mA): 500
          Current Required (mA): 2
          Extra Operating Current (mA): 0
```

But when I open Network settings, I see this:

[image]

RPi is configured like so:

/boot/cmdline.txt:

`dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=2e6c8d06-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether
`

/boot/config.txt:

```
# ...all the defaults, plus:
dtoverlay=dwc2
```

theronic, 2018-12-10:
I followed all the instructions in [this post](https://raspberrypi.stackexchange.com/questions/64112/macos-not-discovering-raspberry-pi-zero-as-an-usb-ethernet-device) with no luck.

theronic, 2018-12-10:
I'm not sure HornDIS driver is being used properly. After Mac reboot, `kextstat` shows no entry for | grep DIS, but RPi is connected as Ethernet gadget and assigned an IP address. If I plug RPi out and back in (which power cycles the RPi), the Ethernet gadget never reconnects.

Explicitly calling `sudo kextload /Library/Extensions/HoRNDIS.kextss` does not help, so I'm not sure which driver is being used here.

Note: `/System/Library/Extensions/HoRNDIS.kext` does not exist after running HornDIS package installer and rebooting.

mikhailai, 2018-12-10:
Ok, I'll look more into it later today, but can you please try the following now:
 - Reboot your Mac
 - Plug in (only one) RPi and observe it connecting as expected.
 - Disconnect RPi, and then plug it into a DIFFERENT USB port.
 - Check if it connects like the first time.

mikhailai, 2018-12-11:
In addition to the above, the following steps would help debugging:
- Reboot your Mac
- Run `ioreg -l -r -c IOUSBHostDevice` and capture it into a file.
- Plug in the RPi, wait for it to connect, run `ioreg -l -r -c IOUSBHostDevice` again and capture it into a different file.
- Run `kextstat | grep RNDIS`
- Disconnect the RPi, wait >= 10 seconds, run `ioreg -l -r -c IOUSBHostDevice` again and capture it into a different file.
- Connect the RPi to the SAME port, wait for it to boot up (about the same time as for the first connection), run `ioreg -l -r -c IOUSBHostDevice` again and capture it into a different file.
- Run `kextstat | grep RNDIS`

Attach the resulting files, as well as the results of `kextstat`