jwise/HoRNDIS #93: Slower than windows RNDIS driver?

dsjzn, 2018-12-13:
Is it possible to change the speed?

mikhailai, 2018-12-13:
Please clarify what speed are you talking about, and how the slower speed manifests itself.

dsjzn, 2018-12-13:
First of all, I want to say thanks for your great work! Really helpful for our project! and sorry for my poor question.
 I'm using an RNDIS to RNDIS cable to connect a Windows PC to a Mac(installed HoRndis), and Remote Desktop the windows pc from this Mac, at the beginning things go well, very smoothly, I can even watch 1080p video smoothly, but after some time, it becomes slow,  we will not encounter this problem in when using Windows to Windows, so we think this problem may be caused by Rndis driver. What do you think?

mikhailai, 2018-12-13:
Thanks for providing the details. I'm very surprised to hear about this application of HoRNDIS: never thought of using it to connect 2 computers. The slowdown you've described may or may not be caused by HoRNDIS driver: this is something we may want to investigate. The problem is that I've never observed such slowdown in my testing. However, I only connected a single-board computer to a Mac, not a Windows machine.

First of all, I would not be able to fix or even investigate the issue without your involvement. So, the primary question is do you really want this to be resolved, and how much effort are you willing to spend helping me track the problem? It's going to take some work even to reproduce the problem on my machine. So, how much would you be able to be involved?

Assuming that you want to proceed, I would need even more details:
 - Which exactly the cable are you using (link to Amazon, or website with specs would be helpful)?
 - Is the link USB2 or USB3 - want to know what connection speed we're talking about?
 - How much time it takes for the connection to become slow?

Also, here is what I want you to check. Once it becomes slow, what happens if you close (exit/kill) remote desktop applications (on both sides), then open them again -all this while NOT touching the RNDIS cable. Does it stat as slow as before, or it (temporarily) becomes fast? I want to make sure the slowness is really caused by RNDIS connections, not the applications.

raj100s, 2020-08-07:
I have a created a opencore installing pendrive but I don't have ethernet I use usb tethering to use internet on my windows so how should I add this to the opencore pendrive should I add the .plist and take of snapshot please reply