jwise/HoRNDIS #94: Network not stable, help to figure out this.

diigu, 2018-12-20:
Hi, I'm using macOS High Sierra.
When I use this driver, it seems not stable, broken/offline sometimes.
How can I get some log or other stuff to fix this?

mikhailai, 2018-12-20:
HoRNDIS does some logging to the system log. You can view it via:
 - Console application.
 - Command line:
   log show --predicate process==\"kernel\" --start "$(date -v-3M +'%F %T')"
Please refer to the HoRNDIS README, "OS Logging" in the Debugging and Development notes.

Are you sure the instability is caused by the driver, as opposed to a network connection? Here are a couple of things you could try to narrow it down:

1. Once you are in a place with stable WiFi and GOOD SIGNAL, connect the phone to that WiFi and setup the USB tethering (disable WiFi on MAC). This way, you are using HoRNDIS driver, but have a good reliable network, so if you start experiencing problems, it's likely due to HoRNDIS.

2. If you are tethering using a cellular connection: once you start experiencing problems, run the internet speed test (e.g. http://www.speedtest.net/) on the Mac and on the phone (not simultaneously). If results are similar, it's NOT the problem with HoRNDIS, but rather with cellular connection. If Speed Test on the phone works way better than on the Mac, then it may be the driver's issue.

diigu, 2018-12-21:
I'm not sure about this, my case is, I use a 4g network router(In truth, it's just an android device with 4g network and USB tethering) I do ping server background to see if the network is online. And I see network failure in other programs, such as chrome download and other programs. But meanwhile, ping server does not lose packets or broken. So, what're your suggestions?

mikhailai, 2018-12-21:
Ok, here is one idea. In the "System Preferences" -> "Network", you can see the connection parameters for USB tethering. Find IP for the gateway (called "Router", see attached screenshot) and try pinging it. If the ping works (e.g. your 4g network router decides to respond), wait till you have network problems, and try pinging it again, and see if that works then.

Also, if your device allows installing applications, you can try to install "iperf" or "iperf3", and check the connection between your mac and the device.

jwise, 2018-12-21:
FWIW, on T-Mobile in the US, packets through tethering are routed with different QoS than packets for on-device data.  So I can very often have no problems with on-device data, and have no problems pinging my device from my Mac, but be completely unable to get traffic from my Mac to the Internet -- even if I'm tethering over WiFi.  May be the case for your carrier too.

diigu, 2018-12-22:
I tested another device using tethering, but that one is stable enough.
So, it's should not be HoRNDIS's problem.

closed 2018-12-22 by diigu