jwise/HoRNDIS #97: HoRNDIS stopping to work after a few seconds

Xekresis, 2019-02-15:
Hey dear HoRNDIS team & GitHub-bers,
My phone get's recognized without problems and ethernet works for about 10 seconds, then it stops working and I need to turn USB Tethering off and on again. It's still shown in the preference pane but it just doesn't load anything anymore.

mikhailai, 2019-02-15:
Hi @Xekresis, thanks for reporting the issue. It sounds like a bug in HoRNDIS and definitely needs investigation. In order to do this, I'd need a bit more information:
 - What MacOS are you using?
 - What Android Phone and Android version are you using?
I presume you're using the latest version of HoRNDIS.

Also, I won't be able to look into the issue right away, but I'll try to find some time within the next week or two as long as I receive the above information.

jvalteren, 2019-03-11:
I've got the same problem. Running HoRNDIS version 9.2 on macOS 10.13.6. 
When I start a command line ping before connecting and then connect my Samsung S7, it only transfers 1 or 2 packets. After that it only gives me request timeouts.

Would really like to use it since the latency is much lower than using my phone's Wi-Fi hotspot. Thanks for creating HoRNDIS!

jvalteren, 2019-03-11:
Update: I ran the Smart Switch uninstall program as mentioned in #78, however there was no 'Remove drivers' checkbox that I could tick. I removed the Samsung Modem manually from the network control panel, answering 'No' to the question if I wanted to have it return after I connected the phone again. Don't know if this actually removed the drivers.

Ping now responds for a longer period, have seen around 20-30 seconds of successful packet transfers. But alas, it still stops working. Couldn't find anything seemingly related in the system.log either.