jwise/HoRNDIS #98: Unable to boot after install on MacOS Mojave

anoldmanthe4th, 2019-02-23:
I just recently installed this tool to use tethering with my OnePlus 6t. After the installation it gave the prompt to reboot, I rebooted and now it will freeze on the booting loading bar. Once the loading bar is full the MacBook freezes and you have to hold the power button to turn it off. I'm now unable to use my MacBook, the recovery mode's disk utility's first aid didn't help.

d235j, 2019-02-23:
Can you boot with Command-V pressed down and provide a photo of the log that is displayed?

anoldmanthe4th, 2019-02-23:
Let me check, I'll be right back with you!

anoldmanthe4th, 2019-02-23:
Having a issue trying to display the log because I dual boot windows and it keeps booting into windows rather than MacOS however I still can choose MacOS through the boot picker.

d235j, 2019-02-23:
Get to the boot picker, select macOS, then hold Command-V and press Return.

anoldmanthe4th, 2019-02-23:
https://drive.google.com/file/d/1px-8gdrlbqsx1mBtKDcE9ysrmOz2BErW/view?usp=drivesdk

anoldmanthe4th, 2019-02-23:
I saw it saw at one point that some extentions library doesn't exist

d235j, 2019-02-23:
Can you boot into single user mode with Command-S then do
   ls /Library/Extensions
?

I see VirtualBox drivers and those are known to cause conflicts.
Apparently your GPU driver is refusing to loc for whatever reason.

anoldmanthe4th, 2019-02-23:
It responded with: -sh: Is: command not found

anoldmanthe4th, 2019-02-23:
Yeah that would explain alot because I do have virtual box installed

d235j, 2019-02-23:
LS (lowercase command) should always work in single user mode... if it doesn’t that’s the sign of a broken system :(

anoldmanthe4th, 2019-02-23:
Wait should I have mounted the hard drive first? It says that to mount the root device as read-write I have to type in 
$ /sbin/fsck -fy
Or
$ /sbin/mount -uw /

anoldmanthe4th, 2019-02-23:
Huh I tried the fsck -fy
It returned:
** The volume /Dev/rdisk1s1 appears to be OK.

anoldmanthe4th, 2019-02-23:
https://drive.google.com/file/d/1MzCKNBpkAgvGe04AJ-EPuuWlySI2wGS0/view?usp=drivesdk

d235j, 2019-02-23:
Does `ls /Library/Extensions` still not work?
Please note that these commands and paths are case-sensitive and any typo would cause them to fail (and that's lowercase LS).

anoldmanthe4th, 2019-02-23:
Ok so after contacting apple support, booting into safe mode and manually removing the HoRNDIS.kext from the Library Extensions my computer is functioning normally again! Thank you for the help thought, if it wasn't for you I would not have known where to look!

d235j, 2019-02-23:
Okay, so safe mode still works. That's good to know.

I suspect this is a conflict between HoRNDIS and another kext you have. While booted normally, can you run the following command in Terminal and paste the output?

`kextstat | grep -v 'com.apple'`

anoldmanthe4th, 2019-02-23:
Sorry but I don't want to risk compromising my computer anyfarther, as I have some crucial documents. Should I close and delete this issue?

d235j, 2019-02-23:
@anoldmanthe4th that command will only give us information on what other kexts are loaded and will not cause any changes to your computer. That way we can hopefully reproduce the bug.

My suspicion is VirtualBox — if you could also provide us with the VirtualBox version that you currently have on your computer (at the time of the failure, without updating it), that would help as well.

anoldmanthe4th, 2019-02-23:
I would love to to help but I can't risk it. However I can tell you the kexts that I have in my extentions folder. 

ACS6x
ArcMSR
ATTOCelerityFC8
ATTOExpressSASHBA2
CalDigitHDProDrv
HighPointIOP
HighPointRR
hp_io_enabler_compound
PromiseSTEX
SoftRAID
SteamInput-DS4v1
SteamInput

Hope this helps, I have removed the HoRNDIS kext obviously. I have Virtual box installed, as well as some legacy HP printer software. I'm using Virtual box version 6.0.2 r128162.

d235j, 2019-02-23:
Thanks, that helps.
The reason that command is useful is because it would list out all the kexts that are being loaded. Some software such as VirtualBox puts kexts elsewhere than in /Library/Extensions, unfortunately.
If you're in doubt, here is the manual page for the `kextstat` command: https://ss64.com/osx/kextstat.html
the `| grep -v 'com.apple'` part is to exclude Apple kexts.

Alternatively you can go to System Profiler or System Information (in Applications -> Utilities), then go to the Software -> Extensions section, then look for ones that are "Loaded: Yes" and "Obtained from: Identified Developer" or "Obtained from: Not Signed". 

All of the kexts you listed except the following come with macOS:

- hp_io_enabler_compound
- SteamInput-DS4v1
- SteamInput

I am very much suspicious of SteamInput, but from what I can tell, Steam will not install this unless one has an appropriate controller.

anoldmanthe4th, 2019-02-23:
I went into the System Report and here is the extention that is Loaded and from a "Identified Developer" or "Not Signed."
and supprisingly only 1 was loaded
SteamInput, Loaded, Obtained from Identified Developer.

anoldmanthe4th, 2019-02-23:
Otherwise, the full "Not Signed" or "Identified developer" list is in this screenshot.
[image]

d235j, 2019-02-23:
> and supprisingly only 1 was loaded
> SteamInput, Loaded, Obtained from Identified Developer.

Yeah... that's the kext that I'm suspicious of causing this conflict. I'm having an awfully difficult time finding a copy to install it, though, or getting Steam to install it.

Do you use a Steam Link, a Steam Controller, or both?

d235j, 2019-02-23:
I found it! I'll see what happens when I install it.

anoldmanthe4th, 2019-02-23:
I have used steam big picture mode,

d235j, 2019-02-23:
@anoldmanthe4th what model of Mac do you have (the Model Identifier from System Information would be helpful here), and do you have any USB devices plugged in?

anoldmanthe4th, 2019-02-23:
I don't have access to my Mac at the moment, but I have a MacBook Air 2017, and I think that I had my OnePlus 6t plugged in when HoRNDIS was done install ing, as I wanted to test it before rebooting and the hotspot tethering didn't work pre-reboot (of course)

svenatkins, 2019-04-20:
Same problem here. After installing the kext my MacBook Pro dosen't boot anymore.
Removed the kext in Single User Mode and booted up again.

It is a MacBookPro 10.1 with macOS Mojave 10.14.5 Beta.

Loaded extension:
Snoopy:~ sven$ kextstat | grep -v 'com.apple'
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
  163    0 0xffffff7f84ba8000 0x6000     0x6000     com.Cycling74.driver.Soundflower (1.6.7) 3579CD1C-ADE2-32EE-87C0-B8DD7E3A4CBB <133 6 5 3>
  179    0 0xffffff7f84cf9000 0x8000     0x8000     com.sophos.kext.sfm (9.7.4) 47394905-E2D3-3643-804D-D278B1BF74FB <6 5 2 1>
  180    0 0xffffff7f84d01000 0x7000     0x7000     com.sophos.nke.swi (9.7.15) 727E1482-79C1-3F33-A3C4-F147A406D5CA <5 1>
  181    0 0xffffff7f84d08000 0x8000     0x8000     com.sophos.kext.oas (9.7.4) FE151087-F3B0-3F51-BD15-3313B8BCDDAB <6 5 2 1>

Thanks.

harmtemolder, 2019-08-16:
Same on a MacBook 12 inch 2017 with Mojave 10.14.6

```
harmtemolder:HoRNDIS harmtemolder$ kextstat | grep -v 'com.apple'
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
harmtemolder:HoRNDIS harmtemolder$ 
```

wayneconnor, 2019-11-06:
I installed horndis on my 2018 Mac Mini Mojave, it is freezing at the end of reboot. It will not boot into safe mode, and single user mode is sitting there with an error saying consider rebuild of pre link Kernel. Org.virtualbox.kwxt.vboxdrv triggered rebuild
And now it's giving me some lines I don't understand.

wayneconnor, 2019-11-06:
Is there anywhere I can get better help eg Facebook chat?

wayneconnor, 2019-11-06:
It's my main computer and it's unusable

wayneconnor, 2019-11-06:
[image]
,

wayneconnor, 2019-11-06:
If you have any idea what to do please email me on wayne@dubbo.org

I've looked in library extensions and there in no horndis.kext file. 

I do not have access to safe mode or single user mode the only thing I can boot into is recovery mode and get into terminal.

wayneconnor, 2019-11-07:
After speaking with Apple I needed to reboot from Recovery partition and reinstall Mojave.

onoakino, 2020-07-18:
HoRNDIS 9.2 working flawlessly on my hackintosh Mojave 10.14.6. Installed via brew.

hgk8, 2020-08-25:
I'm having the same issue. I tried using the terminal to delete it but there is "no such file or directory". I installed the 9.2 version.

I am so desperate. Please help me😭

hgk8, 2020-08-25:
> After speaking with Apple I needed to reboot from Recovery partition and reinstall Mojave. 
> 

Would reinstall loses any data? This is also the only computer I have. And I haven't been backing up for a while.

anoldmanthe4th, 2020-08-25:
Yes the reinstall would cause a loss of data

anoldmanthe4th, 2020-08-25:
I got my Macbook back to working order by booting into safe mode (it took a fair amount of time to boot into it) then opening Finder and deleting the `HoRNDIS.kext` under `/System/Library/Extensions` and `/Library/Extensions` folder