jwise/HoRNDIS #100: GPLv2+ or GPLv3?

swarren, 2019-03-06:
COPYING.rtf says GPLv3, but the source files say GPLv2+. I might like to distribute this driver, but will probably only be allowed to do so if I can distribute under GPLv2 not be required to use GPLv3. Thanks!

jwise, 2019-03-07:
Hi Stephen -- hope things are going well back at the big green.  What's the application?  (If you'd prefer not to say in GitHub, feel free to send me some mail.)  I guess the license in the source file overrides, though I probably meant v3, for the anti-TiVoization clause.

swarren, 2019-03-07:
It's for our Jetson products. They act as a USB device, and implement Ethernet over USB so that you can communicate with them from the host PC/... without having to connect Jetson to a network. Right now we implement both RNDIS (for Windows) and ECM (for Mac), but then a Linux host ends up seeing two Ethernet adapters which is a bit annoying. I'm contemplating enabling just RNDIS by default and asking Mac users to install HoRNDIS or edit the script to enable ECM if they need it. To make it easier, we could ship HoRNDIS with Jetson so people don't need to download it (Jetson is a USB composite gadget so is a USB Mass Storage device too, so the host can access a few files like READMEs right now; we could add the .pkg file there). However, bundling HoRNDIS would likely be much easier to get approved via IP audit if it's GPLv2 not GPLv3. If it's actually GPLv3, we may just leave things as they are, or link people to the website instead.

jwise, 2019-03-07:
I think the source says it's GPLv2+, so you could redist it if SWIPAT needs it to be GPLv2, and I definitely do not object to that application.  But you'd probably be better off, I think, linking to the web site: the system image will end up out of date, and you'll run into weird support issues when Apple changes APIs...

I *think*, by the way, that modern Windows supports ECM, too.  My Pixel 2 only speaks ECM, as far as I can tell.

swarren, 2019-03-08:
I just retested all of RNDIS/ECM/NCM/EEM on a fully updated Windows, and only RNDIS seems to be supported by Microsoft. I searched Google and there are a variety of 3rd-party ECM drivers, but nothing in Windows itself.

Based on your comments about Mac OS API changes, I agree that simply linking to the HoRNDIS website sounds best.

Re: the source files being GPLv2+: I don't think that would have helped much:
a) If we relied on that, we'd have to take the source files and build them ourselves, rather than re-distributing your pre-compiled binaries. That's something I'd rather not take on.
b) While some/all of the files allow that exception, the COPYING.rtf file at the top level still says GPLv3, which makes it look like the overall collection is distributed as GPLv3, even if parts of it could use other licenses. I expect lawyers will still consider everything to be distributed as GPLv3 not GPLv2+ if it's taken from this repo.

Thanks for the comments. I'll close this now based on linking users to the website, so we then don't care about the license details.

closed 2019-03-08 by swarren

jwise, 2019-03-08:
Ok, sounds good.  Note also that if you built it yourself, you'd have to deal with code-signing yourself (you can't load a kext without code-signing it *and* having the key you sign it with get the Extra Special Kext Bit from Apple).  So I think using binaries that I or @mikhailai build is probably the way to go.

(Reminds me that I need to find something to do with my Jetson board...)