jwise/HoRNDIS #102: unable to install on macOS Catalina

dehydratedH2O, 2019-07-03:
HoRNDIS is incompatible with Catalina due to the new read-only system partition. installation via the .pkg or brew will fail.

see the What's New in APFS session from WWDC19: https://developer.apple.com/videos/play/wwdc2019/710/

there may be a workaround by disabling SIP using csrutil in Recovery Mode, but I haven't had a chance to try it yet.

the "right" way to fix HoRNDIS long term is likely to migrate it to using the new DriverKit framework, documented at https://developer.apple.com/documentation/driverkit

M69W, 2019-07-03:
This is a kext that be pick up from HoRNDIS-9.2.pkg in the S/L on Majova.
I don't know whether it's work in the Catalina?
```EFI/Clover/kexts/Other```
[HoRNDIS.kext.zip](https://github.com/jwise/HoRNDIS/files/3354105/HoRNDIS.kext.zip)

FengFeng-is-me, 2019-07-05:
My OS version is 10.15 beat3 it work,it's able work on Catalina. try second times or restart your machine.And if installed  contected phone it always not work or can't find your network You just need to copy  /Library/Preferences/SystemConfiguration/NetworkInterfaces.plist and /Library/Preferences/SystemConfiguration/preferences.plis delete these file. But if you try this,you may not connect the network anymore.and you just backup these two file,you will find it work.It's amazing. You have to remember to backup before you delete system file.

d235j, 2019-07-05:
Is the installer still trying to put files in /System? It should not be doing that.

dehydratedH2O, 2019-07-05:
In 10.16 basically anything not in /Users or /Applications is write protected. The installer is attempting to write to protected locations. Using an old .kext won't work without disabling SIP, which should not be required for users.

DriverKit should be adopted before Catalina is released if this project wants to support it. If I find time to do it myself, I will, but until then if anyone has a chance to do it, it's the "right way".

FengFeng-is-me, 2019-07-05:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    I don’t understand your mean clearly,because I just try to install several pkg （version 8 9 9.1）,and it show me successful.But I can’t cannect my phone at first time. I tried to delete 2 files and backup It work now.

FengFeng-is-me, 2019-07-05:
[image]

lutnos, 2019-10-09:
I now have HoRNDIS installed and working on the release 10.15 Catalina.

The process:

- Shutdown
- Hold command+R and press power
- Launch terminal
- csrutil disable
- Reboot
- Launch terminal
- mount -uw /
- Run the HoRNDIS 9.2 installer
- Shutdown
- Hold command+R and press power
- Launch terminal
- csrutil enable
- Reboot

Ultimately, DriverKit is the way too go.

But for now I hope that this helps others

d235j, 2019-10-09:
Can one of you provide a copy of the Installer log from a failed installation, with all logs enabled, not only errors?

qdtjvszxc, 2019-10-09:
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Product archive /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg trustLevel=350
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: location = file://localhost
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: -[IFDInstallController(Private) _buildInstallPlanReturningError:]: file://localhost/usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg#HoRNDIS-kext.pkg
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Set authorization level to root for session
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Administrator authorization granted.
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Will use PK session
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Using authorization level of root for IFPKInstallElement
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Starting installation:
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Configuring volume "Macintosh HD"
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Preparing disk for local booted install.
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Free space on "Macintosh HD": 453.09 GB (453093011456 bytes).
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: Create temporary directory "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T//Install.289452ieUJ2"
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: IFPKInstallElement (1 packages)
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installd[507]: PackageKit: Adding client PKInstallDaemonClient pid=28945, uid=0 (/usr/sbin/installer)
2019-10-10 03:50:00+08 YipingdeMacBook-Pro installer[28945]: PackageKit: Enqueuing install with framework-specified quality of service (utility)
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: ----- Begin install -----
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: request=PKInstallRequest <1 packages, destination=/>
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: packages=(
	    "PKLeopardPackage <id=com.joshuawise.kexts.HoRNDIS, version=0, url=file://localhost/usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg#HoRNDIS-kext.pkg>"
	)
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: Set reponsibility for install to 18666
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: Extracting file://localhost/usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg#HoRNDIS-kext.pkg (destination=/Library/InstallerSandboxes/.PKInstallSandboxManager/603D0B6A-6DB4-4042-B0B8-020C5A71589D.activeSandbox/Root, uid=0)
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: prevent user idle system sleep
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: suspending backupd
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: Using trashcan path /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/PKInstallSandboxTrash/603D0B6A-6DB4-4042-B0B8-020C5A71589D.sandboxTrash for sandbox /Library/InstallerSandboxes/.PKInstallSandboxManager/603D0B6A-6DB4-4042-B0B8-020C5A71589D.activeSandbox
2019-10-10 03:50:01+08 YipingdeMacBook-Pro install_monitor[28946]: Temporarily excluding: /Applications, /Library, /System, /bin, /private, /sbin, /usr
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: Shoving /Library/InstallerSandboxes/.PKInstallSandboxManager/603D0B6A-6DB4-4042-B0B8-020C5A71589D.activeSandbox/Root (1 items) to /
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: PackageKit: Executing script "./postinstall" in /private/tmp/PKInstallSandbox.tN75vt/Scripts/com.joshuawise.kexts.HoRNDIS.qKEa0q
2019-10-10 03:50:01+08 YipingdeMacBook-Pro installd[507]: ./postinstall: touch: /System/Library/Extensions: Read-only file system
2019-10-10 03:50:01+08 YipingdeMacBook-Pro install_monitor[28946]: Re-included: /Applications, /Library, /System, /bin, /private, /sbin, /usr
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: releasing backupd
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: allow user idle system sleep
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: Cleared responsibility for install from 28945.
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running the script for the package "HoRNDIS-9.2.pkg"." UserInfo={NSFilePath=./postinstall, NSURL=file://localhost/usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg#HoRNDIS-kext.pkg, PKInstallPackageIdentifier=com.joshuawise.kexts.HoRNDIS, NSLocalizedDescription=An error occurred while running the script for the package "HoRNDIS-9.2.pkg".} {
	    NSFilePath = "./postinstall";
	    NSLocalizedDescription = "\U8fd0\U884c\U8f6f\U4ef6\U5305\U201cHoRNDIS-9.2.pkg\U201d\U7684\U811a\U672c\U65f6\U51fa\U9519\U3002";
	    NSURL = "file://localhost/usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg#HoRNDIS-kext.pkg";
	    PKInstallPackageIdentifier = "com.joshuawise.kexts.HoRNDIS";
	}
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: Running idle tasks
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installer[28945]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running the script for the package "HoRNDIS-9.2.pkg"." UserInfo={NSFilePath=./post [... cut ...] ng the script for the package "HoRNDIS-9.2.pkg".}
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: Removing client PKInstallDaemonClient pid=28945, uid=0 (/usr/sbin/installer)
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installd[507]: PackageKit: Done with sandbox removals
2019-10-10 03:50:02+08 YipingdeMacBook-Pro installer[28945]: Install failed: The installer encountered an error that caused the installation to fail. Please contact the software manufacturer for assistance.

d235j, 2019-10-09:
The post-install script needs to be changed.

`sudo touch /System/Library/Extensions` was deprecated a while ago for rebuilding the kextcache. The currently supported method is `sudo kextcache -system-prelinked-kernel`.

@jwise @mikhailai can you take a look? The kext and package will also need notarization. I can do some testing once I have a Catalina install set up.

d235j, 2019-10-11:
Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with the right-click-and-open method). I'll try to get a PR to the project in place soon.
[HoRNDIS-9.2-catalina_install.pkg.zip](https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip)

cheyennerhap, 2019-10-12:
Thanks so much, this update worked perfectly!

ice-cream-headache, 2019-10-14:
It works perfect.Thanks a lot.

alexandrehuat, 2019-10-17:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Wonderful! It works. Thank you. Great job.

mongezik19, 2019-10-20:
Thank you so much @d235j

jadonk, 2019-11-07:
@d235j any updates on a PR? Catalina has a regression related to BeagleBone's USB configuration, which includes a composite adapter including Mass Storage Class, CDC-ACM (serial), RNDIS and CDC-ECM (networking, previously supported by OS X). It appears to get the CDC-ECM working, we have to disable the RNDIS support (or perhaps enumerate it after RNDIS?). Therefore, we are considering just removing CDC-ECM, which otherwise breaks the latest Windows 10 if included in the composite adapter before RNDIS.  Proprietary OS vendors. :-(

whao, 2019-11-28:
@d235j I tried your binary and it worked well! Thanks so much for your job! But there is a small issue. Every time I reconnect my phone to mac, it creates a new interface in Preference -> Network. And the interface number increases. For example, it creates Ethernet Adapter (en6), after plugging out and in, it creates  Ethernet Adapter (en7), and en6 is still there. Now it is alreay en15 on my laptop and I don't know how to delete old ones. Deleting in the Network panel doesn't work.

abdullahsari, 2019-12-13:
Catalina 10.15.2 straight up forbids the whole installation even when you bypass Gatekeeper because of "incompatibility".

suru1994, 2019-12-14:
not working in 10.15.2 update

marcocursi, 2019-12-14:
> not working in 10.15.2 update

Unfortunately...

johanf85, 2019-12-16:
> not working in 10.15.2 update

It's working on my 10.15.2. Did had to accept something with a privacy and safety popup

marcocursi, 2019-12-16:
> > not working in 10.15.2 update
> 
> It's working on my 10.15.2. Did had to accept something on a privacy and safety popup

Really?! I try again.

marcocursi, 2019-12-16:
> > > not working in 10.15.2 update
> > 
> > 
> > It's working on my 10.15.2. Did had to accept something on a privacy and safety popup
> 
> Really?! I try again.

No problem on 10.15.1 updated 10.15.2.
New Mac with 10.15.2 no working.
Error “
Contact software developer “.

aivils, 2019-12-20:
I go to system preferences -> security & privacy. I see - System software from developer "Joshua Wise" was blocked from loading. 
unblock it by pressing "Allow" 
it's working under 10.15.2

svalab-hatci, 2019-12-25:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Thank you so much, you just saved my life !

air2terre, 2019-12-30:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Works perfectly for me and the DJI application on Catalina. Thank you very much !

gmcinalli, 2020-01-24:
> @d235j I tried your binary and it worked well! Thanks so much for your job! But there is a small issue. Every time I reconnect my phone to mac, it creates a new interface in Preference -> Network. [...]

Happens to me too.

Adriano-A3, 2020-01-27:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

MVP TRIGGERED

liho98, 2020-02-15:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Hi, I am using your version, and it works. However, every time I disconnect and connect my device for USB tethering, it create a new network interface, which is quite annoying. I realised that it doesn't recognised my phone model (Xiaomi Pocophone F1). Can you help me solve this issue, thanks in advance!

[image]

dragfly, 2020-02-23:
Thanks @d235j, your version does get correctly installed on Catalina.

@liho98 I'm experiencing the same issue: this is apparently due to the driver generating a new MAC address every time the USB tethering is enabled. I assume this is an Android behaviour rather than a HoRNDIS issue?

liho98, 2020-02-24:
> Thanks @d235j, your version does get correctly installed on Catalina.
> 
> @liho98 I'm experiencing the same issue: this is due to Android generating a new MAC address every time the USB tethering is enabled. It's called [MAC Randomization](https://source.android.com/devices/tech/connect/wifi-mac-randomization) and is considered an official security feature

Great information, hopefully there will be a solution. . . But it looks difficult ... Btw thanks.

billy1380, 2020-02-26:
The installer works in 10.15.3 but I cannot tether my pixel 3

salamander71, 2020-03-05:
The installer works in 10.15.3(19D76) but I cannot tether my LG V35

gshmu, 2020-03-15:
@billy1380  @salamander71  are you try to run `sudo kextload /Library/Extensions/HoRNDIS.kext`?

billy1380, 2020-03-15:
> @billy1380 @salamander71 are you try to run `sudo kextload /Library/Extensions/HoRNDIS.kext`?

/Library/Extensions/HoRNDIS.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).

Just checking the logs now.

billy1380, 2020-03-15:
@gshmu just followed the instructions here https://www.synack.com/blog/high-sierras-secure-kernel-extension-loading-is-broken/ and it now works. Thanks a lot.

Moulick, 2020-03-21:
`brew cask install horndis` failed on Catalina 10.15.3 (19D76) with error 

```Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=moulick.aggarwal USER=moulick.aggarwal USERNAME=moulick.aggarwal /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
installer: Package name is HoRNDIS
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
Follow the instructions here:
  https://github.com/Homebrew/homebrew-cask#reporting-bugs
/usr/local/Homebrew/Library/Homebrew/system_command.rb:176:in `assert_success!'
/usr/local/Homebrew/Library/Homebrew/system_command.rb:53:in `run!'
/usr/local/Homebrew/Library/Homebrew/system_command.rb:29:in `run'
/usr/local/Homebrew/Library/Homebrew/system_command.rb:33:in `run!'
/usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:59:in `block in run_installer'
/usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:65:in `with_choices_file'
/usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:52:in `run_installer'
/usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:34:in `install_phase'
/usr/local/Homebrew/Library/Homebrew/cask/installer.rb:218:in `block in install_artifacts'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/set.rb:777:in `each'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/set.rb:777:in `each'
/usr/local/Homebrew/Library/Homebrew/cask/installer.rb:209:in `install_artifacts'
/usr/local/Homebrew/Library/Homebrew/cask/installer.rb:101:in `install'
/usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:22:in `block in run'
/usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:16:in `each'
/usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:16:in `run'
/usr/local/Homebrew/Library/Homebrew/cask/cmd/abstract_command.rb:36:in `run'
/usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:92:in `run_command'
/usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:158:in `run'
/usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:123:in `run'
/usr/local/Homebrew/Library/Homebrew/cmd/cask.rb:9:in `cask'
/usr/local/Homebrew/Library/Homebrew/brew.rb:103:in `<main>'```

slauhk, 2020-03-25:
> Thank you so much, you just saved my life !

Tks for help, It is the one we need.

Malik056, 2020-03-26:
> `brew cask install horndis` failed on Catalina 10.15.3 (19D76) with error
> 
> `` Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=moulick.aggarwal USER=moulick.aggarwal USERNAME=moulick.aggarwal /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output: installer: Package name is HoRNDIS installer: Installing at base path / installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.) Follow the instructions here: https://github.com/Homebrew/homebrew-cask#reporting-bugs /usr/local/Homebrew/Library/Homebrew/system_command.rb:176:in `assert_success!' /usr/local/Homebrew/Library/Homebrew/system_command.rb:53:in `run!' /usr/local/Homebrew/Library/Homebrew/system_command.rb:29:in `run' /usr/local/Homebrew/Library/Homebrew/system_command.rb:33:in `run!' /usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:59:in `block in run_installer' /usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:65:in `with_choices_file' /usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:52:in `run_installer' /usr/local/Homebrew/Library/Homebrew/cask/artifact/pkg.rb:34:in `install_phase' /usr/local/Homebrew/Library/Homebrew/cask/installer.rb:218:in `block in install_artifacts' /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/set.rb:777:in `each' /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/set.rb:777:in `each' /usr/local/Homebrew/Library/Homebrew/cask/installer.rb:209:in `install_artifacts' /usr/local/Homebrew/Library/Homebrew/cask/installer.rb:101:in `install' /usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:22:in `block in run' /usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:16:in `each' /usr/local/Homebrew/Library/Homebrew/cask/cmd/install.rb:16:in `run' /usr/local/Homebrew/Library/Homebrew/cask/cmd/abstract_command.rb:36:in `run' /usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:92:in `run_command' /usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:158:in `run' /usr/local/Homebrew/Library/Homebrew/cask/cmd.rb:123:in `run' /usr/local/Homebrew/Library/Homebrew/cmd/cask.rb:9:in `cask' /usr/local/Homebrew/Library/Homebrew/brew.rb:103:in `<main>' ``
> 
> but still I ran `sudo kextload /Library/Extensions/HoRNDIS.kext` and the usb tethering worked 🤣 how even

This is working on MacOS Catalina 10.15.4. Great Work! Thanks (y)

johanf85, 2020-03-29:
I got it working installing this from the comment of @d235j 
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

As @aivils was writing:
> I go to system preferences -> security & privacy. I see - System software from developer "Joshua Wise" was blocked from loading.
> unblock it by pressing "Allow"
> it's working under 10.15.2

I got the above message message from @aivils after I plugged in an USB Android device and toggled USB tethering. Hitting Alow got it working for me. 

Using 10.15.3

antoinerousseau, 2020-04-03:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Works for me but I had to reboot my Mac after installing to get it working.
Using a Redmi Note 7.
Thanks @d235j, time to make that PR ;)

darbela4, 2020-04-11:
Muchas gracias, funciona correctamente en MacOS Catalina 10.15.4
[image]

gmcinalli, 2020-04-14:
> @liho98 I'm experiencing the same issue: this is due to Android generating a new MAC address every time the USB tethering is enabled. It's called [MAC Randomization](https://source.android.com/devices/tech/connect/wifi-mac-randomization) and is considered an official security feature

@dragfly Is there anything we can do to prevent this from happening?

shingz96, 2020-04-17:
> @dragfly Is there anything we can do to prevent this from happening?

can try this if needed
https://www.insanelymac.com/forum/topic/101559-remove-and-reinstall-ethernet/

itsashis4u, 2020-04-27:
Not working on catalina 10.15.4 and nokia 5.1 plus

jeremija, 2020-05-14:
I just upgraded to Catalina 10.15.4 and I received this message:

[image]

asozcan, 2020-05-16:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

After a restart worked on 10.15.4 catalina. Device is Huawei Android 10.

philb-lr, 2020-05-19:
> Here's a version packaging the exact same binary-identical .kext together with a modified install script. This does mean the .pkg is unsigned (so you'd have to trust me and bypass Gatekeeper with [...]

Worked for me, too. LG V35.

FLiiPR, 2020-05-23:
Hey Guys!
I just signed up to tell you all that you are my Heroes!
Thank you so much for saving my Day(s)! :) 

Cheers!

roke92, 2020-05-25:
> Works for me but I had to reboot my Mac after installing to get it working.
> Using a Redmi Note 7.
> Thanks @d235j, time to make that PR ;)

hi, i've tried also installing and rebooting my macbook air with the latest version of catalina but it doesn't work with my redmi note 7. what have you done to make it work?

billy1380, 2020-05-25:
@roke92 have you checked your Security and Privacy -> General tab... is there a warning in there when you plug the phone?

roke92, 2020-05-25:
no, there is no warning message

billy1380, 2020-05-25:
> no, there is no warning message

Not even when you run `sudo kextload /Library/Extensions/HoRNDIS.kext`

roke92, 2020-05-25:
can you describe the steps? so i installed the package and rebooted my mac, than i have just to run the kext command?

billy1380, 2020-05-25:
@roke92 yes... for what I remember... but I found the full instructions for something similar here
https://www.synack.com/blog/high-sierras-secure-kernel-extension-loading-is-broken/ take a look if you get a chance.

roke92, 2020-05-25:
it says:    /Library/Extensions/HoRNDIS.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8)

billy1380, 2020-05-25:
@roke92 That is what I was getting... now before you run anything else check your Security and Privacy -> General tab.

roke92, 2020-05-25:
now it works, i just rebooted my mac 
thank you

jipiboily, 2020-06-03:
Thanks @d235j for the patched, compiled version.

Just a quick question: would you mind sharing the code changes you had to do? I don't see anything in your fork at first.

That could help prepare a PR for someone if we had the commits and/or diff of some sorts.

Cheers :)

Tree265, 2020-06-05:
I managed to get my tethering connection working on my Nexus 5 reflashed with Ubuntu Touch (Ubuntu 16.04 OTA-12) using macOS Catalina 10.15.2 (2007 iMac itself modified with the Penryn CPU and patched (Dosdude1.com) for Catalina). On the Nexus 5, I opened a terminal and typed: sudo android-gadget-service enable rndis, then sudo tethering enable. I did not have the usb cable plugged in when I typed this. On the Mac, I tried to install HoRNDIS, but it failed. I tried the updated pkg mentioned here, but it hung there for awhile. I aborted that, but the kext was installed. sudo kextload /Library/Extensions/HoRNDIS.kext seemed to make things work. I plugged the Nexus 5 in with the usb cable and the ethernet connection popped up in Network Preferences. Internet seems fine.

[34 more replies]