jwise/HoRNDIS #104: Documentation suggestion... mention brew to install

drevicko, 2019-08-06:
It might be worth mentioning that you can install HoRNDIS with brew on [the webpage](https://joshuawise.com/horndis) (and indeed anywhere else appropriate). 

I don't know about others, but I usually prefer brew for this sort of thing, as it (hopefully) deals with any installation details etc.. as well as uninstallation and afaik is usually pretty good on version compatibility.

BigD1979666, 2019-10-02:
Could you link it please? I cant find any mention of brew install

drevicko, 2019-10-03:
From the readme file in this repo:

### From Homebrew

```sh
brew cask install horndis
sudo kextload /Library/Extensions/HoRNDIS.kext
```

orlein, 2021-02-27:
The command should be replaced with:

` brew install --cask horndis`