jwise/HoRNDIS #108: А lot of duplicate

AlexanderBuryak, 2019-11-03:
if you turn the device on and off, you can see a lot of duplicates in the network on Mac OS CATALINA  10.15.1 (19B88) MacBook air 2014

billj-korg, 2019-11-08:
Speaking of duplicates, is this a duplicate of issue #54?

billj-korg, 2019-11-09:
In case this might be of some help to some that encounter this issue. The device we were using that exhibited this problem was using ConfigFS to setup the RNDIS USB gadget. The host_addr file, by default, is a generated random MAC address for the host. This seems to be the cause of new records being created by the Mac OS each time. By setting the host_addr contents to a fixed value, new records are no longer created on Mac OS, which then uses the record matching the host_addr contents.

geekeren, 2020-03-29:
> In case this might be of some help to some that encounter this issue. The device we were using that exhibited this problem was using ConfigFS to setup the RNDIS USB gadget. The host_addr file, by [...]

Is any workaround to fix this issue?