jwise/HoRNDIS #111: HoRNDIS not working on Catalina

biaji, 2019-12-02:
After I upgrade my mac mini 2012 to Catalina.  HoRNDIS stop working, and complain:

HoRNDIS: init: HoRNDIS tethering driver for Mac OS X, 9.2
HoRNDIS: rndisInit: 'PH-1': ver=1.0, max_packets_per_transfer=15, max_transfer_size=23700, packet_alignment=2^4
HoRNDIS: init: (network interface) starting up with MTU 1500
ifnet_attach: All kernel threads created for interface en35 have been scheduled at least once. Proceeding.
Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
Boot args: <private>
debug=0x0
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
Found class: <private>
OnQueue
no services left, exiting
DK: IOUserServer(com.apple.DriverKit.AppleUserECMData-0x1000007be)::terminate(AppleUserECMData-0x1000007be) server exit before start()
DK: AppleUserECMData-0x1000007be::start(RNDIS Ethernet Data-0x1000007b0) fail

biaji, 2019-12-02:
After I reinstall through brew and did run the command:    

```shell
sudo mount -uw /
```

It works again.  lol

biaji, 2019-12-02:
And .....  After a while.   It failed again.

Niek, 2019-12-05:
Even installing on Catalina is a no-go (both homebrew and manual):
```
Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=niek USER=niek USERNAME=niek /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
installer: Package name is HoRNDIS
installer: Upgrading at base path /
installer: The upgrade failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
```

pooyabarati, 2019-12-06:
Upgraded to catalina yesterday and it's not even installing using .pkg method

Belcarra, 2019-12-07:
Installing KEXT's in Catalina requires that you have it signed with a KEXT
capable developer certificate and notarized by Apple.

You need to request the KEXT cert from Apple.

Any developer can set up to do notarization. The process for KEXT's is not
well documented in is only slightly painful.

Andreychikov-Vasiliy, 2019-12-07:
Hmm, on hackintosh it's work well. Event it gave error at installation. Also note that I tried install in different run ways (as Ctrl+Open, Option+Open). 
[image]

biaji, 2019-12-09:
> Even installing on Catalina is a no-go (both homebrew and manual):
>
> ```
> Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=niek USER=niek USERNAME=niek /usr/sbin/installer [...]

Maybe you can install it through this way: 

https://github.com/jwise/HoRNDIS/issues/102#issuecomment-541237232

DrSkilled, 2020-02-08:
> > Even installing on Catalina is a no-go (both homebrew and manual):
> > ```
> > Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=niek USER=niek USERNAME=niek /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
> > installer: Package name is HoRNDIS
> > installer: Upgrading at base path /
> > installer: The upgrade failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
> > ```
> 
> Maybe you can install it through this way:
> 
> [#102 (comment)](https://github.com/jwise/HoRNDIS/issues/102#issuecomment-541237232)

Thanks！！

eclansky, 2020-02-17:
I just did a brew cask install of hornsdis but got a nasty kernel panic coming out of sleep. Sifted through the console output a bit but its long n nasty. Anybody have a similar panic running the 9.2 driver on Catalina??

100120796348430\nlast loaded kext at 95239829237006: com.joshuawise.kexts.HoRNDIS\t9.2 (addr 0xffffff7fa2d80000, size 36864)\nlast unloaded kext at 237956662920: >!A!ILpssGspi\t3.0.60 (addr 0xffffff7f9f355000, size 45056)\nloaded kexts:\ncom.joshuawise.kexts.HoRNDIS\t9.2\