jwise/HoRNDIS #114: How to uninstall correctly?

GeertCu, 2020-01-21:
After removing the HoRNDIS.kext file in /System/Library/Extensions and rebooting my Macbook I don't think the extention is loaded:

```
MBP15-2:~ geert$ kextstat | grep HoRNDIS
MBP15-2:~ geert$ 

```

Unfortunately I can still find a HoRNDIS.kext file in /Library/StagedExtensions/Library/Extensions.
How do I get rid of that file too?

eclansky, 2020-02-17:
I had same issue when I deleted horndis 9.2 from my MacBook.

`sudo kextcache --clear-staging`

Found it on stackoverflow
https://stackoverflow.com/questions/46535279/what-is-library-stagedextensions-for-and-how-to-remove-extensions

prismopensource, 2020-05-28:
Same issue here, `sudo kextcache --clear-staging` fixes this, maybe the documentation about uninstalling should be updated?

cking0987, 2020-06-23:
Thanks for the tip! +1 for adding it to the documentation on uninstalling.