jwise/HoRNDIS #116: Using it with Alcatel Link key IK40V USB LTE dongle

dinonovak, 2020-02-05:
Hi,

I am trying to use it with Alcatel LTE USB dongle.
it is working correctly under OpenWrt on a router with RNDIS support installed.

This is what i tried under MacOS Catalina:

1) loading kext
`# kextload /Library/Extensions/HoRNDIS.kext`

2) checking that is is loaded and it looks like it is loaded OK
```
# kextstat | grep RNDIS
  200    0 0xffffff7f83d2d000 0x6000     0x6000     com.joshuawise.kexts.HoRNDIS (9.2) 10ED3B68-FBFE-326E-B848-DC8082240F4A <29 25 8 6 5 3 1>
```

3) I am checking if it is recognized as USB device, and it is showing up (actually as it has disk partition with old 32bit software it is mounted)
# ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 17>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x1000003d8, registered, matched, active, busy 0 (1 ms), retain 10>
    +-o Apple T1 Controller@14100000  <class AppleUSBDevice, id 0x1000003da, registered, matched, active, busy 0 (2 ms), retain 26>
    +-o Mobilebroadband@14600000  <class AppleUSBDevice, id 0x100001d7a, registered, matched, active, busy 0 (1 ms), retain 11>

4) here are more details about USB device
```
# ioreg -p IOUSB -w0 -l

    +-o Mobilebroadband@14600000  <class AppleUSBDevice, id 0x100001d7a, registered, matched, active, busy 0 (1 ms), retain 11>
        {
          "sessionID" = 54027216902811
          "idProduct" = 61440
          "iManufacturer" = 1
          "bDeviceClass" = 0
          "bMaxPacketSize0" = 64
          "bcdDevice" = 578
          "iProduct" = 2
          "iSerialNumber" = 3
          "bNumConfigurations" = 1
          "Bus Power Available" = 250
          "USB Address" = 5
          "Built-In" = No
          "locationID" = 341835776
          "bDeviceSubClass" = 0
          "bcdUSB" = 512
          "USB Product Name" = "Mobilebroadband"
          "PortNum" = 6
          "non-removable" = "no"
          "AppleUSBAlternateServiceRegistryID" = 4294974840
          "bDeviceProtocol" = 0
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "kUSBCurrentConfiguration" = 1
          "Device Speed" = 2
          "USB Vendor Name" = "Alcatel"
          "idVendor" = 7099
          "uid" = "USB:1BBBF0004567890ABCDE"
          "IOGeneralInterest" = "IOCommand is not serializable"
          "USB Serial Number" = "1234567890ABCDE"
          "IOClassNameOverride" = "IOUSBDevice"
        }
```

Any hints what to try next as I do not see any network adapter showing up, or any messages with dmesg or in syslog.

vitospedito, 2020-04-06:
I am also interested. Same problem!

AlirezaH320, 2021-02-21:
Same problem with 4GEE mini which is an Alcatel based. any progress since then?