jwise/HoRNDIS #117: Not working with MacOS HighSierra 10.13.6 and Xiaomi Redmi Note 5 running ROM LineageOS 16.0 (android 9)

ademirml, 2020-02-19:
Hey guys I installed HoRNDIS 9.2 pkg and restarted my mac, but the device is not showing in the Network preferences, even when I try to add it manually, there is no USB option.

I am using MacOS HighSierra 10.13.6
Smartphone Xiaomi Redmi Note 5 running the ROM LineageOS 16.0 (android 9)

I tried to use USB tethering with this smartphone running this ROM with a Windows computer and it worked, so the problem is not with the device or the custom ROM.

Do I need to do any extra configuration to make it work with this device or HoRNDIS driver does not support it?

Also, I tried to use a MotoG 1ºst generation with this HoRNDIS version (9.2) and it also did not work. Is there a list of supported devices? If so, where can I find it? I also have a spare Xperia Z3 Compact, does it work with it? This way I could use it, but would have to flash the original ROM again.

Thanks
[image]
[image]

mikevaux, 2020-04-07:
I think this is probably an issue with High Sierra rather than your specific phone. I have spent ages trying to get it to work and have finally managed it...

I can explain more if that's helpful, but the steps I followed to get it to work are:

1. Boot into recovery mode and disable kext consent in the terminal `spctl kext-consent disable`
2. Boot normally and install Horndis via [_Homebrew_](https://github.com/jwise/HoRNDIS#from-homebrew) (manual install of the binary didn't work for some reason for me), plus `kextload`ing as per the instructions. At this point, you should be able to plug your phone in and tether as expected.
3. Boot into recovery mode and reenable kext consent in the terminal `spctl kext-consent enable`
4. Boot normally. It now won't work, as the Security & Privacy panel will show the 'allow' button for the extension, but you can't click it. To fix this, as per [this SE answer](https://apple.stackexchange.com/a/325768), set your keyboard preferences to 'All Controls', quit System Prefs, reopen System Prefs, go to Security & Privacy, tab to allow and hit space bar. Put your keyboard prefs back to normal.

Yuck! What an annoyingly convoluted set of security features High Sierra has!

Hope it works for you?!

mikevaux, 2020-04-07:
On reflection, you can _probably_ get away with just doing 2 and 4... I had assumed that disabling the kext consent would mitigate the crazy 'Allow' button issues, but I had to do 4 anyway, so it might be worth trying that first...

ademirml, 2020-04-07:
> On reflection, you can _probably_ get away with just doing 2 and 4... I had assumed that disabling the kext consent would mitigate the crazy 'Allow' button issues, but I had to do 4 anyway, so it might be worth trying that first...

Thanks man, I will try it later, hope it works