jwise/HoRNDIS #118: creates a new Ethernet connection on each phone restart

Stevemoretz, 2020-03-22:
I have already 10 Ethernets right now after using it for 10 days.In my network section.So this might not be related to your package and it could be related to something to change on the phone to get this fixed.But anyway if anyone's going to tell what's the problem and how to fix it,I think it could be maybe here.

How do I make it not create any new Ethernets?

I do a lot of setup on the ethernet setting up dns proxies and ... and new Ethernet comes on with no configuration.So I have to do all over again.

jamescrowley, 2020-04-30:
Same issue since upgrading to macOS Catalina

romaink-es, 2020-05-12:
same here, a new interface is created each time I enable USB tethering on my phone (OnePlus 6 running OxygenOS 10.3.2). I'm running macOS 10.15.3.

EETagent, 2020-05-15:
Same problem. OpenCore Hackintosh. 

Simple fix to remove clones is 
`rm -rf /Library/Preferences/SystemConfiguration`
That will delete all network settings though.

Clones in the NetworkInterfaces.plist could be easily deleted with Bash and PlistBuddy. Don't know how to delete them from preferences.plist. 

Is HoRNDIS still maintained? Last update was more than a year ago. There were probably some changes in Catalina for which HoRNDIS wasn't updated.

Stevemoretz, 2020-05-15:
> Same problem. OpenCore Hackintosh, Catalina.
> 
> Simple fix to remove clones is
> `rm -rf /Library/Preferences/SystemConfiguration`
> That will delete all network settings though.
> 
> Clones in the NetworkInterfaces.plist could be easily deleted with Bash and PlistBuddy. Don't know how to delete them from preferences.plist.
> 
> Is HoRNDIS still maintained? Last update was more than a year ago. There were probably some changes in Catalina for which HoRNDIS wasn't updated.

I'm the poster of this issue,about
`Clones in the NetworkInterfaces.plist could be easily deleted with Bash and PlistBuddy. Don't know how to delete them from preferences.plist.`
I didn't get everything you said,where is NetworkInterfaces.plist? If we delete stuff in it the interfaces will be removed? so what is preferences.plist for then?

EETagent, 2020-05-15:
> I'm the poster of this issue,about
> `Clones in the NetworkInterfaces.plist could be easily deleted with Bash and PlistBuddy. Don't know how to delete them from preferences.plist.`
> I didn't get [...]

Those are two file in /Library/Preferences/SystemConfiguration where interfaces are stored. Deleting them or changing them solves too many Enxxxx in settings but is useless for your problem with overwrited configuration. I don't think it would be too hard for someone who understand driver development to fix it, but it looks like there is nobody interested in maintaining.

Stevemoretz, 2020-05-15:
Oh I get it now, preferences.plist contains stuff about Dns settings and other stuff. Though it's hard to understand how it is implemented.I think that's what you meant.So yeah seems like this should be fixed for Catalina.I remember I couldn't even install it and I used a package from some other guy who hacked the installer so it could be installed on Catalina

EETagent, 2020-05-15:
Hackintosh users don't have to deal with unsigned kexts problems in Catalina. OpenCore bootloader does all the kext injection without even breaking SIP. Sadly Intel Wifi cards are not supported by Apple, so there are 4 options for me
1. Buy compatible Broadcom network card and deal with driver problems in Windows.
2. Buy USB adapter.
3. Wait for Intel Wifi community driver.
4. Use mobile phone for USB tethering.

From my testing HoRNDIS is stable as a rock, but this Enxxxx problem is relatively annoying.

Stevemoretz, 2020-05-15:
I was talking about the package installer that wasn't supported I have a Hackintosh too but on clover.That's not true you can use Intel wifi now with Hackintosh.

https://www.tonymacx86.com/threads/success-working-intel-wifi-drivers-for-7265ac-on-catalina.292207/

You can find two 3 kexts here for Intel wifi, they are in development so not that stable.
Ask for Chinese kext v2 that is pretty stable and work though it won't be developed further unfortunately, itwlm kext which is posted on the page 31 is told that works but has upload problems the kext on page 1 doesn't work very well not stable and crashes but still it works. I'm using the Chinese one it works the only problem is it only connects through the plist ssid not the ui interface.

EETagent, 2020-05-15:
There is probably some way how to fix it.
Based on this comment in similar issue
> In case this might be of some help to some that encounter this issue. The device we were using that exhibited this problem was using ConfigFS to setup the RNDIS USB gadget. The host_addr file, by default, is a generated random MAC address for the host. This seems to be the cause of new records being created by the Mac OS each time. By setting the host_addr contents to a fixed value, new records are no longer created on Mac OS, which then uses the record matching the host_addr contents.

There is different MAC adress for every connection, so macOS creates new interface every time.
There is [old way in Linux ](https://www.monblocnotes.org/node/1895) how to fix that. Searching now how to do that on macOS.

Stevemoretz, 2020-05-15:
That's some good news so if you got the solution let us know too thanks.

EETagent, 2020-05-15:
> I was talking about the package installer that wasn't supported I have a Hackintosh too but on clover.That's not true you can use Intel wifi now with Hackintosh. [...]

Yeah, I am watching these two
https://github.com/zxystd/itlwm
https://github.com/AppleIntelWifi/adapter

They are still not fully stable

Stevemoretz, 2020-05-15:
> Yeah, I am watching these two
> https://github.com/zxystd/itlwm
> https://github.com/AppleIntelWifi/adapter
>
> They are still not fully stable

itlwm seems stable check my link goto the last page and try that it's stable but has upload problems,
https://github.com/AppleIntelWifi/adapter
This one doesn't even work yet, no connection is happening right now,
There is another kext on that topic from some Chinese guy who did 2 updates and then they banned him because he did something illegally but his v2 kext is amazing.I'm using it all days no upload or download problem speed is very good as well.The only problem is it doesn't support gui for connecting.you can only get connected via the plist editing and only to one network you want another one you must restart.And connect to the new one. It's a disgrace he got banned cause he was this close to finish it. :)

Stevemoretz, 2020-07-19:
Anybody is gonna care to fix this?

rdmitry0911, 2020-08-14:
For those who still suffering from this problem. Apple fixed "IOProviderClass" problem in recent updates. At least in 10.15.6 this problem is resolved and thus there is no need for the hack presented in the driver code to fix the problem. So just comment the lines 289-291 in HoRNDIS.cpp, rebuild the kext, put it in oc or clover and you are all set. Reboot and enjoy. For those who have problems rebuilding the kext, I attached the one I built for myself. It works perfectly in 10.15.6 booted from oc 0.6.0
[HoRNDIS.kext.zip](https://github.com/jwise/HoRNDIS/files/5076570/HoRNDIS.kext.zip)

Stevemoretz, 2020-08-16:
> For those who still suffering from this problem. Apple fixed "IOProviderClass" problem in recent updates. At least in 10.15.6 this problem is resolved and thus there is no need for the hack [...]

Will this also work in lower versions? Like 10.15?

rdmitry0911, 2020-08-16:
> Will this also work in lower versions? Like 10.15?

As far as your version has a problem with a stock kext, it should. However, you have to try. I built mine for 10.15 For systems earlier then 10.15 you have to built the kext by yourself. Mine with much probability will not work.

Stevemoretz, 2020-08-16:
> > Will this also work in lower versions? Like 10.15?
> 
> As far as your version has a problem with a stock kext, it should. However, you have to try. I built mine for 10.15 For systems earlier then 10.15 you have to built the kext by yourself. Mine with much probability will not work.

Thanks what command will build a kext?Out of this source code

rdmitry0911, 2020-08-16:
> Thanks what command will build a kext?Out of this source code

I used xcode. Start it and open HoRNDIS.xcodeproj. Then inside xcode open HoRNDIS.cpp and comment lines 289-291. They should look like 
```
//		if (providerClass) {
//			setProperty(kIOProviderClassKey, providerClass);
//		}
```
Then go to Project -> Build. It should build the kext.

Stevemoretz, 2020-08-16:
> I used xcode. Start it and open HoRNDIS.xcodeproj. Then inside xcode open HoRNDIS.cpp and comment lines 289-291. They should look like
>
> ```
> //		if (providerClass) {
> // [...]

Took me sometime because it was my first time building a kext and had some permission problems and also had to change the sdk version from 10.11 to the one I had (10.15).

But guess what genius it worked.And not only it worked but it did also recognize my phone name which didn't before and changed the name to Ethernet Adapter en 60... But now it writes the name of my phone and also I could finally setup static ip with it.

Who could think by removing 3 or 4 lines all these things could happend,obviously not me,and clearly only you.

Thanks you're the best.

closed 2020-08-16 by Stevemoretz