jwise/HoRNDIS #120: Bluetooth tethering support

MyAlexro, 2020-04-25:
Probably a stupid question but, is it possible to use my Android phone with Bluetooth tethering enabled as a Bluetooth card with this driver? If not, could this feature me implemented?

ErrorErrorError, 2020-05-11:
> Probably a stupid question but, is it possible to use my Android phone with Bluetooth tethering enabled as a Bluetooth card with this driver? If not, could this feature be implemented in the future?

You should already be able to bluetooth tethering without HoRNDIS. Just make sure you acc bluetooth PAN on the Network Settings.

MyAlexro, 2020-05-11:
> You should already be able to bluetooth tethering without HoRNDIS. Just make sure you acc bluetooth PAN on the Network Settings.

Ooh wait I didn't explain myself clearly, I used the wrong term. I thought "Bluetooth tethering" meant sharing my phone's Bluetooth with the pc, like if it was a Bluetooth card. Is it possible?

jwise, 2020-08-18:
This issue does not appear to be related to HoRNDIS.  HoRNDIS provides a mechanism to connect to an Android device's network interface.

closed 2020-08-18 by jwise