jwise/HoRNDIS #122: The Installer encountered an error that caused the installation to fail

tiholic, 2020-06-07:
The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”

on MacOS Catalina: 10.15.4

tiholic, 2020-06-07:
Was able to build it by `git grep macosx10.11` and changing all references to my sdk version.

SDK version can be found by running `xcodebuild -showsdks`. Pick the one from `macOS SDKs`

tiholic, 2020-06-07:
However, unable to install kext as it  requires to be signed

tiholic, 2020-06-12:
Thanks for that @jeevatattva !

I couldn't successfully install the kext anyways. Dropped the idea of tethering for now and using a hotspot+wifi

TristanMarion, 2020-07-06:
Hi @tiholic 
Check this issue, it should help you : https://github.com/jwise/HoRNDIS/issues/102