jwise/HoRNDIS #124: Can't install on MacOs Catalina

voquanghoa, 2020-08-06:
Hello, 

I have just download the latest version (9.2) but it can't be installed on my Mac. Could you help me to check this.

[image]

Thanks

eathtespagheti, 2020-08-07:
Same problem here, 10.15.6

eathtespagheti, 2020-08-07:
Ok apparently there's a workaround, a user fixed the install script and solved the compatibility issues with Catalina, you can find it here  [https://github.com/jwise/HoRNDIS/issues/102#issuecomment-541237232]()

voquanghoa, 2020-08-08:
It doesn't work on my side, after disabling csrutil from the recovery mode, I restart my mac and try again but I have the same error. You can see in my screenshot, the csrutil is disabled.

 
[image]

voquanghoa, 2020-08-08:
Try again with file in the comment, it works. Thank you so much

closed 2020-08-08 by voquanghoa