jwise/HoRNDIS #125: Please help how to reinstall hackintosh which requires net so what should I do

raj100s, 2020-08-07:
I have a created a opencore installing pendrive but I don't have ethernet I use usb tethering to use internet on my windows so how should I add this to the opencore pendrive should I add the file and take oc snapshot in config
 please reply

eathtespagheti, 2020-08-07:
This has nothing to do with HoRNDIS

raj100s, 2020-08-08:
But it requires net I don't have ethernet so how should I download it is not supporting usb tethering

raj100s, 2020-08-08:
By reinstall I mean installing hackintosh for first time by reinstall option that i get when I insert my usb containing opensource hackintosh so what changes should I do to this usb drive

jwise, 2020-08-18:
If you are having trouble installing Mac OS, please contact Apple's support.

closed 2020-08-18 by jwise