jwise/HoRNDIS #129: Failed to install on 10.15.6

leon0707, 2020-10-04:
```shell
Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=XXX USER=XXX USERNAME=XXX /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
installer: Package name is HoRNDIS
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
```
Tried to install via brew cask and the pkg file.

Karatla, 2020-10-06:
same error

johndwells, 2020-10-07:
FWIW I'm encountering the same error on 10.15.7.

fgardavaud, 2020-10-07:
Works on 10.15.6 with SIP disable when installing HoRNDIS.
Here we go for the process
- shutdown your mac,
- start your computer in recovery mode : hold Command+R keys on your keyboard and without stop holding press power to start your mac,
- launch terminal with utilities menu,
- in your bash window execute `csrutil disable` command line
- Then reboot your system in normal mode
- launch terminal
- execute `sudo mount -uw /` command line
- install HoRNDIS
- shutdown
- start your computer in recovery mode (hold Command + R keys)
- launch terminal with utilities menu,
- in your bash window execute `csrutil enable` command line
- Reboot and enjoy !

Aleph-Design, 2020-10-19:
@fgardavaud,
Your third step no longer works in my 10.15.7 version. There's simply no Utilities menu available. So there's no terminal and as a consequence no way to issue the command.
Does someone know a workaround?

hibobmaster, 2020-10-24:
> @fgardavaud,
> Your third step no longer works in my 10.15.7 version. There's simply no Utilities menu available. So there's no terminal and as a consequence no way to issue the command.
> Does someone know a workaround?

I am using 10.15.7, but i can disable SIP in recovery mode as what @fgardavaud mensions(I have Utilities on menu~).
Maybe you have to check your system or ask someone for help

Aleph-Design, 2020-10-25:
Thanks for the reply.
Which option do you choose when starting in recovery mode?
When I choose “disk utility” I have the trouble stated.

hibobmaster, 2020-10-25:
> Thanks for the reply. Which option do you choose when starting in recovery mode? When I choose “disk utility” I have the trouble stated.
> […](#)

No special options, just normal recovery mode. When you are in recovery mode, the terminal will live in utilities->terminal which is the third option.
Okay, if you still not see the terminal option, you can disable SIP in this way.
Edit your config.plist  in NVRAM->`7C436110-AB2A-4BBB-A880-FE41995C9F82`->csr-active-config
change the value of csr-active-config into `E7030000` which can be used to disable SIP in boot process.

baskarangappan, 2020-11-12:
Mac Catalina 10.15.7: The same issue was reported long back and it was supposedly fixed in release 9.0. But it is coming again...

==> Downloading https://github.com/jwise/HoRNDIS/releases/download/rel9.2/HoRNDIS-9.2.pkg
Already downloaded: /Users/XYZ/Library/Caches/Homebrew/downloads/eabca1a09c01d4ed6d3818d32f50051a3f3071ceba98f6c8fa9704de0284dfcf--HoRNDIS-9.2.pkg
==> Verifying SHA-256 checksum for Cask 'horndis'.
==> Installing Cask horndis
==> Running installer for horndis; your password may be necessary.
==> Package installers may write to any location; options such as --appdir are ignored.
installer: Package name is HoRNDIS
installer: Upgrading at base path /
installer: The upgrade failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
==> Purging files for version 9.2 of Cask horndis
Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=XYZ USER=bangappa USERNAME=XYZ /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
installer: Package name is HoRNDIS
installer: Upgrading at base path /
installer: The upgrade failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)

But still, luckily the fixed intermediate version is given by d235j in https://github.com/jwise/HoRNDIS/issues/102#issuecomment-541237232 is still working. Hopefully, that intermediate fix will come into the mainstream with PR. Thanks in advance.