jwise/HoRNDIS #130: Link speed capped at 480Mbps

hy1o, 2020-10-28:
Thanks for the driver! I appreciate your work.

I have an RNDIS cable that supports USB3 interface and thus it can support the speed up to 5Gbps.
I'm using it to connect two computers. I'm aware of the fact that this is not the purpose of HoRNDIS but I could use your help.

On Windows side, the Ethernet(over USB) link speed is indeed 5Gbps.
On macOS side, it is capped at 480Mbps (in Network Utility)

I found out that the max speed(480Mbps) is hard-coded in HoRNDIS source code.
`medium = IONetworkMedium::medium(kIOMediumEthernetAuto, 480 * 1000000);`
The maximum speed I could get by playing with this was 705.033 Mbps.

Is there any RNDIS or USB parameters to increase the link speed?

Belcarra, 2020-10-28:
What RNDIS USB-3 cable are you using?

Is your problem displayed speed or the actual measured transfer speed?

hy1o, 2020-11-01:
I'm using Streaming Cable from LattePanda..
https://www.dfrobot.com/product-1851.html
http://docs.lattepanda.com/content/streaming_cable/get_started/

Well, I felt some lagging in remote desktop but the actual file transfer speed was 50MB/s.
I think that exceeds typical, real-life USB 2.0 speed which is, if I recall correctly, around 20MB/s.
(Theoretically it's still under 480Mbps? 50MB*8=400Mbps)
However, USB 3.0 speed in my experience was near 100MB/s so there's some room for improvement.

AlekssZeltins, 2022-03-16:
I would also want to find a way for faster speed.

Usb link speed: Up to 480Mb/s

Which seems are max speeds of USB 2.0

Is there any way to get to speed of USB 3.0 ?