jwise/HoRNDIS #132: macOS 11 Big Sur Support?

krjw-eyev, 2020-11-13:
I am hesitant of upgrading to macOS 11 Big Sur since they have deprecated certain functionalities that seem to be necessary fo HoRNDIS to work.

Here is a list:
https://developer.apple.com/support/kernel-extensions/

Does anyone know if there is a possibility to transition? I have no experiences with kernel extensions.

sfstpala, 2020-11-13:
Can confirm it doesn't work on big Sur :(

krjw-eyev, 2020-11-13:
have you tried with disabling SIP?

sfstpala, 2020-11-13:
Okay wow, that worked. I would prefer having SIP enabled but I will deal with it for now.

(PS I did compile it myself and had Xcode apply a bunch of migrations. I have no idea if that has made a difference.)

d235j, 2020-11-13:
I would expect disabling SIP to be a short-term workaround. Long-term someone would need to rewrite this as a System Extension (using `NetworkingDriverKit`). I don't know if anyone is planning to do that.

Belcarra, 2020-11-13:
The official route forward is a DriverKit extension using
NetworkingDriverKit and USBDriverKit.

It is difficult to get the appropriate USB entitlements for anything other
than your own vendor ID (i.e. one that is owned by the organization that is
applying for the entitlement.)

USB device matching is limited to the vendor ID that is in your
entitlements. Class only matching will not work (i.e. match USB Class,
Sub-Class and Protocol).

d235j, 2020-11-13:
@Belcarra Ugh. This is the type of situation where Apple should either implement the standard officially, or provide an exception to that rule. Has anyone made a request for Apple to add RNDIS support via Feedback Assistant?

cpboyd, 2020-11-14:
> @Belcarra Ugh. This is the type of situation where Apple should either implement the standard officially, or provide an exception to that rule. Has anyone made a request for Apple to add RNDIS support via Feedback Assistant?

That seems extremely unlikely as RNDIS is a proprietary Microsoft protocol. Microsoft might want some sort of licensing scheme if Apple were to try to include it with macOS.

Further, as the primary use-case is wired internet access via Android devices, Apple's likely to simply suggest you buy an iPhone.

RNDIS is very finicky. I tried getting this driver working with a Windows CE device a few years back to no avail. (While it did work for my Android phone.)
Not all RNDIS devices work in the exact same way and it's not clearly documented by Microsoft.

d235j, 2020-11-14:
Apple did add support for CDC-ECM to macOS, which some Android phones have. Ideally, Android phones would support both RNDIS for Windows hosts and CDC-ECM for non-Windows hosts, but that is not the case.

> Further, as the primary use-case is wired internet access via Android devices, Apple's likely to simply suggest you buy an iPhone.

Perhaps, but Apple is interested in minimizing the need for third party drivers. For instance, they have added built-in support for all major USB-serial devices.

The lack of RNDIS standardization is probably a bigger problem. That said, this driver has plenty of bugs anyway.

cpboyd, 2020-11-14:
> Apple did add support for CDC-ECM to macOS, which some Android phones have. Ideally, Android phones would support both RNDIS for Windows hosts and CDC-ECM for non-Windows hosts, but that is not the case.

Not all Android devices support RNDIS. It appears to be a manufacturer decision: https://en.wikipedia.org/wiki/RNDIS
Yes, ideally manufacturers would support more than just Windows.

I believe Apple even allows iPhones to enumerate as a network device on Windows machines if you've installed iTunes and their drivers.

> Perhaps, but Apple is interested in minimizing the need for third party drivers. For instance, they have added built-in support for all major USB-serial devices.

USB serial is standardized. Even Windows 10 provides open serial drivers for any device.
For RNDIS, even on Windows, devices have to enumerate properly to automatically (per Microsoft's custom enumeration protocols) have RNDIS support without third-party drivers.

> The lack of RNDIS standardization is probably a bigger problem. That said, this driver has plenty of bugs anyway.

There is no standard. It's a proprietary implementation.

jwise, 2020-11-14:
> > The lack of RNDIS standardization is probably a bigger problem. That said, this driver has plenty of bugs anyway.
> 
> There is no standard. It's Microsoft's proprietary implementation of CDC. If anything, Microsoft should support the open CDC standards.

https://winprotocoldoc.blob.core.windows.net/productionwindowsarchives/WinArchive/%5BMS-RNDIS%5D.pdf

https://docs.microsoft.com/en-us/windows-hardware/drivers/network/remote-ndis-to-usb-mapping

jwise, 2020-11-14:
> The official route forward is a DriverKit extension using NetworkingDriverKit and USBDriverKit. It is difficult to get the appropriate USB entitlements for anything other than your own vendor ID [...]

Yes, this is what I was thinking, though I was thinking that the answer would be to use `libusb` and NetworkingDriverKit (which might be a reasonable workaround for the USBDriverKit matching trouble).

That said, I don't even have hardware that's capable of running Big Sur, and I have no plans to upgrade past Mojave (and my primary machine is a Linux machine these days, anyway...).  So if someone else wants to do this, they are welcome to do so, but I'm not convinced that you'd even get to reuse much code...

I agree that Apple really ought to support this natively.

bisaalz, 2020-11-15:
Horndis is working fine on macOS big sur final version 
1. make sure you have working ethernet kext such as rtl8111 kext, intelmausi kext or other compatible kext
2. https://www.icloud.com/iclouddrive/0jlcfKZP8E_6g64mhb86iSg_g#horndis download horndis.kext from here
3. copy it into oc/kext folder
4. oc snapshot it with propertree 
5. restart pc.
6. in android enable usb debugging in developer option
7. connect wifi in android and enable usb tethering while connecting it from usb cable.

androidmalin, 2020-11-16:
@bisaalz I have a question. Where is this directory "oc/kext" ?

bisaalz, 2020-11-17:
> @bisaalz I have a question. Where is this directory "oc/kext" ?
> The file you provided is a compressed package. This package has two file HoRNDIS-9.2.pkg and HoRNDIS.kext.

oh sorry this for hackintosh only on real mac i don't think it will work on big sur Because i have tested by putting hordis.kext on library/extension and rebuild the cache of library/extension  and it didn't work but in catalina it worked. if u have catalina with real mac https://www.youtube.com/watch?v=lzbBfAF1rrI follow this guide. This method didn't work for me on big sur.
   if you have hackintosh put horndis.kext file in opencore bootloader inside Kext folder of OC folder. like OC/Kext/horndis.kext and oc snapshot it with propertree software https://github.com/corpnewt/ProperTree..

androidmalin, 2020-11-17:
@bisaalz thank you very much, I try it, It didn't work in mac os big sur.

ben-xD, 2020-11-18:
Excellent, thanks @bisaalz. I added it to my hackintosh and its working beautifully.

sfatih, 2020-11-21:
Hey guys,
I give feedback about the issue to Apple and the answer was:

[image]

For the SIP-enabled usage, they recommend kext notarization: https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution

ForceGT, 2020-11-22:
If anyone had it working, kindly leave a link to an executable or something

ForceGT, 2020-11-22:
[This](https://github.com/OpenIntelWireless) could be an alternative to people looking for a wireless solution without changing the wifi card

Roboromeo1, 2020-11-23:
disabling sip and running this command worked for me with big sur 

sudo kextload /Library/Extensions/HoRNDIS.kext

i'm using 2015 macbook pro i5 8g 512gb ssd

Kavithacypress, 2020-11-26:
> disabling sip and running this command worked for me with big sur
> 
> sudo kextload /Library/Extensions/HoRNDIS.kext
> 
> i'm using 2015 macbook pro i5 8g 512gb ssd

@Roboromeo I have tried but it tells me 

`Extension with identifiers com.joshuawise.kexts.HoRNDIS not approved to load. Please approve using System Preferences.` and nothing in System Preferences... I have disabled SIP by the way.

krjw-eyev, 2020-11-27:
@Kavithacypress I think you need to go to `Settings -> Security & Privacy` and then the tab `General`. In the lower half of the view you should be able to allow loading the kext. This appears only for a couple of minutes after you tried to load the kext.

Velolong, 2020-12-04:
Same probleme here. I tried to install Horndis on Big Sur (real Mac, not Hackintosh), with the following lines in terminal (recovery mode and normal mode)
- csrutil disable
- spctl kext-consent disable
- sudo spctl --master-disable
But it always ends with the kext being installed in library/preferences but not being loaded or activated. and the I enter :
- sudo kextload /Library/Extensions/HoRNDIS.kext
I get the same message "Extension with identifiers com.joshuawise.kexts.HoRNDIS not approved to load. Please approve using System Preferences" but nothing to allow in System preferences > Security and privacy > general. The button "allow" juste does not appear....
Any clues ?

ForceGT, 2020-12-04:
The maintainer already said  that this won't be - officially supported

Velolong, 2020-12-04:
Yes, I understand, but for Catalina, some found a workaround, I thought maybe some one could advise a way to do. My only issue here is how to get the "allow" button to appear. It seems to be a bug in Big Sur rather than a incompatibility of Horndis.

Or maybe some one has an advice for an USB tethering solution from android to Mac ?

Roboromeo1, 2020-12-04:
I got it working in my Big sur update , I have macbook book pro i5 , 512 gb ssd  early 2015 .  First thing i did is went to recovery mode , terminal  csrutil disable i.e i disabled the sip and then install hornidis it wont open . go to system preferences and allow horndis installer then again click hornidis install and system preferences allow installer and it should install. then restart mac once and it worked and then after install i again  enabled my sip and it still works :)

Velolong, 2020-12-04:
@Roboromeo1 

"go to system preferences and allow horndis installer" The problem is that I don't have any button or whatever allowing me to _allow_ neither the install nor the kext in the System Preferences Panel. I go there, "Security", under "general" tab, but nothing. I would really like to allow, but I can't :D

Roboromeo1, 2020-12-05:
@Velolong not sure whats the issue on your side it works fine for me , after installation i enabled sip and still it works . Have attached a screen shot below for your reference . 
[image]

Prophetia086, 2020-12-07:
There is a way to install AX88179 Drivers for Big Sur 11.0, and which did work in may mac is this:

"
Revised instructions to install AX88179-179A USB-Ethernet driver on Mac OS 11 (Big Sur)
Thanks for all who confirmed these instructions actually works even with scenarios that one find himself stuck like after re-enabling SIP using csrutil enable in recovery mode. Point 12 solves this particular issue. However, everybody should follow all these instructions including point 12. Thanks volodymyrk for correction.

Follow these instructions carefully:
1- Unplug your USB-Ethernet device.
2- Get the latest version and discard all advices about using older versions. Get version 2.19 (beta2) which is currently available. Do NOT install driver now !
3- Uninstall old driver first using the uninstaller that comes with the latest driver. You will have to restart.
4- Wait your screen to be off due to Restart, then directly... (next point)
5- Keep holding Command+R wile your PC is booting. Keep holding until the Apple logo appears. (If you fail to enter recovery mode, shut down and repeat this point).
6- Select your account and enter password.
7- Choose Utilities > Terminal.
8- csrutil disable
9- /usr/sbin/spctl kext-consent list
10- If 5RHFAZ9D4P is not in the list, enter: /usr/sbin/spctl kext-consent add 5RHFAZ9D4P . Otherwise, skip this point.
11- Restart you mac.
12- This is the critical point which you might miss from all other instructions available: Remove Allow permission you provided earlier to asix team (5RHFAZ9D4P). This is important to let your OS accept the device again. Follow these steps carefully:
sudo su
(enter your password if prompted), then enter these four lines by sequence:

sqlite3 /var/db/SystemPolicyConfiguration/KextPolicy
--
delete from kext_load_history_v3 where team_id='5RHFAZ9D4P';
delete from kext_policy where team_id='5RHFAZ9D4P';
.quit

13- Install driver.
14- Now you will be prompt to Allow this driver in System Preferences -> Security & Privacy.
15- Connect your USB-Ethernet device.
16- Your USB-Ethernet device should work by now. Ensure you have ethernet connected to router/Switch to verify.
17- If it does not yet work, try running this in terminal: sudo kextload -b com.asix.driver.ax88179-178a
"
from here: https://developer.apple.com/forums/thread/651132?page=2

When I delete the date in kext_load_history_v3 and kext_policy, I can see the Allow button again.

But I don't know what does 5RHFAZ9D4P mean, and what code it should be in HoRNDIS.

I hope it can help someone here, and get a better way to do.

Velolong, 2020-12-08:
@Prophetia086 

Thanks a lot for the tip! It may be helping indeed! I'll try it and tell you.

Velolong, 2020-12-14:
Unfortunately, I can't make it work, since I have no such a file  /usr/sbin/spctl kext-consent list....

I'm stuck. The USB tethering is very important for my daily work, and I can't find a way to make it work on Big Sur.... It always end with "Extension with identifiers com.joshuawise.kexts.HoRNDIS not approved to load. Please approve using System Preferences" with nothing to allow on Pref Panel. It's really annoying....

Any clue, anyone ? Please ? :D

Velolong, 2020-12-15:
I come back with good news: I finally made it work ! Thank a lot to Prophetia086 and to others tips I collected [here](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-739660615), [here](https://github.com/jwise/HoRNDIS/issues/102#issuecomment-540072360)  and [there](https://eclecticlight.co/2019/06/01/how-to-bypass-mojave-10-14-5s-new-kext-security/) to make it work. Here is my path :
> First, we need to get the developer ID and check whether that extension has been notarized. One easy way to do this is to type most of the command
	•	spctl -a -vv -t install
then drag and drop the extension from your Downloads folder to the end of that line, where its path and name should appear, e.g.  /Users/Downloads/Horndis.kext
Then press Return, and you should see three lines of response: 
	•	mykext.kext: accepted
	•	source=Developer ID
	•	origin=Developer ID Application: DeveloperName (54GTJ2AU36)
If the extension is notarized already, they will instead look like 
	•	mykext.kext: accepted
	•	source=Notarized Developer ID
	•	origin=Developer ID Application: DeveloperName (54GTJ2AU36)
Make a note on paper or your iOS device of the developer ID provided in parentheses, as you’ll need those in a few moments. 

> Close your apps down and restart your Mac in Recovery mode. There, open Terminal and type in the command
	•	csrutil disable
	•	/usr/sbin/spctl kext-consent list
If 54GTJ2AU36 is not in the list, enter
	•	/usr/sbin/spctl kext-consent add 54GTJ2AU36.
Press Return, wait for the command prompt to appear again, then quit Terminal and restart in normal mode.

> Remove Allow permission you provided earlier to Horndis (54GTJ2AU36). This is important to let your OS accept the device again. Follow these steps carefully:
	•	sudo su
(enter your password if prompted), then enter these four lines by sequence:
	•	sqlite3 /var/db/SystemPolicyConfiguration/KextPolicy
	•	delete from kext_load_history_v3 where team_id='54GTJ2AU36';
	•	delete from kext_policy where team_id='54GTJ2AU36';
	•	.quit

> Manually erase any previous horndis.kext remained in library/extensions (just to be sure). Install [Horndis 9.2 Catalina Install pkg](https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip)). Now you will be prompt to Allow this driver in System Preferences -> Security & Privacy.

> Restart in recovery mode to enable csrutil
	•	scrutin enable

> Restart in normal mode and enjoy it working (at least, for me it did work)

I am so happy :D I hope this walkthrough will help others people with the same issue on Big Sur.

Prophetia086, 2020-12-17:
@Velolong You really did a great job! I made my mac work by following your steps in just a quarter. Thanks a lot!

Velolong, 2020-12-17:
:)

Velolong, 2020-12-17:
:)

VKrau, 2020-12-21:
@Velolong Thank you! It really works on Big Sur 11.1

Velolong, 2020-12-21:
You're welcome ;)

Malik056, 2020-12-31:
csrutil not found in recovery mode

tibeemts, 2021-01-03:
> I come back with good news: I finally made it work ! Thank a lot to Prophetia086 and to others tips I collected [here](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-739660615), [here](https://github.com/jwise/HoRNDIS/issues/102#issuecomment-540072360) and [there](https://eclecticlight.co/2019/06/01/how-to-bypass-mojave-10-14-5s-new-kext-security/) to make it work. Here is my path :
> 
> > First, we need to get the developer ID and check whether that extension has been notarized. One easy way to do this is to type most of the command
> > •	spctl -a -vv -t install
> > then drag and drop the extension from your Downloads folder to the end of that line, where its path and name should appear, e.g.  /Users/Downloads/Horndis.kext
> > Then press Return, and you should see three lines of response:
> > •	mykext.kext: accepted
> > •	source=Developer ID
> > •	origin=Developer ID Application: DeveloperName (54GTJ2AU36)
> > If the extension is notarized already, they will instead look like
> > •	mykext.kext: accepted
> > •	source=Notarized Developer ID
> > •	origin=Developer ID Application: DeveloperName (54GTJ2AU36)
> > Make a note on paper or your iOS device of the developer ID provided in parentheses, as you’ll need those in a few moments.
> 
> > Close your apps down and restart your Mac in Recovery mode. There, open Terminal and type in the command
> > •	csrutil disable
> > •	/usr/sbin/spctl kext-consent list
> > If 54GTJ2AU36 is not in the list, enter
> > •	/usr/sbin/spctl kext-consent add 54GTJ2AU36.
> > Press Return, wait for the command prompt to appear again, then quit Terminal and restart in normal mode.
> 
> > Remove Allow permission you provided earlier to Horndis (54GTJ2AU36). This is important to let your OS accept the device again. Follow these steps carefully:
> > •	sudo su
> > (enter your password if prompted), then enter these four lines by sequence:
> > •	sqlite3 /var/db/SystemPolicyConfiguration/KextPolicy
> > •	delete from kext_load_history_v3 where team_id='54GTJ2AU36';
> > •	delete from kext_policy where team_id='54GTJ2AU36';
> > •	.quit
> 
> > Manually erase any previous horndis.kext remained in library/extensions (just to be sure). Install [Horndis 9.2 Catalina Install pkg](https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip)). Now you will be prompt to Allow this driver in System Preferences -> Security & Privacy.
> 
> > Restart in recovery mode to enable csrutil
> > •	csrutil enable
> 
> > Restart in normal mode and enjoy it working (at least, for me it did work)
> 
> I am so happy :D I hope this walkthrough will help others people with the same issue on Big Sur. Yes, I almost forgot : I upgraded to Big Sur 11.1 before doing this whole process. It may have be helping.

This is totally work on Big Sur with my Mi10T pro. Thank you guy

Velolong, 2021-01-03:
@tibeemts Glad it helped you!
@Malik056 What do you mean by "not found" ? If you hav Big Sur, csrutil is automatically enabled. You did try to launch the command via the recovery mode, right ?

rreitz, 2021-01-12:
While executing the step:
sqlite3 /var/db/SystemPolicyConfiguration/KextPolicy
• delete from kext_load_history_v3 where team_id='54GTJ2AU36'
...
I get:
sqlite> delete from kext_load_history_v3 where team_id='54GTJ2AU36';
Error: attempt to write a readonly database

It looks like another process is using the database:
zippy:~ $ sudo lsof /var/db/SystemPolicyConfiguration/KextPolicy
Password:
COMMAND   PID USER   FD   TYPE DEVICE SIZE/OFF   NODE NAME
syspolicy 249 root    4u   REG    1,5    53248 742717 /private/var/db/SystemPolicyConfiguration/KextPolicy
zippy:~ $ 

Do I kill 'syspolicy'?
Thanks.

Velolong, 2021-01-12:
@rreitz I am not sure this error is linked to another process using the database. With " attempt to write a readonly database", it looks like csrutil is still activated. 
But I am no expert in terminal or coding, maybe someone wiser would be able to advise you.

Prophetia086, 2021-01-12:
@rreitz The error "attempt to write a readonly database", maybe you haven't disable the SIP yet, I guess .

rreitz, 2021-01-12:
Yes, I neglected to disable SIP.  I have installed HoRNDIS on Big Sur.
Thanks for you response.

Prophetia086, 2021-01-13:
@rreitz You're welcome ：）

ForceGT, 2021-01-18:
Just updated to Big Sur today and can confirm that it works seamlessly as before

Petchav, 2021-03-29:
Hello,

Personally, I still have problems, the HoRNDIS driver does not work with macOS Big Sur 11.2 or 11.3 (beta).
I tried @Prophetia086's solution and to disable SIP as in the Apple documentation:
https://developer.apple.com/documentation/security/disabling_and_enabling_system_integrity_protection

But it doesn't work for me

luxagroove, 2021-03-31:
Hi, 

after trying all the ways mentioned above, I found one, which is much easier and faster.
Just downloaded the DJI Assistant 2 for Mavic from the DJI support website (https://www.dji.com/de/downloads/softwares/assistant-dji-2-for-mavic) and ran the installer. 
This installer included the HoRNDIS driver, and no jumping around with enable/disable csrutil and Recovery Mode needed. Just one normal reboot needed. 
And it works for me now.

luxagroove, 2021-03-31:
Hmm, yes, I am using an Intel Mac. 
So what is not working for you. Does the installation of the DJI Assistant app fails? 
Which version are you trying to install?

wglas85, 2021-04-05:
> Yes, this is what I was thinking, though I was thinking that the answer would be to use `libusb` and NetworkingDriverKit (which might be a reasonable workaround for the USBDriverKit matching [...]

Hello @jwise,

  do you have an active developer group on developer.apple.com and might it be possible to add me to this group, so we might try develop a NetworkingDriverKit extensionen, which might replace HoRNDIS?

  Best Regards, Wolfgang

Emilien-Lambert-Rio, 2021-04-06:
@luxagroove Hi sorry to reply after so long, the driver installation works because the DJI Assistant application starts well, but during the installation there is no HoRNDIS validation step in Security and Privacy. So, I think the problem is with the HoRNDIS driver. If it works with Big Sur and intel chip, then it means that the problem is probably related to the M1

anafaggiano, 2021-04-08:
@luxagroove Hello,

I am on a MacBook Pro 2018 Intel i7.

I downloaded and ran the installer for DJI2, following the link you provided. It installed correctly and when prompted, I enabled the permissions for a joshua wise app (I think it even said horndis) in my System Preferences. So far, so good.
I restarted my computer and... no luck: I plugged in my Android phone via USB, enabled `USB tethering` in my phone options and it never shows up in the `Network` window in the `System Preferences`.

In doubt, I unplugged my phone, restarted my computer a second time, re-plugged it and re-enabled usb tethering. I listed my drivers `/System/Library/Extensions ls | grep ".kext"`, also tried with `grep -i "horndis"`, I can't seem to find the horndis driver.

Last, I tried moving the mavic Application to trash and re-installing it, but I never got prompted for the permissions again (I guess that makes sense). Any advice ?

brendangibat, 2021-04-30:
To chime in, what worked for me was to install then reboot:
https://www.dji.com/downloads/softwares/assistant-dji-2-for-mavic

Macbook Pro (16-inch, 2019) (Intel)
macOS Big Sur 11.3

Thanks for the suggestion @luxagroove !

gorquan, 2021-05-09:
working!
macOS Big Sur 11.3.1

amanjain08, 2021-05-14:
Doesn't work for me.

Download and installed successfully DJI Assistant 2 For Mavic Mac V2.0.14
https://www.dji.com/downloads/softwares/assistant-dji-2-for-mavic

macOS Big Sur 11.2.3.

Joll123, 2021-06-24:
@amanjain08 Have you solved it? DJI Assistant 2 For Mavic Mac V2.0.14 does't work for me too.

amanjain08, 2021-06-24:
@Joll123 No, I haven't as I made peace with the WIFI Hotspot instead of tethering.

Joll123, 2021-06-26:
> I come back with good news: I finally made it work ! Thank a lot to Prophetia086 and to others tips I collected [here](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-739660615), [...]

Have you updated system to macOS Big Sur 11.4? It doesn't work for  me.

krjw-eyev, 2021-06-26:
@Joll123 I used Velolongs steps [here](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-745607263) to install it on macOS 11.4.

martin-braun, 2021-06-27:
Could someone please tell me how to get the horndis.kext file, directly? I was looking into the pkg with the Archive Explorer, but the file is missing.

[27 more replies]