jwise/HoRNDIS #133: Installation failed on MacOS 10.15.7 with installer 9.2

cmosetick, 2020-11-27:
MacOS:
`10.15.7 Catalina`

```uname -a
Darwin 19.6.0 Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64 x86_64

Installation via `brew cask`

```
==> Verifying SHA-256 checksum for Cask 'horndis'.
==> Installing Cask horndis
==> Running installer for horndis; your password may be necessary.
==> Package installers may write to any location; options such as --appdir are ignored.
installer: Package name is HoRNDIS
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
==> Purging files for version 9.2 of Cask horndis
Error: Failure while executing; `/usr/bin/sudo -E -- /usr/bin/env LOGNAME=chris USER=chris USERNAME=chris /usr/sbin/installer -pkg /usr/local/Caskroom/horndis/9.2/HoRNDIS-9.2.pkg -target /` exited with 1. Here's the output:
installer: Package name is HoRNDIS
installer: Installing at base path /
installer: The install failed. (The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. An error occurred while running scripts from the package “HoRNDIS-9.2.pkg”.)
```

cmosetick, 2020-11-27:
Note that the v9.2 GUI .pkg installer from here, also fails:
http://joshuawise.com/downloads/HoRNDIS-9.2.pkg

imekachi, 2020-11-30:
Try this one
https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip

cmosetick, 2020-12-03:
@imekachi The `HoRNDIS-9.2 2.pkg` driver installer you linked me to seems to complete / install successfully.

However, After rebooting, the Mac Catalina system, never shows that the Android USB is functioning, nor does it actually function.

Also, after rebooting the Mac Catalina system,`kextstat` I do not notice any "android" related usb kexts. So not sure what is actually getting "installed".

The Android device, does notice when the USB-C cable is connected and disconnected to the Mac Catalina system.
FYI - in case anyone is curious, the USB-C cable I am testing with is an authentic Apple manufactured USB-C to USB-C cable.

The Android phone is running Linux kernel version `4.14.141-19064881`

[image]

[image]

```
➜  ~  kextstat | grep -i usb
   19    4 0xffffff7f80dd2000 0x4000     0x4000     com.apple.driver.AppleUSBHostMergeProperties (1.2) 7D9D2E98-5B93-3287-89FB-16CBD09E351A <5 3 1>
   20    9 0xffffff7f80dd6000 0xe000     0xe000     com.apple.driver.usb.AppleUSBCommon (1.0) B5515CE4-E548-3AEE-9980-1967FB191C23 <6 5 3 1>
   21   21 0xffffff7f80de4000 0xfa000    0xfa000    com.apple.iokit.IOUSBHostFamily (1.2) 80AD6206-0EDB-39B5-A7B9-8FEAFD6407F3 <20 19 18 8 7 6 5 3 1>
   24    1 0xffffff7f8130a000 0x34000    0x34000    com.apple.iokit.IOUSBMassStorageDriver (157.140.1) 5452090C-D1C9-3280-B382-390A6EB4C586 <23 22 21 13 7 6 5 3 1>
   42    2 0xffffff7f81456000 0x15000    0x15000    com.apple.driver.AppleUSBTDM (489.120.1) F998617F-7AC6-3E47-A78D-5176C342E7C3 <41 36 24 23 22 21 8 7 6 5 3 1>
   62    1 0xffffff7f839a5000 0x58000    0x58000    com.apple.driver.usb.AppleUSBXHCI (1.2) D877721D-A4F1-3271-B4E2-C3668D189793 <21 20 12 8 7 6 5 3 1>
   63    0 0xffffff7f83c5b000 0x35000    0x35000    com.apple.driver.usb.AppleUSBXHCIPCI (1.2) D5238DED-D891-3BED-A4B2-8FF9E58629DC <62 21 20 13 12 8 7 6 5 3 1>
   64    2 0xffffff7f818c1000 0x99000    0x99000    com.apple.iokit.IOUSBFamily (900.4.2) 15879BD6-8025-376B-9085-F56C7C002FB1 <21 20 13 8 6 5 3 1>
   79    3 0xffffff7f8197e000 0x9000     0x9000     com.apple.driver.usb.AppleUSBVHCICommon (1.0) 1D932E76-DA61-38A3-A469-C70C4E85E71F <21 20 8 6 5 3 1>
   80    1 0xffffff7f81987000 0xd000     0xd000     com.apple.driver.usb.AppleUSBVHCICommonBCE (1.0) DFEDBF9F-7518-3A7D-A0AB-68E6622E181C <79 76 20 8 6 5 3 1>
   81    1 0xffffff7f81994000 0x28000    0x28000    com.apple.driver.usb.AppleUSBVHCI (1.2) F1A77490-176E-3435-8A7F-0A82B698D119 <79 21 20 8 7 6 5 3 1>
   82    0 0xffffff7f819c0000 0x4000     0x4000     com.apple.driver.usb.AppleUSBVHCIBCE (1.2) 2AAF6148-F34C-3B26-B7BF-A5AD4DE696B5 <81 80 79 76 21 20 12 8 7 6 5 3 1>
  106    4 0xffffff7f80eee000 0x8000     0x8000     com.apple.driver.usb.AppleUSBHostCompositeDevice (1.2) 813575C0-65CA-36E1-853F-4BD44F81935F <21 20 7 6 5 3 1>
  107    4 0xffffff7f81b8b000 0x8000     0x8000     com.apple.driver.usb.networking (5.0.0) 9A201177-A20D-3145-A4AB-A2B9E0EAAEF1 <21 7 6 5 3 1>
  108    3 0xffffff7f82a95000 0x8000     0x8000     com.apple.driver.usb.cdc (5.0.0) 7D401B3D-415A-3D8E-92AA-1172B49CF325 <107 106 21 6 5 3 1>
  110    0 0xffffff7f82a9d000 0x11000    0x11000    com.apple.driver.usb.cdc.ncm (5.0.0) 309B5FF5-631E-3294-B18B-68BE9FEF897D <108 107 106 32 21 17 7 6 5 3 1>
  186    1 0xffffff7f83915000 0x3000     0x3000     com.apple.iokit.IOUSBUserClient (900.4.2) AA958E1D-D020-33B4-A915-C1633FF7A135 <64 19 8 6 5 3 1>
  187    0 0xffffff7f8777d000 0x8000     0x8000     org.virtualbox.kext.VBoxUSB (6.1.16) CA33FA52-5933-3F0E-9B40-8B1ECF4D6A36 <186 185 64 8 6 5 3 1>
  191    0 0xffffff7f87790000 0x6a000    0x6a000    com.apple.driver.AppleUSBAudio (323.4) 0E8068B1-3B7D-3D93-A17B-8BBBD71A06AF <106 87 21 19 13 6 5 3 1>
  192    1 0xffffff7f842ec000 0x6000     0x6000     com.apple.driver.usb.serial (6.0.0) 2CD7A00D-7FE0-32B6-8356-0282E8A56108 <85 21 6 5 3 1>
  193    0 0xffffff7f84aa0000 0x8000     0x8000     com.apple.driver.usb.cdc.acm (5.0.0) 3AE15D79-8672-372E-AB15-D1C7B751B09C <192 108 107 106 85 21 6 5 3 1>
  194    0 0xffffff7f84aa8000 0xc000     0xc000     com.apple.driver.usb.cdc.ecm (5.0.0) E7034D01-BB4C-3148-99BD-3904CE5DD589 <108 107 32 21 19 6 5 3 1>
```

ninikolov, 2020-12-24:
Hello, I am having the same problem -- installed the package successfully, but the Ethernet option in the networking remains red when I turn on the USB tethering. I'm testing with a OnePlus 6 phone and the same Mac version. Any update on this? 

Thanks!

iahmedbacha, 2020-12-27:
> Try this one
> https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip

Thanks, this worked for me.
Afterwards, the phone network didn't appear in the list of networks. I went to "Preferences->Security & Privacy" and allowed jwise program. That solved the issue.

cmosetick, 2021-02-10:
FYI - I ran more tests on my MacOS 10.15.7 device recently trying to get this to work.
HoRNDIS does still not work for me.
I will test on a MacOS 11.1 BigSur device at some point, but not very hopeful that it will work.

In case I did not already mention, this same Android Device with Linux `4.14.141-19064881` lets a vanilla Linux PC connect to the Android WiFi via tether immediately, with no issues hesitation or special software installed.
All I have to do in Android is turn on the share internet switch in settings.
So I am quite baffled by the issues MacOS has with something that seems so simple in my opinion in 2020/2021.

cmosetick, 2021-02-23:
FYI - I just discovered that on MacOS 10.15.7 device this package/driver shows up in system information report as **"Disabled Software."**
And I have no clue why it shows up there, as I consented for the installation!
See screenshot below for more information:

[image]

ryanjAA, 2022-04-12:
Worked for me on 10.15.3. Try disabling SIP. I installed it from the above link and rebooted. Nothing else (already had SIP disabled when I installed it).

Leebgit, 2024-08-05:
Worked for me on 10.15.7. Had to allow software from Joshua Wise during the middle of the installation and the installation took quite a long time, but other than that it worked quite well.

Leebgit, 2024-08-05:
> Try this one https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip

This version of HoRNDIS works on MacOS Catalina 10.15.17. During the installation I had to open Security Privacy to allow installation of software by Joshua Wise, but aside from taking longer than normal to install things worked perfectly. No reboot was required - it just worked. In my Network Settings 3 new Ethernet Adaptor (en4, en5, en6) appeared with en6 connected to my Android phone. I had to unplug my phone and then plug it back in again to pull up the screen on Android setting how the phone was connected to the PC. I chose USB Tethering and everything works. The original HoRNDIS 9.2 did not work.