jwise/HoRNDIS #135: M1 support?

shlowe, 2020-12-22:


mark-netalico, 2020-12-25:
In my testing of M1 MacBook Air HoRNDIS does not support M1 Macs right now.

xianfei, 2020-12-27:
I can build it for Apple M1, but I don't know how to sign it. 'Kext signature validation error code -67050'

mark-netalico, 2020-12-27:
https://developer.apple.com/library/archive/technotes/tn2206/_index.html#//apple_ref/doc/uid/DTS40007919-CH1-TNTAG19

caffeinum, 2020-12-31:
DJI Assistant software depends on this kext, so even though I don't need Android tethering per se, I can't use the software to access files from my drone. Will love to see a fix, too

noc2spam, 2021-01-14:
I don't think it's working in M1, not atleast in my Mac Mini M1. I get "Incompatible architecture: Binary is for x86_64, but needed arch arm64e". Did somebody manage to install a version on the new mac mini?

mark-netalico, 2021-01-14:
Kind of crazy that Android USB connections rely on this open-source project. Google or Apple should have their 1st party own solution.

zinwalin, 2021-03-05:
same issue here.  I can compile from Xcode, but failed to load it because of signing errors.

➜  horndis git:(master) ✗ brew install --cask horndis

Updating Homebrew...
==> Caveats
horndis requires a kernel extension to work.
If the installation fails, retry after you enable it in:
  System Preferences → Security & Privacy → General

For more information, refer to vendor documentation or this Apple Technical Note:
  https://developer.apple.com/library/content/technotes/tn2459/_index.html

Error: macOS El Capitan, Sierra, High Sierra or Mojave is required for this software.

chaealarm, 2021-03-05:
I succeed without kext signing on M1

( For Korean, check this page below )
 https://itap.tistory.com/2 

Open terminal in recovery mode, type
csrutil enable --without kext

after, reboot

build kext, paste to Library/Extensions and repair permission and install(open terminal and type)
sudo chown -R root:wheel /Library/Extensions/horndis.kext
sudo chmod -R 755 /Library/Extensions/horndis.kext       
sudo kextload -t /Library/Extensions/horndis.kext

after, open settings-security, and install button will be added. 
Install it and reboot

Done!

chaealarm, 2021-03-06:
> I succeed without kext signing on M1
>
> ( For Korean, check this page below )
> https://itap.tistory.com/2
>
> Open terminal in recovery mode, type
> csrutil enable --without kext
>
> after, reboot
>
> build kext, [...]

Somebody Have an KEXT sign-able Apple developer account?
HoRNDIS have to be rebuilt(for Universal or arm64) and signed.

caly8464, 2021-03-14:
When I try to do kextload -t .....   it appears the next text.

Executing: /usr/bin/kmutil load -p /Library/Extensions/horndis.kext
Error Domain=KMErrorDomain Code=71 "Incompatible architecture: Binary is for x86_64, but needed arch arm64e" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for x86_64, but needed arch arm64e}

chaealarm, 2021-03-15:
> When I try to do kextload -t ..... it appears the next text.
>
> Executing: /usr/bin/kmutil load -p /Library/Extensions/horndis.kext
> Error Domain=KMErrorDomain Code=71 "Incompatible architecture: [...]

You should build that code for Universial(x64, arm64e) in Xcode.

dave-moriarti, 2021-04-05:
> I succeed without kext signing on M1
>
> ( For Korean, check this page below )
> https://itap.tistory.com/2
>
> Open terminal in recovery mode, type
> csrutil enable --without kext
>
> after, reboot
>
> build kext, [...]

Hello there, thanks for your instructions, are they working fine for you?
I'm a total noob when it comes to programming, but i really need HoRNDIS to work on my M1 MacBook.

I tried to follow your step again and again but I couldn't get it done (I'm struggling with the cloning and building passages, in particular)
Could you be so kind to write something more of a step-by-step guide for us poor beginners ? You'd save my day and I'm sure it would be useful to many others as well!

I'm sorry if this post it's inappropriate, it's my first one here on GitHub.

tvqt, 2021-04-15:
@dave-moriarti the Korean website has a build that you can use- no need to build it yourself (I think)!

dave-moriarti, 2021-04-15:
@tvqt Thank you for your response! Already tried to, but failed. Can't understand what i'm doing wrong, i followed the steps in the Korean site but i'm probably missing something

tvqt, 2021-04-15:
@dave-moriarti the process for me was:

1. downloading the file from the Korean site (https://blog.kakaocdn.net/dn/nRMG3/btqZkqGNzgR/U1XLxka8DvDA5zgx4X1yY0/HoRNDIS.kext.zip?attach=1&knm=tfile.zip), unzipping it, and pasting the HoRNDIS.kext file it into the ~/Library/Extensions folder

2. booting into recovery, opening terminal, and running the lines the Korean site tells you to do (this threw an error for me, and after restarting the computer, my Mac wouldn't boot, but eventually, after booting back into Recovery, and disabling authenticated -root and csrutil, the Mac was working again, after I restarted it)

3. Back on MacOS, I ran the three terminal lines from @chaealarm 's comment:

sudo chown -R root:wheel /Library/Extensions/horndis.kext
sudo chmod -R 755 /Library/Extensions/horndis.kext
sudo kextload -t /Library/Extensions/horndis.kext

it came up with a pop-up, saying I needed to changed the security settings in recovery to run the extension, so
4. I restarted to recovery, changed the settings in the startup security utility to Reduced Security, and checked both the boxes underneath, and restarted the Mac (only for it to not boot to MacOS again (lol))
5. I booted to recovery again, ran csrutil disable  and csrutil authenticated-root disable, rebooted
6. ran the three terminal lines again, and this time, it just said I needed to restart the computer to use them. I restarted, and then it worked!

It was a bit of an ordeal 😅

dave-moriarti, 2021-04-15:
@tvqt Thank you for being so helpful! 

I did everything as described, 
When I enter the line:

sudo kextload -t /Library/Extensions/horndis.kext

I get no pop-up but this message from terminal:

Notice: -print-diagnostics (-t) ignored; use kextutil(8) to test kexts.
Executing: /usr/bin/kmutil load -p /Library/Extensions/horndis.kext
Error Domain=KMErrorDomain Code=71 "Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext" UserInfo={NSLocalizedDescription=Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext}

And this is the point where I get stucked... Any idea?
We were so close!

tvqt, 2021-04-16:
@dave-moriarti I had a bit of a search around the internet, and the error code seems to indicate that kmutil cannot find a part of the kext, the plist file. A few things I would try, to troubleshoot it:

- running kextutil, to see if the kext file has any issues/wasn't a bad download
- check csrutil status and csrutil authenticated-root status both return disabled for me, (so probably should return disabled for you too)
- downloading the kext again from the Korean site- the file should have a little cube-like icon in Finder- copying it to /Library/Extensions (I misspoke before, it's not ~/Library/Extensions, but /Library/Extensions) running the three lines listed before, and seeing if that returns anything

let me know if any of this helps!

dave-moriarti, 2021-04-16:
@tvqt  (and @chaealarm) 
I downloaded the kext from korean site again and checked everything from scratch. 

authenticate-root was enabled and so was Safe Startup option. I followed your steps and I tried everything again with this configuration:

a. I'm running MacOS with Reduced Security option flagged from Startup Security Utility (underneath boxes both checked)
b. csrutil status --> returns disabled
c. csrutil authenticated-root status --> returns disabled

When i run the three command lines:
1. sudo chown -R root:whell /Library/Extensions/horndis.kext --> asks for password and gives no problem
2. sudo chmod -R 755 /Library/Extensions/horndis.kext --> runs without saying a thing
3. sudo kextload -t /Library/Extensions/horndis.kext --> gives this message

Notice: -print-diagnostics (-t) ignored; use kextutil(8) to test kexts.
Executing: /usr/bin/kmutil load -p /Library/Extensions/horndis.kext
Error Domain=KMErrorDomain Code=71 "Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext" UserInfo={NSLocalizedDescription=Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext}

At this point I ran kextutil as suggested:

4.kextutil /Library/Extensions/horndis.kext --> You must be running as root to load kexts or send personalities into the kernel.
5. su
6. password
7. kextutil /Library/Extensions/horndis.kext ---> gives same error as 3) 

Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/horndis.kext
Error Domain=KMErrorDomain Code=71 "Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext" UserInfo={NSLocalizedDescription=Loading extension failed: Unable to create extension from bundle: codeless (missing CFBundlePackageType) bundle (nil) at /Library/Extensions/horndis.kext}

I also noted that other users up in this thread had same error code 

Error Domain=KMErrorDomain Code=71 

but different specification:

Incompatible architecture: Binary is for x86_64, but needed arch arm64e" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for x86_64, but needed arch arm64e}

At this point is it possible that the korean file isn't working properly? It's like the kext file is missing some part I guess.

dave-moriarti, 2021-04-16:
********* Update: SOLVED (in the most stupid and nonsense way). But I'm not sure of what i've actually done.

Having excluded every other problem I insisted with working on the korean kext. 

I tried unzipping it multiple times with different programs but the problem was still there, until I accidentally unzipped the archive in a folder were there was already the extracted kext file . 

MacOS default unzipper found that there was already a file with that name in the folder, and renamed it "horndis.kext 2". Adding the 2 to the extension made the finder not recognising it as a .kext and instead showed it as a folder .

Opening the folder, there was the .plist file that @tvqt mentioned and that was missing, causing the Error 71 when trying to load the kext from terminal.

I copied the entire folder in /Library/Extensions and renamed it "HoRNDIS.kext", system asked to conferm extension change. It now appeared as a brick again. And it was infact identical to the previous kext. But when I hopelessly tried to run the 3 command lines again, it installed successfully!

I don't know exactly what happened. Maybe changing the file extension kinda forced a sort of extension's internal indexing and allowed the extension to find the .plist, but I am just guessing.

I'm writing everything here so that noobs like me struggling with the installation know what worked for me. 

Thanks everyone, now my Macbook M1 can browse the internet via Android USB Tethering!
This may also be useful connecting to DJI's drone interface in a M1 Mac.

P.S. It's surprising that Apple didn't take care of a feature so easy for them to add. Thank god the  internet community exist!

tvqt, 2021-04-17:
glad I could help 😊

qugu, 2021-04-28:
Hi, another M1 user here. Can indeed confirm that the kext from the Korean site works, but this installation required some back and forth with restarting and calling `kextload` a few times. The final time it worked, the Security Preferences window had a pop-up showing something like 'Reindexing / reloading extensions...' and after a followed reboot it worked. As a side note, only a native type-c to type-c cabe worked when connecting the phone to the computer. 

Saved me from high-latency death with bluetooth tethering 🥇

dave-moriarti, 2021-04-28:
I'm tethering trough a type-c to usb-a cable, with apple usb-a to usb-c adapter, and it works just fine

thpryrchn, 2021-06-01:
For those of you who want to get this up and running for Big Sur, I have crudely modified the MakeFile and xcode project to make a pkg file that will work on M1 and most likely Intel Big Sur (I haven't tested Intel). 

First step is to allow unsigned kext's to be loaded on your M1 Mac. to do that with a Macbook, power on the laptop holding the power button until it loades the options window. Should show your SSD, and a Options that has a setup icon on it. Select that. Then it will most likely have you select a user. Select one, and enter the password. This is so it unlocks the Drive. Then in the menu bar after that, under Utilities, select Terminal. then type in `csrutil enable --without kext`, and press enter, Then it will ask you to confirm that is what you want to do, and even say it is a unsupported config. confirm it. then reboot your mac.

After that, clone this from my repo. `git clone  https://github.com/thpryrchn/HoRNDIS.git -b BigSur` , then `cd HoRNDIS`, then run the `make` command. IF that doesn't work, make sure you have XCode installed.. Then in your build/pkg folder, you will see "HoRNDIS-kext.pkg". Install that, then enable it in System Prefrences, Security & Preferences. Then it will have you reboot to make it take effect.

Joll123, 2021-06-25:
@tvqt Would you tell me the command lines of the Korean site , because I don't find it. Thanks.

tvqt, 2021-06-27:
> @tvqt Would you tell me the command lines of the Korean site , because I don't find it. Thanks.

the link from the @chaealarm's comment:
> 3. In recovery mode, open the upper utility-terminal and enter the command below

 

csrutil disable

 

(Enter password)

 

csrutil enable --without kext

 

(Enter password)

출처: https://itap.tistory.com/2 [Chaealarm's Project]

FawenYo, 2021-07-10:
Is there any way to use this without disable csrutil disable? Cause that will make sideload app not openable...

tvqt, 2021-07-13:
> Is there any way to use this without disable csrutil? Cause that will make sideload app not openable...

There doesn't appear to be good documentation online about this- you can try experimenting with using things like csrutil enable --without kext, and seeing if they work- anecdotally, it seems like any non-default SIP configuration (i.e. disabling it, or disabling certain parts of it) would appear to make MacOS block sideloading (but you're welcome to try- no guarantees of anything)

tvqt, 2021-07-13:
> @dave-moriarti I meet this problem. Would you know how to solve it?
> Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/horndis.kext
> Error Domain=KMErrorDomain Code=71 "Incompatible architecture: Binary is for arm64e, but needed arch x86_64
> Unsupported Error: one or more extensions are unsupported to load Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /Library/Extensions/HoRNDIS.kext" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for arm64e, but needed arch x86_64
> Unsupported Error: one or more extensions are unsupported to load Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /Library/Extensions/HoRNDIS.kext}

do you have an M1 Mac? "Binary is for arm64e, but needed arch x86_64" seems to indicate you have an x86-based CPU

L1cardo, 2021-11-15:
I have a developer account, just tell me how to build it and sign it? I will do it.

chaealarm, 2021-11-15:
> I have a developer account, somebody just tell me how to build it and sign it. I will do it.

Do you have kext signable account?
Sadly, I have developer but apple don't grant me to sign kext because I am personal developer.

L1cardo, 2021-11-18:
> > I have a developer account, somebody just tell me how to build it and sign it. I will do it.
> 
> 
> 
> Do you have kext signable account?
> 
> Sadly, I have developer but apple don't grant me to sign kext because I am personal developer.

This is sad. Mine is personal developer account as well. I didn't know that personal developer can't sign kext. :(

chaealarm, 2021-12-09:
I just built PCB Design that support ethernet tethering between Android-Apple Silicon Mac.
(Not for RNDIS, Not tested yet)
You could be use it without disabling SIP.
https://github.com/chaealarm/DualUSBETH
(Of course, can run iOS sideload apps.)

ono7, 2022-01-04:
> Hello everyone,
> 
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
> 
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip)
> 
> **Installation steps**
> 
> 1. Disable SIP for kernel extensions
>    1.1. Restart your Mac in recovery mode by powering it off then restarting with a long press on the power button
>    1.2. Execute `csrutil disable` **then** `csrutil enable --without kext`
>    1.3. Restart mac to normal mode
> 2. Run the .pkg that you'll find in the .zip I uploaded.
> 3. Try tethering data with a nearby Android phone -- currently using my data plan ;)

thank you thank you thank you!

xros, 2022-04-25:
Can anyone (verified developer) just sign the the build? I know it works but disabling kext protection totally on macos isn't secure.

thpryrchn, 2022-04-25:
> I just built PCB Design that support ethernet tethering between Android-Apple Silicon Mac. (Not for RNDIS, Not tested yet) You could be use it without disabling SIP. https://github.com/chaealarm/DualUSBETH (Of course, can run iOS sideload apps.)

Awesome! It may be cheaper to just get 2 of these: https://www.amazon.com/Ethernet-Adapter-ABLEWE-Network-Compatible/dp/B0837QGZMY and put a Ethernet cable together, as that is basically what you did on the hardware level... But your setup also charges the phone at the same time right?

chaealarm, 2022-04-27:
> Awesome! It may be cheaper to just get 2 of these: https://www.amazon.com/Ethernet-Adapter-ABLEWE-Network-Compatible/dp/B0837QGZMY and put a Ethernet cable together, as that is basically what you [...]

That's right. 
Some phones are not be supported to charge.

Using cheap pcb manufacturer(like JLCPCB), might be cheaper than buying 2 of USB Ethernet adapter.

krystianmagdziarz, 2022-07-03:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Epic. Thanks!

brunoaduarte, 2022-07-25:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Hi, solution does not work for latest Mac Monterey 12.4 

[image]

[image]

brunoaduarte, 2022-07-27:
Any updates on this?

brunoaduarte, 2022-07-28:
> The Xcode on my Mac is very old because my Mac is still running Mojave; the last Xcode that runs on Mojave is Xcode 11.3. You need Xcode 12 to build for Apple Silicon. It's open source software, and you're free to build it and sign it yourself. I don't see why I should go upgrade my production Mac environment to do this, though.
> 
> To be honest, I am not sure what my incentive is here to spend time on this. I haven't touched HoRNDIS in over five years; countless large organizations -- most notably, DJI, but the Beagleboard folks too -- distribute my code with their commercial application (and, in fact, rely on my code for their product to work at all on Mac), and I have never seen a letter of thank you from them, let alone a product or a dollar bill: the only thing I see from them is complaints from their users when my code goes wrong. If you want to send me a Mac to get this working on, that would make this project actually interesting again. But, in the seven years since I initially relesed it, I haven't seen anyone "really need"ing it enough to feel that it's worth sponsoring.

Why don't you enable "Sponsor this project" so everyone can donate/sponsor it? In a few days you will be gettin a new mac.

brunoaduarte, 2022-07-28:
> The Xcode on my Mac is very old because my Mac is still running Mojave; the last Xcode that runs on Mojave is Xcode 11.3. You need Xcode 12 to build for Apple Silicon. It's open source software, and [...]

Hi @jwise , why don't you enable "Sponsor this project" so everyone can donate/sponsor it? In a few days you will be gettin a new mac.

dimmyv, 2022-08-15:
> thank you thank you thank you!

For anyone having trouble updating their DJI FPV Goggle V2 using M1 MacBook on macOS Monterey

This worked for me. Follow the instructions and then the goggles will update.

I also want to say thank you for this fix. 

> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Thank you for this.

For anyone who has purchased DJI FPV Goggles V2 and running M1 Mac with macOS Monterey.

This solution worked for me and managed to update my Goggles to the latest firmware. I confirm successful update as of 15 August 2022.

I ran this on the Mac not Windows, even after DJI Assistant download states its not working, it worked with this solution.

Thanks again.

inspiredearth, 2022-10-11:
> Why would I distribute something I couldn't test? That would be irresponsible of me.

I am sure there's plenty of people more than happy to test it for you, and report their results. Release it as a "beta" if that feels more responsible, and once the beta testing results come in from the public, you're good to go.

LayerCorp, 2022-10-12:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Big thanks! is working flawless in my Macbook Pro M1 Pro with Monterey.

I have a few questions, maybe someone can help me.

Since I did this, I cant open iPad Apps in my mac, I think is because kext are disabled in my system? If so, is there a way to re-enable the kexts on my computer and how? Also, if I re-enable the kexts I will lost the Horndis connection or will stay working? I want to use horndis and also my ipad apps if possible.

Thanks!

samip5, 2022-10-16:
Hmm, can I actually load a custom system extension with kext enabled aka can I sign my own modules so I can actually keep it enabled?

dmytro-fedko, 2022-10-17:
Bump. MacOS 12.5.1 (M1)

hyeonchang, 2022-11-07:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

This is very helpful for me. Thanks.

I have tested this on M1 (MacOS Ventura). 

https://support.apple.com/guide/security/startup-disk-security-policy-control-sec7d92dc49f/web

This article is helpful for me to enable the HoRNDIS. 

[Summary]

Step.1. run "crsutil disable" then reboot. (in recovery mode)
Step.2. install HoRNDIS.
Step.3. reboot and entering recovery mode.
Step.4. check "allow user menagement... " in permissive security below.

dimmyv, 2022-12-01:
> This is very helpful for me. Thanks.
>
> I have tested this on M1 (MacOS Ventura).
>
> https://support.apple.com/guide/security/startup-disk-security-policy-control-sec7d92dc49f/web
>
> This article is [...]

I can confirm this solution still works when updating your DJI Goggles V2 to latest firmware ending .0607 

https://youtu.be/1Y-Hgcz1sT4

Thanks again Brian for this download, much appreciated.

samip5, 2022-12-01:
Any ideas as to how to get this WITHOUT running crsutil disable?

dimmyv, 2022-12-01:
> Any ideas as to how to get this WITHOUT running crsutil disable?

Run what you need to update and once you have updated go back into recovery mode and turn the SIP settings back to default.

samip5, 2022-12-01:
> > Any ideas as to how to get this WITHOUT running crsutil disable?
> 
> Run what you need to update and once you have updated go back into recovery mode and turn the SIP settings back to default.

Huh? I don't want to disable it to begin with.

samip5, 2022-12-01:
> Can anyone (verified developer) just sign the the build? I know it works but disabling kext protection totally on macos isn't secure.
> 
> [HoRNDIS-kext.pkg.zip](https://github.com/jwise/HoRNDIS/files/9949151/HoRNDIS-kext.pkg.zip)
> 
> Here's my build.

Can you explain why it needs a verified developer to sign it?

Gorgonthedestroyer, 2023-02-03:
ATTENTION M1/M2 USERS ON OS VENTURA
(Sorry for yelling, I just spent a few hours fixing this issue)

There is a new set of instructions to follow if you want to make this work given the layout of some things has changed. 

1. Shut down, long hold the power button to open _Disk Recovery_
2. In the top bar options click _Utility_ and then click _Terminal_ (once in the terminal put in: **csrutil disable** (You may be asked for a password, enter it, then you’ll be asked **Y/N** put in **Y** then click enter.)
3. Go out of the terminal in your top bar options, (Select _Utilities_, where you found _Terminal_.)
5. Click _Security_ (the top option)
6. Open _Security_, under permissible access (the bottom of the 3 screen options, you didn’t have this option before and now you do) you’ll want to select the top box of the 2 options given (DO NOT SELECT THE BOTTOM ONE AS WELL). This is a necessary step if you skip this then it won’t work in my experience. 
7. Click the _Apple_ logo and _Restart_.
8. Go try it again. After opening the Kext it will need to be allowed 2 times in _security_ (have this open when installing .kext) then it will restart. 
9. Open Dji assistant and it should work. I just spent 2 hours doing this process over again after following the instructions above.

Let me know if this works for anyone else. There is some better terminology I'm sure I could have used but someone can refine this list if they like haha.

achandec, 2023-03-01:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Just tried it today, and this just solved my issue. Thanks for uploading this. After 3 hours, I am finally able to connect my DJI Mini 3 Pro to DJI Assistant 2 (Consumer Drones) and calibrate it.

achandec, 2023-03-01:
> **date && hostname && sw_vers && kmutil dumpstate | grep --invert-match com.apple**

This one also worked for me.

idDreameR, 2023-04-12:
> Hello everyone,
>
> I'm uploading my M1 binary so that other people don't have to install Xcode on their 256GB mac ;)
>
> [HoRNDIS-M1.zip](https://github.com/jwise/HoRNDIS/files/7323710/HoRNDIS-M1.zip) [...]

Работает на чипе M2! СПАСИБО!!!

xjzyy, 2023-06-15:
我执行：sudo kextload -t /Library/Extensions/horndis.kext 的时候报下面的错： 有人遇到过吗？

Notice: -print-diagnostics (-t) ignored; use kextutil(8) to test kexts.
Executing: /usr/bin/kmutil load -p /Library/Extensions/horndis.kext
Error Domain=KMErrorDomain Code=29 "Authenticating extension failed: Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /private/var/db/KernelExtensionManagement/Staging/com.joshuawise.kexts.HoRNDIS.3F64G1/HoRNDIS.kext:

Authenticating extension failed: Bad code signature" UserInfo={NSLocalizedDescription=Authenticating extension failed: Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /private/var/db/KernelExtensionManagement/Staging/com.joshuawise.kexts.HoRNDIS.3F64G1/HoRNDIS.kext:

Authenticating extension failed: Bad code signature}

sumsethan, 2023-07-11:
> 我执行：sudo kextload -t /Library/Extensions/horndis.kext 的时候报下面的错： 有人遇到过吗？
>
> Notice: -print-diagnostics (-t) ignored; use kextutil(8) to test kexts. Executing: /usr/bin/kmutil load -p [...]

至少要把SIP的Kext关了，步骤上面说的很清楚了