jwise/HoRNDIS #138: HoRNDIS fails when the smartphone is connected to a dock

dtoniolo, 2021-04-16:
Hi,
I'm a long-time HoRNDIS user and I've never had any issues. However, recently I started connecting my smartphone to a dock, instead of going in directly into the Mac USB port. The tethering works for a while, but then stops and I have to restart the computer to solve the issue. Unloading and loading the kernel extension doesn't work.

ngfuong, 2021-09-25:
Hi, have you found any way to avoid this?
I plugged my smartphone to my Dell XPS via a USB dock and it also stops randomly with the log `processing a network change notification` and a notification `PKDownloadError error 8`
IMO usb dock is the root of all evil. However, when I try connecting it directly to the usb slot via a usb-c to usb-c cable, it also fails.

dtoniolo, 2021-09-28:
Hi, no, I didn't. I ended up just using the wifi hotspot instead of the USB tethering. No other alternative, as right now HoRNDIS doesn't even work on macOS Big Sur

ngfuong, 2021-10-06:
Thank you for replying.
I ended up creating an offline installer of Big Sur and avoided using HoRNDIS.