jwise/HoRNDIS #139: DriverKit Port

sgulls, 2021-04-22:
Clearly the writing is on the wall for KEXTs. We can get away with it for now by running `bputil -k` in 1TR for arm Macs but sooner or later KEXTs are going to go the way of 32 bit support and it's clear the original developer/maintainer cannot/will not port the code so it's up to us honestly.

I'm not that experienced with Obj-C and I've played around with the code but not everything has a drop in replacement from kernel IOKit to DriverKit so I'm in way over my head. 

among some libraries/headers that I'm figuring out how to replace

In the headers file:
`IOKit/usb/IOUSBHostFamily.h`
`IOKit/network/IOEthernetController.h` & `IOKit/network/IOEthernetController.h` (I guess `IOUserNetworkEthernet` is probably the way to go but it's not quite the same)
`IOKit/IOTimerEventSource.h`
`IOKit/IOMessage.h`

in the main cpp file:
`mach/kmod.h` & `libkern/version.h` (they don't seem necessary given we're moving code away from the kernel)
`IOKit/network/IOGatedOutputQueue.h`

In addition, DriverKit now enumerates interfaces using a c++ class in an `.iig` file which handles starting, stopping and cleanup versus the kext which has everything in the `.cpp` body so reformatting the code will be necessary. 

I only have DriverKit entitlements for a few VID/PIDs as I have to request each pair individually from Apple so I only bothered request devices I personally own. With that being said, an unsigned driver extension is vastly preferably to running in the kernel as evidenced by the kernel panics and crashes this software has wrought so far. 

Anyone with any Obj-C experience willing to pitch in? I'll try learning as I go but I can't do this by myself.

sgulls, 2021-04-22:
starting my fork [here](https://github.com/sgulls/netlink)

Belcarra, 2021-04-25:
I have a working DriverKit networking driver for RNDIS, ECM, NCM, EEM etc.

I hope to make it available in the Mac store eventually. But can make it
available to HoRNDIS users to use now, just finalizing the Installation
application.

I have the same issue with VID/PID's, I need to request each pair
individually from Apple. Although I have my own VID (0x15ec) that I can
allocate any PID with.

If anyone wants to try it contact me with the VID/PID you need.

jwise, 2021-04-25:
Here's a dumb question: can you use DriverKit for the network portion, and libusb for the USB portion?  For a while I was thinking that the future of HoRNDIS would be tun/tap-based for network and libusb for USB.

krjw-eyev, 2021-06-10:
@Belcarra that sounds great! When are you planning to release it?

@sgulls I have looked at your repository and have noticed that you added code for a SwiftUI App. What was your plan? How far did you come porting it to DriverKit?

YiquanGu, 2021-07-24:
> starting my fork [here](https://github.com/sgulls/netlink)

Great initiative! Is your version working on M1 Big Sur?

vricosti, 2022-09-22:
Any news about this ?