jwise/HoRNDIS #142: Big Sur not supported!

CodeChari, 2021-05-29:
Error: macOS El Capitan, Sierra, High Sierra, or Mojave is required for this software.

PrincipeAzzurro, 2021-05-30:
I can confirm that experience - please have a look at the screenshot attached: 
[image]

PrincipeAzzurro, 2021-05-30:
In a related issue, the most recent version of Homebrew seems to not like the "cask" command anymore. See above screenshot "unknown command: cask"

PrincipeAzzurro, 2021-05-30:
Also, the latest installer 9.2 from Joshua's site won't install either (macOS 11.4, MacBook Pro 16,1:
[image]

danielehrhardt, 2021-06-03:
> Also, the latest installer 9.2 from Joshua's site won't install either (macOS 11.4, MacBook Pro 16,1:
> [image]

Same for M1 Macbook Pro 13. BigSur 11.5 Beta

krjw-eyev, 2021-06-12:
try [this solution](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-745607263)

Joll123, 2021-06-25:
@PrincipeAzzurro have you successfully installed horndis? my pc is same with yours.

Joll123, 2021-06-26:
@all Have anyone installed Horndis on MacBook Big Sur 11.4 successfully?

martin-braun, 2021-06-27:
> try [this solution](https://github.com/jwise/HoRNDIS/issues/132#issuecomment-745607263)

But how to get the kext file? I want to try to install it on 11.4.