jwise/HoRNDIS #146: HoRNDIS 9.2 works perfectly on Apple Silicon (M1), macOS 12 Monterey with no code changes, as long as you compile the kext from source!

akemin-dayo, 2021-11-24:
I'm glad to report that the latest version of HoRNDIS (9.2) works perfectly on Apple Silicon machines with no code changes required!

Basically, just compiling an additional arm64e (not arm64) binary slice to the HoRNDIS kext works!

Tested on macOS 12.0.1 21A559, "Monterey".

---

## If you're a general user coming who just wants to use HoRNDIS on Apple Silicon right this instant…
Ever since macOS / OS X 10.10 Yosemite, Apple has required kernel extensions (kexts) to be signed by developers who are subscribed to the Apple Developer Program, who _also_ have to apply for a special kext signing privilege.

You're… probably not one of those people. ;P

Luckily, there _is_ a way for you to sign your own kexts using an Xcode feature called ad-hoc signing! But it does require changing some settings first.

### Switching to "Reduced Security" mode
If you've _already_ placed your Mac in "Reduced Security" mode before, simply skip this section.
1. Shut down your Apple Silicon Mac.
2. Press and hold down the power button _until_ the text under the Apple logo says "Loading startup options…", then let go.
3. Select "Options".
4. You are now in recoveryOS — enter your password if it asks.
5. Go to Utilities → Startup Security Utility.
6. Select "Reduced Security" and enable Allow user management of kernel extensions from identified developers".
7. Shut down your Apple Silicon Mac.

### Disabling SIP (System Integrity Protection)
**IMPORTANT:** Disabling SIP in _any_ capacity, even partially, will also disable Apple Pay, as well as any iOS-on-macOS apps you may have downloaded from the App Store. This is a strange (and annoying) decision that Apple has decided to make specifically on Apple Silicon, as Apple Pay actually works fine even when SIP is disabled on x86_64 (Intel) Macs.

1. Follow steps 2〜4 from above.
2. Go to Utilities → Terminal.
3. Type in the following to _fully_ disable SIP: `csrutil disable`
    **Note:** It is possible to only _partially_ disable the part of SIP that enforces kext signature verification (`csrutil enable --without kext `), but according to Apple, this is apparently an unsupported configuration. Use it if you wish, but please read and understand the warning that `csrutil` gives if you try.
4. Reboot your Apple Silicon Mac.

### Compiling HoRNDIS for Apple Silicon (arm64e)
1. Download and install [Xcode](https://apps.apple.com/app/xcode/id497799835).
2. Run the following in a Terminal session. When it asks for your password, it is normal for no characters to show when when you type!
```shell
git clone --recursive https://github.com/jwise/HoRNDIS.git
cd Development/HoRNDIS/
xcodebuild -sdk macosx -configuration Release
sudo cp -rv build/Release/HoRNDIS.kext /Library/Extensions/
```
3. Go to System Preferences → Security & Privacy and approve the HoRNDIS kernel extension.
4. Reboot, connect an Android device in USB tethering mode, and enjoy using HoRNDIS again!

kentcdodds, 2021-12-01:
Any chance someone could do this and upload the resulting `HoRNDIS.kext` file? Will that not work with the ad-hoc signing required?

kentcdodds, 2021-12-02:
In any case, I followed these instructions and it worked for me. Thanks 👍

imekachi, 2021-12-02:
@akemin-dayo Do I have to keep the "Reduced Security" mode on and SIP disabled? Or is it just for the installation and I can safely enable them back on later?

akemin-dayo, 2021-12-03:
> Any chance someone could do this and upload the resulting `HoRNDIS.kext` file? Will that not work with the ad-hoc signing required?

@kentcdodds While I _can_ do that and provide a downloadable precompiled `HoRNDIS.kext`, I do not think it is good security hygiene to encourage users to download a compiled binary kernel extension from effectively, some random person who is not actually affiliated with the HoRNDIS project.

> @akemin-dayo Do I have to keep the "Reduced Security" mode on and SIP disabled? Or is it just for the installation and I can safely enable them back on later?

@imekachi Reduced Security _must_ be kept enabled if you want to use _any_ kernel extensions on Apple Silicon macOS at all. This includes a future, fully-signed binary release by @jwise, if one ever comes. (As well as any other kernel extensions used by any software you may be using, such as NoSleep, Soundflower, TunTap, Paragon NTFS, etc.)

SIP on the other hand, only needs to be kept enabled as long as you are using unsigned / ad-hoc signed kernel extensions, like the one that my instructions creates. This means that if a future, fully-signed HoRNDIS arrives, you can re-enable SIP and still retain kernel extension functionality.

chaealarm, 2021-12-07:
Same solution as previous issue "M1 Support?"
https://github.com/jwise/HoRNDIS/issues/135#issuecomment-791469750

the kext built for arm64 : https://blog.kakaocdn.net/dn/nRMG3/btqZkqGNzgR/U1XLxka8DvDA5zgx4X1yY0/HoRNDIS.kext.zip?attach=1&knm=tfile.zip

As you said, the critical problem is 'Can't run iOS App Sideload', because SIP is disabled.

So, there are two solutions.

1. Sign the kext.
  - To sign kext, kext singable Apple developer account needed.
  - But, Apple won't grant to individual developer.(I've tried to ask it, but failed.)
2. DriverKit port.
  - DriverKit can use some allowed VIDs.
  - But RNDIS devices, VIDs and PIDs are various.
  - So, hard to apply DriverKit.

AlessandroChecco, 2022-01-28:
> I'm glad to report that the latest version of HoRNDIS (9.2) works perfectly on Apple Silicon machines with no code changes required!
> 
> Basically, just compiling an additional arm64e (not arm64) binary slice to the HoRNDIS kext works! @jwise
> 
> Tested on macOS 12.0.1 21A559, "Monterey".
> 
> (I do realise that this isn't really an _issue_ per se, but I just felt really bad for all the users complaining about broken functionality here…)
> 
> (Plus, speaking from my own personal experience as a fellow developer, I _do_ find it quite helpful when someone else already tested compatibility for me ;P)
> 
> ## If you're a general user coming who just wants to use HoRNDIS on Apple Silicon right this instant…
> Ever since macOS / OS X 10.10 Yosemite, Apple has required kernel extensions (kexts) to be signed by developers who are subscribed to the Apple Developer Program, who _also_ have to apply for a special kext signing privilege.
> 
> You're… probably not one of those people. ;P
> 
> Luckily, there _is_ a way for you to sign your own kexts using an Xcode feature called ad-hoc signing! But it does require changing some settings first.
> 
> ### Switching to "Reduced Security" mode
> If you've _already_ placed your Mac in "Reduced Security" mode before, simply skip this section.
> 
> 1. Shut down your Apple Silicon Mac.
> 2. Press and hold down the power button _until_ the text under the Apple logo says "Loading startup options…", then let go.
> 3. Select "Options".
> 4. You are now in recoveryOS — enter your password if it asks.
> 5. Go to Utilities → Startup Security Utility.
> 6. Select "Reduced Security" and enable Allow user management of kernel extensions from identified developers".
> 7. Shut down your Apple Silicon Mac.
> 
> ### Disabling SIP (System Integrity Protection)
> **IMPORTANT:** Disabling SIP in _any_ capacity, even partially, will also disable Apple Pay, as well as any iOS-on-macOS apps you may have downloaded from the App Store. This is a strange (and annoying) decision that Apple has decided to make specifically on Apple Silicon, as Apple Pay actually works fine even when SIP is disabled on x86_64 (Intel) Macs.
> 
> 1. Follow steps 2〜4 from above.
> 2. Go to Utilities → Terminal.
> 3. Type in the following to _fully_ disable SIP: `csrutil disable`
>    **Note:** It is possible to only _partially_ disable the part of SIP that enforces kext signature verification (`csrutil enable --without kext`), but according to Apple, this is apparently an "unsupported configuration". Use it if you wish (as many do already), but please make sure to read and fully understand the warning that `csrutil` gives if you try.
> 4. Reboot your Apple Silicon Mac.
> 
> ### Compiling HoRNDIS for Apple Silicon (arm64e)
> 1. Download and install [Xcode](https://apps.apple.com/app/xcode/id497799835).
> 2. Run the following in a Terminal session. When it asks for your password, it is normal for no characters to show when when you type!
> 
> ```shell
> git clone --recursive https://github.com/jwise/HoRNDIS.git
> cd Development/HoRNDIS/
> xcodebuild -sdk macosx -configuration Release
> sudo cp -rv build/Release/HoRNDIS.kext /Library/Extensions/
> ```
> 
> 3. Go to System Preferences → Security & Privacy and approve the HoRNDIS kernel extension.
> 4. Reboot, connect an Android device in USB tethering mode, and enjoy using HoRNDIS again!

I followed the steps using last version of monterey and xcode. Unfortunately I get kernel panic.
Can you let us know which version of Xcode you used?
Thanks!

fxh-lab, 2022-03-16:
有用

bs702, 2022-06-02:
@akemin-dayo Thanks a lot for your help. but it still didn't work and I got the following error

Executing: /usr/bin/kmutil load -p /Library/Extensions/HoRNDIS.kext
Error Domain=KMErrorDomain Code=29 "Authenticating extension failed: Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /private/var/db/KernelExtensionManagement/Staging/com.joshuawise.kexts.HoRNDIS.ZoHTq5/HoRNDIS.kext:

Authenticating extension failed: Bad code signature" UserInfo={NSLocalizedDescription=Authenticating extension failed: Kext com.joshuawise.kexts.HoRNDIS v9.2 in executable kext bundle com.joshuawise.kexts.HoRNDIS at /private/var/db/KernelExtensionManagement/Staging/com.joshuawise.kexts.HoRNDIS.ZoHTq5/HoRNDIS.kext:

Authenticating extension failed: Bad code signature}

altitudedashboard, 2022-06-17:
Works for me on macOS Monterey 12.4. no Rosetta installed, MacBook Pro M1, Xcode 13.4.1. I need HoRNDIS to connect to an ADALM-Pluto SDR device.

olarroche, 2022-09-21:
Hello everybody

I successfully run the git command above, and then cd to the "HoRNDIS" directory created in my $HOME. But then the command "xcodebuild -sdk macosx -configuration Release" yields an error message :

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

However, I do have Xcode installed, version 13.4.1 on my M1 MacStudio under Monterey 12.6 ... Then I noticed the Makefile in the HoRNDIS directory, so I tried to run "make", after setting the environment variable HORNDIS_XCODE=/Applications/Xcode-13.4.1 . Then I got the following error :

Makefile:20: *** Cannot find xcodebuild under /Applications/Xcode-13.4.1. Please either download Xcode 7.3.1 from: "https://developer.apple.com/download" and install as /Applications/Xcode-7.3.1/ or point HORNDIS_XCODE to your preferred Xcode app path.  Stop.

Then I opened the project file "HoRNDIS.xcodeproj", but the build failed, saying that it could not find the file machine/limits.h and the macosx10.11 sdk...

Any leads to proceed ? It would be nice to be able to use USB tethering since my WiFi connection is so low...

Thank you for your attention,
Olivier

happy-dev, 2022-11-20:
@akemin-dayo Thank you so much for this tutorial! 
Without you I was stuck

MohanPeriyasamy071, 2022-12-16:
Is it possible to transfer internet from mac ventura to an android device through USB cable ??
Can someone please help 

Thanks in advance

akemin-dayo, 2022-12-20:
@MohanPeriyasamy071 This is unrelated to HoRNDIS (and by extension, this issue thread), but you can use `gnirehtet` ("tethering" spelled backwards) for macOS, Windows, Linux to achieve your goal of reverse tethering to an Android device: https://github.com/Genymobile/gnirehtet

MohanPeriyasamy071, 2022-12-20:
> @MohanPeriyasamy071 This is unrelated to HoRNDIS (and by extension, this issue thread), but you can use `gnirehtet` ("tethering" spelled backwards) for macOS, Windows, and Linux to achieve your goal of reverse tethering to an Android device: https://github.com/Genymobile/gnirehtet

Thanks for your help, but even using gnirehtet, iam unable to use my mac's internet in my pixel device

ekawahyu, 2022-12-24:
@akemin-dayo I can confirm that your guide is working on my Apple M1 Pro machine, Monterey 12.6.2, Xcode 13.3. However, my device can only be recognized after load/unload the kext, every time. So the steps are:
1. Unplug my USB device.
2. Plug the USB device back in.
3. sudo kextunload /Library/Extensions/HoRNDIS.kext
4. sudo kextload /Library/Extensions/HoRNDIS.kext

Any tips on how to fix this non plug-and-play issue? Thank you

justtryingthingsout, 2022-12-28:
@ekawahyu this is how I solved that issue:
1. make a plist at /Library/LaunchDaemons (e.g. /Library/LaunchDaemons/com.joshuawise.HoRNDIS-loader.plist)
2. put the following inside the plist:
```plist
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>KeepAlive</key>
        <false/>
        <key>Label</key>
        <string>com.domain.identifier</string>
        <key>ProgramArguments</key>
        <array>
                <string>/sbin/kextload</string>
                <string>/Library/Extensions/HoRNDIS.kext</string>
        </array>
        <key>RunAtLoad</key>
        <true/>
        <key>StandardErrorPath</key>
        <string>/dev/null</string>
        <key>StandardOutPath</key>
        <string>/dev/null</string>
        <key>UserName</key>
        <string>root</string>
</dict>
</plist>
```
3. run `sudo launchctl load -w /Library/LaunchDaemons/your-plist-name-here.plist`

you should now be able to plug and unplug (or reboot) how many times as you want without the kext just unloading.

akemin-dayo, 2022-12-28:
@ekawahyu @justtryingthingsout

Hmn, that's… odd.

At least with HoRNDIS, it doesn't use any of the [deprecated KPIs listed here](https://developer.apple.com/support/kernel-extensions) as far as I can tell, so it should be able to automatically load whenever an Android device in USB tethering mode is connected.

The device IDs that the HoRNDIS kext is specifically looking for are defined in the `IOKitPersonalities` key inside [`Info.plist`](https://github.com/jwise/HoRNDIS/blob/master/HoRNDIS-Info.plist) (※ also available locally on disk at `/Library/Extensions/HoRNDIS.kext/Contents/Info.plist`).

I can personally confirm that this behaviour works as intended, at least on my macOS 12.6.2 21G320 machine with my Android device, a Sony Xperia X Compact F5321 SO-02J running LineageOS.

ekawahyu, 2022-12-28:
@justtryingthingsout Thanks for sending this, but it does not seem to work. When I issue `launchctl list`, I don't see it loaded there. How can I debug this thing?

@akemin-dayo I have also tested this steps on Intel Mac with Big Sur and it does recognize my device only once after system booting up or system reboot. Other than that, kextload/kextunload does not work.

Cornul11, 2023-01-26:
Moreover, if you are using OpenCore or Clover (OpenCore is the way to go these days), you can simply extract the .kext or build from source yourself and add it to your kexts.

bartolomeo902, 2023-06-28:
It worked for me in MacOS Ventura 13.4.1 with an Apple Silicon M1 Max!
Just a note: when you clone from git it will save the repository to your user home folder. 
That means the command you will need to run can be easily fixed:
```shell
git clone --recursive https://github.com/jwise/HoRNDIS.git
cd ~/HoRNDIS/
xcodebuild -sdk macosx -configuration Release
sudo cp -rv build/Release/HoRNDIS.kext /Library/Extensions/
``` 
I will attach my Kext so that somwone may benefit from it without having Xcode installed (you will still need to disable SIP and enter Reduced Security mode)
[HoRNDIS.kext (2).zip](https://github.com/jwise/HoRNDIS/files/11893420/HoRNDIS.kext.2.zip)

david65536, 2023-08-13:
Just taking a moment to confirm that the above also works for Apple M2 (Mac Mini) Mac OS 13.4.1 (22F770820d), allowing connection to BeagleBone single board computer via ssh.

elkSal, 2024-09-24:
Worked on MAC M1 Pro Sonoma:
Followed these steps : https://github.com/jwise/HoRNDIS/issues/146#issue-1062238006
Used this updated command for installing on terminal: https://github.com/jwise/HoRNDIS/issues/146#issuecomment-1611221420
Thanks a lot to everybody!

d235j, 2024-09-24:
> Just taking a moment to confirm that the top instructions work for Apple M2 (Mac Mini) Mac OS 13.4.1 (22F770820d), allowing connection to BeagleBone single board computer via ssh.

Please note — you don't need this (and shouldn't install it) if your BeagleBone Linux image is reasonably up-to-date.

BSthun, 2024-12-27:
Worked with M4:
1. turn off sip: https://developer.apple.com/documentation/security/disabling-and-enabling-system-integrity-protection
2. follow the step per previous comment
3. after cp kext file, go to system settings > secirity & privacy and allow extension to load, then reboot.

kalimbamusic, 2024-12-28:
can you turn on sip after finishing the steps?
I saw tutorial on youtube with "csrutil enable -- without  kext" command and now kinda wish I used it as well. 

anyway, worked for me on m1 somona. disabled sip & allowed kernel extensions from recovery. then ran:
> ```shell
> git clone --recursive https://github.com/jwise/HoRNDIS.git
> cd ~/HoRNDIS/
> ```
[downloaded the kext file bartolomeo902 provided](https://github.com/jwise/HoRNDIS/files/11893420/HoRNDIS.kext.2.zip) and copied it to HoRNDIS folder, then:
> ```shell
> sudo cp -rv HoRNDIS.kext /Library/Extensions/
> ```
and allowed it in security.

alexandregz, 2025-03-15:
Thx, works with macbook air M4 👍

hamer13, 2025-05-10:
https://github.com/jwise/HoRNDIS/issues/146#issuecomment-1611221420
Thanks!
It works on M4 pro with Sequoia 15.4.1

zhouzhonghao, 2025-08-19:
Need this.

whollacsek, 2025-08-20:
I was curious recently and plugged my Pixel 7 into my M4 Max, and USB tethering just worked out of the box without HoRNDIS. Not sure what changed between macos and Android

censiir, 2026-02-22:
@whollacsek The Pixel switched to using CDC-NCM (natively supported by macOS), but some Android devices like Samsungs still use RNDIS.