jwise/HoRNDIS #147: Notarized HoRNDIS + HoRNDIS-Consent.pkg

chris1111, 2021-12-18:
Its now possible to install HoRNDIS without Error, I made a macOS Package to do this without issue.
Download ➤ [Notarized HoRNDIS.zip](https://github.com/jwise/HoRNDIS/files/7738545/Notarized.HoRNDIS.zip)
Enjoy 🤙  Follow the steps in the GUIDE Consent HoRNDIS.pdf
[image]

jarryyen, 2021-12-31:
Thank You @chris1111

brucedjones, 2022-01-18:
I am having the same issue as noted here: https://github.com/chris1111/HoRNDIS/issues/4

I couldn't comment there because your repo is archived. I'm on M1 and need to have SIP enabled, I followed the SIP enabled instructions and had the issue @pradhyumk observed

mlostekk, 2022-01-23:
@chris1111 Same issue as above.

pandres95, 2022-10-13:
@chris1111 same issue. Now the repo is not found.

aaronkaiyangg, 2023-05-23:
thx for 分享

closed 2023-06-20 by chris1111

aaronkaiyangg, 2023-06-20:
邮件已收到，会尽快回复，谢谢！