jwise/HoRNDIS #148: Project now support Monterey 12 and BigSur 11

chris1111, 2021-12-18:
I [Have Update the Project for macOS12 and macOS11](https://github.com/chris1111/HoRNDIS)

droidguyg1, 2022-02-12:
Great, @chris1111! Anyone can confirm that it works with Big Sur?  Which minor versions of Big Sur.

solus9, 2022-03-03:
Hi @droidguyg1 , Can't confirm for Big Sur but I have just installed this in intel based MacBook Pro with macOS Monterey. I did the installation from pre-build release package file from the instructions given there. It all work just fine, all I did was follow the steps. I think it's safe to assume that it will gonna work in Big Sur as well.

Thanks @chris1111, great work.

clauderoy790, 2022-05-27:
The link is dead, can someone post the new link if it's still available?

solus9, 2022-06-22:
Hi, sorry missed this one. You guys can check out @chris1111 's other repositories, I found this [one](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter). As far as I understand from the readme, it should work.

closed 2023-06-30 by chris1111